pbatard/rufus #1541: Errors encountered regarding the compilation on Linux by MinGW

Ricky-Tigg, 2020-05-18:
Hey. I still have had to rely on Microsoft Windows in order to keep using that best in class application – **version 3.10**,  Though i started a project to compile it on Linux side, The respectable _Wine_ does not list Linux resources, hence does not allow the selection of a source ISO file, which is another source of motivation to accomplish that goal. On that site, the compilation instructions are clear enough while minimalist.

- _Visual Studio 2019_ (version 16.0 ; SDK 10.0.18362 –  invoke  .sln; _Use [...] Visual Studio Community Edition to build, run or develop for Rufu_s.
- _MinGW_ – invoke `./configure`/`make`.

That _Community Edition_ is not yet intended for Linux. Then i started a project to accomplish the installation from code source using MinGW, to a Fedora system. I found out that required packages for an x86_64 architecture were mingw64-gcc, mingw64-gcc-c++, mingw64-zlib, mingw64-pkg-config.
```
$ dnf list installed mingw64-{gcc,gcc-c++,zlib,pkg-config}
mingw64-gcc.x86_64                        9.2.1-6.fc32
mingw64-gcc-c++.x86_64                    9.2.1-6.fc32
mingw64-pkg-config.x86_64                 0.28-14.fc32
mingw64-zlib.noarch                       1.2.11-2.fc32
```
Packages _dh-autoreconf_, _automake_, were required as well:
```
$ dnf list installed dh-autoreconf automake
automake.noarch                               1.16.1-14.fc32
dh-autoreconf.noarch                          19-7.fc32
```
_autoreconf_ indicated that **m4 folder** was missing, so I created it. At that state, the classical prerequisite checks returned as follows:
```
$ autoreconf -vif calls aclocal
autoreconf: 'configure.ac' or 'configure.in' is required
```
Those as required mentioned component, hence missing, indicate an issue I believed to be related to automake, hence reported accordingly. The following output errors may not be related at all to Rufus. I need confirmation from a developper though.
```
$ git clone git://github.com/pbatard/rufus && cd rufus
$  ./configure && make check
Making check in src
make[1]: Entering directory '/home/yk/Lataukset/rufus/src'
Making check in bled
make[2]: Entering directory '/home/yk/Lataukset/rufus/src/bled'
  CC       libbled_a-bled.o
In file included from bled.c:14:
libbb.h:20:2: error: #error Only Windows platforms are supported
   20 | #error Only Windows platforms are supported
      |  ^~~~~
In file included from libbb.h:24,
                 from bled.c:14:
./../msapi_utf8.h:22:10: fatal error: windows.h: No such file or directory
   22 | #include <windows.h>
      |          ^~~~~~~~~~~
compilation terminated.
make[2]: *** [Makefile:339: libbled_a-bled.o] Error 1
make[2]: Leaving directory '/home/yk/Lataukset/rufus/src/bled'
make[1]: *** [Makefile:500: check-recursive] Error 1
make[1]: Leaving directory '/home/yk/Lataukset/rufus/src'
make: *** [Makefile:294: check-recursive] Error 1
```

pbatard, 2020-05-18:
> `#error Only Windows platforms are supported`

means that you are invoking the regular `gcc` instead of MinGW `gcc`.

Here's what you need to do (provided you are using `x86_64-w64-mingw32`, if `x86_64-w64-mingw32-gcc -v` returns nothing then you will need to adapt `--host` accordingly):

```
./bootstrap.sh
./configure --host=x86_64-w64-mingw32
make
```

You also need to make sure that your version of MinGW is recent enough, or you're going to run into `error: ‘PROCESSOR_ARCHITECTURE_ARM64’ undeclared`.

closed 2020-05-18 by pbatard

Ricky-Tigg, 2020-05-18:
Above mentioned  `make`output was produced then while i had that output for `x86_64-w64-mingw32-gcc -v`. Is that last output the correct one in order to invoke the MinGW?
```
$ x86_64-w64-mingw32-gcc -v
Using built-in specs.
COLLECT_GCC=x86_64-w64-mingw32-gcc
COLLECT_LTO_WRAPPER=/usr/libexec/gcc/x86_64-w64-mingw32/9.2.1/lto-wrapper
Target: x86_64-w64-mingw32
Configured with: ../configure --prefix=/usr --bindir=/usr/bin --includedir=/usr/include --mandir=/usr/share/man --infodir=/usr/share/info --datadir=/usr/share --build=x86_64-redhat-linux-gnu --host=x86_64-redhat-linux-gnu --with-gnu-as --with-gnu-ld --verbose --without-newlib --disable-multilib --disable-plugin --with-system-zlib --disable-nls --without-included-gettext --disable-win32-registry --enable-languages=c,c++,objc,obj-c++,fortran --with-bugurl=http://bugzilla.redhat.com/bugzilla --enable-threads=posix --with-cloog --enable-libgomp --target=x86_64-w64-mingw32 --with-sysroot=/usr/x86_64-w64-mingw32/sys-root --with-gxx-include-dir=/usr/x86_64-w64-mingw32/sys-root/mingw/include/c++
Thread model: posix
gcc version 9.2.1 20190827 (Fedora MinGW 9.2.1-6.fc32) (GCC)
```
Are suggested commands in case of null output, still valid despite relevant output was not null?

pbatard, 2020-05-18:
Again, the __only__ way you are going to see ` #error Only Windows platforms are supported` and the other error is if you didn't add `--host=x86_64-w64-mingw32` when invoking `./configure` (I do not see any mention of how you invoked `./configure` above) or if your MinGW compiler is broken, which is unlikely.

Be mindful that the `Configure with...` output from `x86_64-w64-mingw32-gcc -v` simply reports what the person who compiled that specific executable (in other words the folks who produced the MinGW archive) used on their machine. But it has no bearing whatsoever on whether __your__ machine will use the same settings automatically. This is unrelated to the `./configure` command I am asking you to type.

In other words, please try the exact commands I gave and report the exact errors you see then, if there are still any.

Ricky-Tigg, 2020-05-18:
Mention is still there, as code; a search for pattern "./configure && make check", could help you. Indeed errors are produced:

```
$ make
[...]
make[2]: Entering directory '/home/yk/Lataukset/rufus/src'
  CC       rufus-stdfn.o
stdfn.c: In function 'GetCpuArch':
stdfn.c:243:7: error: 'PROCESSOR_ARCHITECTURE_ARM64' undeclared (first use in this function); did you mean 'PROCESSOR_ARCHITECTURE_ARM'?
  243 |  case PROCESSOR_ARCHITECTURE_ARM64:
      |       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |       PROCESSOR_ARCHITECTURE_ARM
stdfn.c:243:7: note: each undeclared identifier is reported only once for each function it appears in
make[2]: *** [Makefile:458: rufus-stdfn.o] Error 1
make[2]: Leaving directory '/home/yk/Lataukset/rufus/src'
make[1]: *** [Makefile:500: all-recursive] Error 1
make[1]: Leaving directory '/home/yk/Lataukset/rufus/src'
make: *** [Makefile:294: all-recursive] Error 1
make[1]: *** [Makefile:500: all-recursive] Error 1
make[1]: Leaving directory '/home/yk/Lataukset/rufus/src'
make: *** [Makefile:294: all-recursive] Error 1
```

pbatard, 2020-05-18:
As I said:

> You also need to make sure that your version of MinGW is recent enough, or you're going to run into error: ‘PROCESSOR_ARCHITECTURE_ARM64’ undeclared.

I'm afraid that this is on you to solve, as I am designing a Windows application and therefore only officially supporting a Windows compilation environment (on Windows, you can easily update MinGW to get rid of that error). I may provide hints and tips on how to compile from Linux, as I did above, but in the end, if you are using an outdated version of MinGW, it's up to you to solve that.

Ricky-Tigg, 2020-05-19:
_An outdated version_; Is it even the case? Version against which compilation are made:
```
$ rpm -qi mingw64-gcc | sed -n '10,11p;17p'
Source RPM  : mingw-gcc-9.2.1-6.fc32.src.rpm
Build Date  : Wed Jan 29 16:39:25 2020
Summary     : MinGW Windows cross-compiler (GCC) for C for the win64 target
```
At [Source Code (includes MinGW.org patches)](https://osdn.net/projects/mingw/releases/)  last build matches _Released at 2020-01-13 05:33_ for version _9.2.0_.!

pbatard, 2020-05-19:
Then please log an issue with either MinGW or your distro.

`PROCESSOR_ARCHITECTURE_ARM64` is explicitly defined in the `mingw32\i686-w64-mingw32\include\winnt.h` and `mingw64\x86_64-w64-mingw32\include\winnt.h` headers for the version of MinGW that is installed by msys2 on Windows, which, again, is the __only__ version I care about or support.

If your version of MinGW is missing updated headers (please remember that MinGW is not just the compiler, but the compiler + the headers), it's entirely your problem to solve, because, as long as the version both myself and [AppVeyor](https://ci.appveyor.com/project/pbatard/rufus/build/job/xippfhr8bm7jt39f) are using, I'm not going out of my way to sort compilation issues, sorry.

pbatard, 2020-05-19:
Also: https://sourceforge.net/p/mingw-w64/mingw-w64/ci/master/tree/mingw-w64-headers/include/winnt.h#l4730

Ricky-Tigg, 2020-05-20:
Some available components required to be installed **may not have been installed** yet. I shall report though since i cannot confirm that it is the case. Amount of available components:
```
$ dnf list available mingw64-* | wc -l
407
```
Components that were installed for the purpose of this task:
```
$ rpm -qa mingw64-* | sort
mingw64-binutils-2.32-7.fc32.x86_64
mingw64-cpp-9.2.1-6.fc32.x86_64
mingw64-crt-6.0.0-4.fc32.noarch
mingw64-filesystem-111-1.fc32.noarch
mingw64-gcc-9.2.1-6.fc32.x86_64
mingw64-gcc-c++-9.2.1-6.fc32.x86_64
mingw64-headers-6.0.0-3.fc32.noarch
mingw64-pkg-config-0.28-14.fc32.x86_64
mingw64-winpthreads-6.0.0-3.fc32.noarch
mingw64-zlib-1.2.11-2.fc32.noarch
```

pbatard, 2020-05-20:
`mingw64-headers-6.0.0-3.fc32.noarch` __is__ old. Current version on my system is `8.0.0`. That's your problem right there, because `PROCESSOR_ARCHITECTURE_ARM64` was only introduced after `6.0.0`.

Again, you need to file a request with the package maintainers for MinGW on your system that they update `mingw64-headers` to something that is not close to obsolete.

Ricky-Tigg, 2020-05-20:
A [report](https://bugzilla.redhat.com/show_bug.cgi?id=1838022) has been submitted accordingly thanks to you. I hardly could have noticed that cause by myself on my system since the **build dates** of _mingw64-headers_ and _mingw64-gcc_ are there identical, while that last was up-to-date. 
```
$ rpm -qi mingw64-headers | sed -n '10,11p'
Source RPM  : mingw-headers-6.0.0-3.fc32.src.rpm
Build Date  : Wed Jan 29 16:48:10 2020
```