pbatard/rufus #1544: Error: [0xC0030090] The directory is not a subdirectory of the root directory.

Mr-x2020, 2020-05-20:
Any solution for this error

Error: [0xC0030090] The directory is not a subdirectory of the root directory.
[image]

pbatard, 2020-05-20:
Any chance for a full log, __as requested by the check list__?

Mr-x2020, 2020-05-20:
Rufus x86 v3.10.1647 (Portable)
Windows version: Windows 10 32 bit (Build 15063.413)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
Found USB 2.0 device 'General UDisk USB Device' (048D:04D2)
1 device found
Disk type: Removable, Disk size: 545 GB, Sector size: 512 bytes
Cylinders: 66322, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x231F4DAE
Drive has an unknown Master Boot Record
Partition 1:
  Type: FAT32 (0x0b)
  Size: 508.1 GB (545520615424 bytes)
  Start Sector: 2048, Boot: No
Found USB 2.0 device 'General UDisk USB Device' (048D:04D2)
1 device found
Disk type: Removable, Disk size: 545 GB, Sector size: 512 bytes
Cylinders: 66322, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x231F4DAE
Drive has an unknown Master Boot Record
Partition 1:
  Type: FAT32 (0x0b)
  Size: 508.1 GB (545520615424 bytes)
  Start Sector: 2048, Boot: No
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.04/20191223 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'Kali Live'
  Size: 2.9 GB (Projected)
  Has a >64 chars filename
  Uses: Syslinux/Isolinux v6.04
  Uses: EFI
  Note: This ISO uses symbolic links, which will not be replicated due to file system limitations.
  Because of this, some features from this image may not work...
Using image: kali-linux-2020.2-live-amd64.iso (2.8 GB)
Will reuse 'ldlinux.sys' and 'ldlinux.bss' from './rufus_files/rufus_files/syslinux-6.04/20191223/' for Syslinux installation

Format operation started
Requesting disk access...
Will use 'H:' as volume mountpoint
Deleting partitions...
Deleting ALL partition(s) from disk '\\?\PhysicalDrive1':
● Partition 1 (offset: 1048576, size: 508.1 GB)
Opened \\.\PhysicalDrive1 for shared write access
Analyzing existing boot records...
Drive has an unknown Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 2176 sectors
Deleting partitions...
Partitioning (MBR)...
● Creating Main Data Partition (offset: 1048576, size: 407.1 GB)
● Creating Linux Persistence Partition (offset: 437073751552, size: 101.0 GB)
Closing existing volume...
Waiting for logical drive to reappear...
Using Debian-like method to enable persistence
Formatting (ext3)...
1654783 possible inodes out of 26476537 blocks (block size = 4096)
1323826 blocks (5.0%) reserved for the super user
Creating 808 inode sets: [1 marker = 10.1 set(s)]
++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
Failed to create ext3 'lost+found' dir: Corrupted entry
Re-mounted volume as H: after error

Found USB 2.0 device 'General UDisk USB Device' (048D:04D2)
1 device found
No volume information for drive 0x81
Disk type: Removable, Disk size: 545 GB, Sector size: 512 bytes
Cylinders: 66322, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x23250A8D
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 407.1 GB (437072702976 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2:
  Type: GNU/Linux (0x83)
  Size: 101.0 GB (108447897600 bytes)
  Start Sector: 853659671, Boot: No
Found USB 2.0 device 'General UDisk USB Device' (048D:04D2)
1 device found
No volume information for drive 0x81
Disk type: Removable, Disk size: 545 GB, Sector size: 512 bytes
Cylinders: 66322, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x23250A8D
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 407.1 GB (437072702976 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2:
  Type: GNU/Linux (0x83)
  Size: 101.0 GB (108447897600 bytes)
  Start Sector: 853659671, Boot: No
Closed Windows format prompt

Mr-x2020, 2020-05-20:
[image]

Mr-x2020, 2020-05-20:
[image]

Mr-x2020, 2020-05-20:
Rufus x86 v3.10.1647
Windows version: Windows 10 32 bit (Build 15063.413)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 2.0 device 'General UDisk USB Device' (048D:04D2)
1 device found
Disk type: Removable, Disk size: 545 GB, Sector size: 512 bytes
Cylinders: 66322, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x23250A8D
Drive has an unknown Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 508.1 GB (545520615424 bytes)
  Start Sector: 2048, Boot: No
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.04/20191223 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'Kali Live'
  Size: 2.9 GB (Projected)
  Has a >64 chars filename
  Uses: Syslinux/Isolinux v6.04
  Uses: EFI
  Note: This ISO uses symbolic links, which will not be replicated due to file system limitations.
  Because of this, some features from this image may not work...
Using image: kali-linux-2020.2-live-amd64.iso (2.8 GB)
Downloading https://rufus.ie/files/syslinux-6.04/20191223/ldlinux.sys
Unable to access file: 404
Extended version was not found, trying main version...
Downloading https://rufus.ie/files/syslinux-6.04/ldlinux.sys
File length: 66.5 KB
Successfully downloaded 'ldlinux.sys'
Download signature is valid ✓
Downloading https://rufus.ie/files/syslinux-6.04/20191223/ldlinux.bss
Unable to access file: 404
Extended version was not found, trying main version...
Downloading https://rufus.ie/files/syslinux-6.04/ldlinux.bss
File length: 512 bytes
Successfully downloaded 'ldlinux.bss'
Download signature is valid ✓

Format operation started
Requesting disk access...
Will use 'H:' as volume mountpoint
Deleting partitions...
Deleting ALL partition(s) from disk '\\?\PhysicalDrive1':
● Partition 1 (offset: 1048576, size: 508.1 GB)
Opened \\.\PhysicalDrive1 for shared write access
Analyzing existing boot records...
Drive has an unknown Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 2176 sectors
Deleting partitions...
Partitioning (MBR)...
● Creating Main Data Partition (offset: 1048576, size: 407.1 GB)
● Creating Linux Persistence Partition (offset: 437073751552, size: 101.0 GB)
Closing existing volume...
Waiting for logical drive to reappear...
Using Debian-like method to enable persistence
Formatting (ext3)...
1654783 possible inodes out of 26476537 blocks (block size = 4096)
1323826 blocks (5.0%) reserved for the super user
Creating 808 inode sets: [1 marker = 10.1 set(s)]
++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
Failed to create ext3 'lost+found' dir: Corrupted entry
Re-mounted volume as H: after error

Found USB 2.0 device 'General UDisk USB Device' (048D:04D2)
1 device found
No volume information for drive 0x81
Disk type: Removable, Disk size: 545 GB, Sector size: 512 bytes
Cylinders: 66322, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x232E3376
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 407.1 GB (437072702976 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2:
  Type: GNU/Linux (0x83)
  Size: 101.0 GB (108447897600 bytes)
  Start Sector: 853659671, Boot: No
Found USB 2.0 device 'General UDisk USB Device' (048D:04D2)
1 device found
No volume information for drive 0x81
Disk type: Removable, Disk size: 545 GB, Sector size: 512 bytes
Cylinders: 66322, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x232E3376
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 407.1 GB (437072702976 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2:
  Type: GNU/Linux (0x83)
  Size: 101.0 GB (108447897600 bytes)
  Start Sector: 853659671, Boot: No
Closed Windows format prompt

pbatard, 2020-05-20:
> `Failed to create ext3 'lost+found' dir: Corrupted entry`

Ah, so that's with the ext formatting process from the persistent partition. Now that gives me something to try to replicate the problem, as your original report was too short on details (hence the reason why you should always provide a log, regardless of whether you think it applies or not).

I can't seem to replicate the issue when creating a 101 GB persistent partition on a 128 GB drive...

I'll try to see what happens on a larger drive, but from what I am seeing, this is unlikely to be a common error.

I would also __strongly__ encourage you to run a bad blocks check against your drive (which you can see in the advanced options), because there's a high probability that this could be your issue, and this possibility needs to be eliminated anyway.

Below is the log I get when creating a persistent drive using Kali:
<details>
<summary>log</summary>

```
Rufus x86 v3.10.1647
Windows version: Windows 10 64-bit (Build 18363.836)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0809 (en-GB)
Will use default UI locale 0x0809
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found VHD device 'Microsoft Virtual Disk'
Device eliminated because it's a Windows Sandbox VHD
Found USB 3.0 device 'SanDisk Extreme USB Device' (0781:5580)
Using autorun.inf label for drive F: 'Run Debian GNU/Linux'
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1946, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x02CDC16C
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 14.9 GB (16012894208 bytes)
  Start Sector: 2048, Boot: Yes
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.04/20191223 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'Kali Live'
  Size: 2.9 GB (Projected)
  Has a >64 chars filename
  Uses: Syslinux/Isolinux v6.04
  Uses: EFI
  Note: This ISO uses symbolic links, which will not be replicated due to file system limitations.
  Because of this, some features from this image may not work...
Using image: kali-linux-2020.2-live-amd64.iso (2.8 GB)
Will reuse 'ldlinux.sys' and 'ldlinux.bss' from './rufus_files/rufus_files/syslinux-6.04/20191223/' for Syslinux installation
Search for conflicting processes was interrupted due to timeout

Format operation started
Requesting disk access...
No drive letter was assigned...
Will use 'F:' as volume mountpoint
Deleting partitions...
Deleting ALL partition(s) from disk '\\?\PhysicalDrive9':
● Partition 1 (offset: 1048576, size: 18.2 GB)
● Partition 2 (offset: 19595794432, size: 101.0 GB)
Opened \\.\PhysicalDrive9 for shared write access
No logical drive found (unpartitioned?)
Drive does not appear to be partitioned
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
[... 1118 lines cut ...]
Finalizing, please wait...
F:autorun.inf already exists - keeping it

Found VHD device 'Microsoft Virtual Disk'
Device eliminated because it's a Windows Sandbox VHD
Found USB 3.0 device 'SanDisk Extreme Pro USB Device' (0781:5588)
Using autorun.inf label for drive F: 'Run Debian GNU/Linux'
1 device found
Disk type: Removable, Disk size: 128 GB, Sector size: 512 bytes
Cylinders: 15567, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x02D51D26
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 18.2 GB (19594745856 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2:
  Type: GNU/Linux (0x83)
  Size: 101.0 GB (108447897600 bytes)
  Start Sector: 38273036, Boot: No
```
</details>

By the way, what software did you use to record and create your GIF animations? I've been looking for a good way to do that, but haven't found anything that satisfied me so far.

Mr-x2020, 2020-05-20:
I runned "check device for bad blocks" option but I still get same error
I am using "Gif Recorder" software to record and create GIF animations http://gifrecorder.com/
@pbatard

pbatard, 2020-05-22:
Thanks for the update, and the link to gifrecorder.

I tried some more with various drives, including drives that are larger than 500 GB as well as a 500 GB drive, but I never got your error.

I am therefore going to close this issue and wait to see if other people report something similar. If not, then it means it's probably something specific to your environment.

closed 2020-05-22 by pbatard

zooose18, 2021-04-07:
I have the same issue and can't seem to find a resolution. I can create a bootable USB, but I can't partition it. 

My log file is below.
[rufus.log](https://github.com/pbatard/rufus/files/6274882/rufus.log)

pbatard, 2021-04-07:
@zooose18, what is the total capacity of your drive? Is it more than 1 TB?

pbatard, 2021-04-07:
Nevermind, I can see from your log that it's 25 GB.

Infamous222, 2021-05-05:
I am having the same issue with a 2TB USB drive. Did you ever find a fix?