pbatard/rufus #1545: Rufus- Windows to Go - Speed/time taken.

fendertelemusik, 2020-05-20:
﻿

Checklist
---------
- [x ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ x] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ x] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [x ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Hi, Im sorry in advance if i have done any of this wrong or if i have broken any rules regarding posting, i joined to ask on here as i had seen a similar post to mine although with slight differences.

Was hoping someone might be able to tell me if there is a problem or what im experiencing is the norm whilst performing a Rufus - windows to go operation.

My PC is new and all the Drivers are up to date according to Driver booster and Windows update, They are USB 3.0 however my USB stick isnt so was expecting things to be slower due to this however.

I have left the operation running for over and hour and a half and when i returned it was sitting at 3 percent, the log was constantly moving and no signs of stalling just slow speed writing i would guess ?

Is this to be expected or is it possible to increase this speed a little as i was hoping to use it by the morning for working at home and at this rate it might be done for tea tomorrow.

I have since stopped and started a few times hoping to find a different result with other ports etc but no joy.

The log attached is after 30 or so minutes.

thanks.
[rufus.log](https://github.com/pbatard/rufus/files/4659657/rufus.log)

Log
---
```
<FULL LOG>
```

leumasme, 2020-05-20:
Oh damn, this is a miracle, you were just one hour faster than me.
I have the same issue. Rufus has been writing for 4 hours 30+ minutes and is at 65.1%.
It is connected to usb3.0 and i have used this stick for windows installs(non-togo) before.
It also seems to be getting slower as it progresses. While i was able to watch it rise by 0.1% every few seconds at the beginning, it now takes multiple minutes per 0.1%

leumasme, 2020-05-21:
Im gonna go insane. After 6+ hours and over 90%, it failed with a write error. log attached
[rufuslog.txt](https://github.com/pbatard/rufus/files/4660213/rufuslog.txt)

pbatard, 2020-05-22:
I see nothing out of the ordinary from the logs. The error @leumasme got can unfortunately happen because Windows is an ass at handling multiple partitions, and even worse at allowing access to Efi System Partitions (which is why we have no choice but to create it last).

I'm afraid that the slow speed comes from your environment, not Rufus.
Especially, as pointed in __[this very relevant FAQ entry](https://github.com/pbatard/rufus/wiki/FAQ#Why_is_the_Windows_To_Go_media_creation_process_so_slow), you must make sure that the __random__ write speed of your USB Flash Drive is fast enough, or else the whole Windows To Go creation process is going to be excruciatingly slow. This being said, if Rufus takes a long time creating your Windows To Go drive, it's likely to be as dreadful when you try to run Windows To Go from it.

If temporarily disabling the Antivirus, as recommended in the FAQ entry, doesn't help, then your only option is to get a better drive.

I will therefore close this issue, as there isn't much I can do in Rufus, even with the ESP creation error...

closed 2020-05-22 by pbatard

leumasme, 2020-05-22:
Why is the non-togo installation much faster?
Also, why does the progress in % get much slower as it gets further into applying the image?

pbatard, 2020-05-22:
2 questions that received a notification for, but curiously can't see an entry here:

> Why is the non-togo installation much faster?

This is explained in the FAQ entry. We're not copying files, but we are applying a WIM or ESD image, which are completely different things. Because of the way Microsoft designed WIM files, copying files is always much much faster than WIM application.

> Also, why does the progress in % get much slower as it gets further into applying the image?

Because the progress report from Microsoft is broken. We ask the Microsoft APIs to report the progress (because we don't handle WIM application ourselves) and therefore we rely on Microsoft for the progress information. We actually apply some skewing to try to make it more accurate, but there's only so much we can do.

leumasme, 2020-05-22:
Sorry, i had deleted that after noticing that my first question was already answered in the faq ^^
Well, i tried flashing again, and again got a write error after 6h19m.
Last time i checked a few minutes earlier, it was at 97% so it seems to fail at pretty much exactly at 100%.
As this now happened twice in a row, is there something i should change on my side to prevent that error?
(log attached again)
[rufus.log](https://github.com/pbatard/rufus/files/4670016/rufus.log)

pbatard, 2020-05-22:
> is there something i should change on my side to prevent that error?

Yes. Get a faster USB Flash Drive.

I think the problem is that Windows gets too much time to interfere with the creation of the EFI System Partition.

For the record, I have tested Windows To go creation on a drive with __decent__ random write speed (SanDisk Extreme PRO USB 3.1 - SDCZ880 128 GB, used with a USB 3.0 port), and saw no issues. The whole operation took about 14 minutes.

Here is my log:
```
Rufus x86 v3.10.1647
Windows version: Windows 10 64-bit (Build 18363.836)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0809 (en-GB)
Will use default UI locale 0x0809
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found VHD device 'Microsoft Virtual Disk'
Device eliminated because it's a Windows Sandbox VHD
Found USB 3.0 device 'SanDisk Extreme Pro USB Device' (0781:5588)
Using autorun.inf label for drive F: 'Run Debian GNU/Linux'
1 device found
Disk type: Removable, Disk size: 128 GB, Sector size: 512 bytes
Cylinders: 15567, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x02D51D26
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 18.2 GB (19594745856 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2:
  Type: GNU/Linux (0x83)
  Size: 101.0 GB (108447897600 bytes)
  Start Sector: 38273036, Boot: No
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'CCCOMA_X64FRE_EN-GB_DV9'
  Size: 5 GB (Projected)
  Has a >4GB file
  Uses: EFI
  Uses: Bootmgr (BIOS and UEFI)
  Uses: Install.wim (version 0.13.1)
Using image: Win10_1909_EnglishInternational_x64.iso (5.1 GB)
WIM extraction method(s) supported: 7-Zip, wimgapi.dll
WIM apply method supported: wimgapi.dll
Opening: \\.\CDROM1/sources/install.wim:[0] (API)
Extracting: C:\Users\nil\AppData\Local\Temp\Ruf1CA.tmp (From [1].xml)
Closing: \\.\CDROM1/sources/install.wim
Will use 'Windows 10 Pro' (Build: 18362, Index 6) for Windows To Go
Search for conflicting processes was interrupted due to timeout

Format operation started
Requesting disk access...
Will use 'F:' as volume mountpoint
Deleting partitions...
Deleting ALL partition(s) from disk '\\?\PhysicalDrive9':
● Partition 1 (offset: 1048576, size: 18.2 GB)
● Partition 2 (offset: 19595794432, size: 101.0 GB)
Opened \\.\PhysicalDrive9 for exclusive write access
No logical drive found (unpartitioned?)
Drive does not appear to be partitioned
Analyzing existing boot records...
Drive has a Syslinux Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
[... 121 lines cut ...]
Max parts: 128, Start Offset: 17408, Usable = 128043678208 bytes
Partition 1:
  Type: Microsoft System Reserved Partition
  Name: 'Microsoft Reserved Partition'
  ID: {CDF1BDB3-4B0A-4BF9-BF17-5EE602ABB705}
  Size: 128 MB (134217728 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  ID: {7C2A9BFE-494D-4BE4-8AC5-13113AD574E7}
  Size: 118.8 GB (127593833472 bytes)
  Start Sector: 264192, Attributes: 0x0000000000000000
Partition 3:
  Type: EFI System Partition
  Name: 'EFI System Partition'
  ID: {B5FA0248-B107-439C-BD0A-80E64E97B076}
  Size: 300 MB (314592768 bytes)
  Start Sector: 249470898, Attributes: 0x0000000000000000
Found VHD device 'Microsoft Virtual Disk'
Device eliminated because it's a Windows Sandbox VHD
1 device found
```

If your device is taking 6 hours to write, you are kidding yourself about being able to use it as a Windows To Go drive, because it will also be way to slow to run Windows from. Not being able to create a Windows To Go drive in less than 30 mins is usually a good sign that the drive you are trying to use is not suitable for Windows To Go.

waveletlet, 2020-07-05:
Does this poor write speed = poor run speed hold true for Windows To Go drives created from a Windows VM? On linux, passing through the USB device to the qemu VM. I left it to run over night and it's still going, only 80% so far.

pbatard, 2020-07-05:
> Does this poor write speed = poor run speed hold true for Windows To Go drives created from a Windows VM?

Most likely yes. For a properly designed VM environment, there's shouldn't be that much translation enacted for USB transfer, so slow in VM pretty much equates slow in real life.

Your Windows To Go drive is unlikely to achieve the speed required for a usable Windows experience.

waveletlet, 2020-07-06:
For anyone else with the "fail at 99%" problem, it's probably because Windows doesn't want to set up boot configuration data on your usb stick. Because of @pbatard 's earlier comment about it being finicky with the EFI partition, I decided to follow some "manually fix windows EFI partition" guides. Quite a few of the steps give the "we don't support that on removable media" error, but some of them I could get around using linux (like formatting the FAT32 partition). The step that is 100% windows only that I couldn't get past with my drive was running `bcdboot` at the very end, which seems to be the step that makes it bootable.

Actually I found a workaround for that, too, by passing the USB stick as a block device through QEMU (instead of as USB passthrough) so Windows thinks it's a normal hard drive, but I've waited at least an hour or so for it to boot and it's still "setting up devices" or something, so it's not looking like it'll be very usable. (All I need it for is running *one* program to try to diagnose a hardware issue on a laptop whose manufacturer doesn't provide any support or details for Linux users. If this doesn't work, the screwdrivers are coming out...)

I wonder if there's a way to catch that it's an unsupported drive and make it fail earlier before extracting all the wim. The steps it complained about it being a removable drive on were (in diskpart) formatting the EFI partition as fat32 and assigning a volume letter, and the bcdboot step mentioned earlier.