pbatard/rufus #1548: Old "usb to usable" issue

1TekPro, 2020-05-25:
﻿I have read most of the similar posts and my issue is different. After trying with multiple partition editors and 8 usb drives, none are detectable.
I am attempting to burn a ubuntu iso to usb and it has had the same result 8 times.
Burn completes, disk is undetectable.
The usb disks varied in size and brand.

pbatard, 2020-05-25:
Can you please provide a full log of what happens when you try the following:

https://github.com/pbatard/rufus/wiki/FAQ#Help_how_do_I_restore_my_USB_to_how_it_was_before_I_created_a_bootable_drive

You can access the Rufus log with <kbd>Ctrl</kbd>-<kbd>L</kbd>.

1TekPro, 2020-05-25:
Can I access the old logs somehow?
I tried again for the 9th time and it worked.

pbatard, 2020-05-25:
> Can I access the old logs somehow?

Nope. It's usually not very kind to leave unwanted files on someone's computer, so the log is not saved unless the user does it manually.

> I tried again for the 9th time and it worked.

I was asking you to try the produced against any of the existing 8 drives that "failed", not to try on a new drive. But considering that you seem to have gotten your hands on quite a large number of drives, are you sure you didn't just get a batch of fake drives (i.e. drives that are sold cheap because they have been modified to pretend to have something 8x or 16x the amount of flash that they actually have, i.e. the flash drive controller can be made to report that the drive is 32 GB when it's really 2 GB, and of course, as soon as you write over that limit, all kind of bad things happen).

Rufus can detect for fake drive as part of its bad blocks check (which you __REALLY__ should have run after the 2nd or 3rd failure).

So again, what does the Rufus log report when you try to recreate the drive in Rufus as non-bootable, as I asked above. And if that works, can you please try to run a bad block check?

1TekPro, 2020-05-25:
I'm a bit unclear what is being asked.
If I use one of the failed drives to try and return it, rufus does not see the drive at all. 
The drives were a collection of different sizes and brands (sandisk being one of the name brands) so I doubt "cheap" drives related to the issue.

pbatard, 2020-05-25:
Can you still try <kbd>Alt</kbd>-<kbd>.</kbd> in Rufus, with one of the faulty drives plugged in, and post the log you get (regardless of whether the drive is reported)?

Also, you may want to check that you don't have power supply or cabling issues (for instance USB 3.0 drives are very sensitive to extension cables), coz there's no way you're going to lose 8 drives unless you have a major hardware problem, even more so as, contrary to what many users believe, Rufus does not do anything special to create the drives, and, even if the code was full of bugs, repartitioning or formatting a drive is absolutely no more dangerous than writing a file to a drive.

1TekPro, 2020-05-25:
None of the drives appear in the drop down.
So, I am going to attempt it with another drive and if it fails, I will post the log.

pbatard, 2020-05-25:
> None of the drives appear in the drop down.

Yes, I understand that, but I __still__ want to see the log.

> So, I am going to attempt it with another drive and if it fails, I will post the log.

No need. I just want to see the log with one of the failed drives and <kbd>Alt</kbd>-<kbd>.</kbd>. __Even if Rufus doesn't list the device__. Can you please provide that?

1TekPro, 2020-05-25:
The START button is not an option

1TekPro, 2020-05-25:
Rufus x86 v3.10.1647
Windows version: Windows 10 64-bit (Build 18363.836)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
0 devices found
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.04/20191223 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'Ubuntu 20.04 LTS amd64'
  Size: 2.5 GB (Projected)
  Has a >64 chars filename
  Uses: Syslinux/Isolinux v6.04
  Uses: EFI
  Note: This ISO uses symbolic links, which will not be replicated due to file system limitations.
  Because of this, some features from this image may not work...
Using image: ubuntu-20.04-desktop-amd64.iso (2.5 GB)

pbatard, 2020-05-25:
Good. Now please try the same thing with <kbd>Alt</kbd>-<kbd>. You don't  need to select an ISO.

1TekPro, 2020-05-25:
Rufus x86 v3.10.1647
Windows version: Windows 10 64-bit (Build 18363.836)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
0 devices found
Processing Hub 1:
  Hub[1] = '\\?\usb#root_hub30#4&2188f0d&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[170]: USB\VID_046D&PID_C52B\5&3AC2BAEE&0&9
Processing Hub 2:
Processing Hub 3:
Processing Hub 4:
  Hub[2] = '\\?\usb#root_hub20#4&1c476887&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[026]: USB\VID_8087&PID_8001\5&27B27D6B&0&1
Processing Hub 5:
  Hub[3] = '\\?\usb#root_hub20#4&245fdec6&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[202]: USB\VID_8087&PID_8009\5&255CBB93&0&1
Processing 'SCSI' device:
  Hardware ID: 'SCSI\DiskNVMe____Samsung_SSD_970_EXM7'
Found non-USB non-removable device 'Samsung SSD 970 EVO Plus 500GB' => Eliminated
0 devices found

pbatard, 2020-05-25:
Okay, so this is definitely hardware issue with either your flash drive or your USB bus.

closed 2020-05-26 by pbatard

1TekPro, 2020-05-26:
Food for thought:
Universal USB Installer works without issue. I have made 5 successful drives. (5 for 5)
Rufus (1 for 10)

pbatard, 2020-05-26:
https://github.com/pbatard/rufus/wiki/FAQ#Help_Rufus_damaged_my_flash_drive

1TekPro, 2020-05-26:
Simple fact is other software does not "brick" usb drives, yet Rufus did. I don't care that they are garbage now, I only wish to help find a potential bug. Posting "no it didn't" is unprofessional. The drive is not broken... but it is definitely not readable.

pbatard, 2020-05-26:
I'm sorry, but if you brick 8 drives without running a single bad blocks check or getting a log of how exactly the drive got to fail while the application was writing to it, there's really nothing I can do.

Also, EVERY SINGLE TIME someone comes here with a _"Rufus bricked my drive"_, they never seem to want to report what brand and model drives they were using and where exactly they acquired them. Millions and millions of people do create flash drives using Rufus without the slightest issue, so, no, there doesn't exist a "major bug" that bricks drives and if you're not reporting what is special about your environment and the exact drives you used, then it doesn't look to me like you are taking the matter seriously (even more so I'm afraid, if you're finding yourself bricking 8 drives in a row, and act as if there's no need to check for bad blocks).

You may also want to know that there exist some drives (e.g. [PNY ones](https://github.com/pbatard/rufus/wiki/FAQ#Incompatible_Hardware)) where the internal firmware has bugs, so, instead of pointing the finger straight at Rufus without providing any information that could even remotely hell troubleshoot the issue, if issue there was, I would encourage you to do your research.

If you are positive that Rufus bricked your drives, then by all means, please RMA them and report that problem to the manufacturer, as I am sure they will be very interested in replicating the issue and demonstrating with irrefutable proof how Rufus manages to brick their hardware and why it should not be used. Otherwise, sorry, but there's literally nothing I can do to help. I am not withholding any "inside knowledge" about even remotely dangerous practices that Rufus might perform that could lead to bricking a drive, and I can guarantee you that any developer with some knowledge of the Windows Hardware Abstraction Layer, and how any USB drive access translates to telling Windows: _"Write data Y to block X"_, with absolutely know possibility of interacting with the underlying USB bus or flash drive firmware or internal reserved blocks, will, just like me, have no choice but to dismiss the possibility that an application like Rufus could damage drives.

But again, if people are adamant that this is really what happens, then you should have no issue convincing at least one flash drive manufacturer to investigate the matter and __prove me wrong__, since it's in their utmost interest not to waste their money on RMA.

1TekPro, 2020-05-26:
Most recent attempt:
32gb Cruiser drive (grey) purchased from Staples
[https://www.staples.com/staples-32-gb-usb-2-0-flash-drive-2-pack/product_2832750](url)

*Drive previously had an operating system on it*
Opened rufus and set the usual iso options then enabled bad block check

Log:
Attached file

I then closed rufus and immediately opened it again.

Log:
Rufus x86 v3.10.1647
Windows version: Windows 10 64-bit (Build 18363.836)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Processing Hub 1:
  Hub[1] = '\\?\usb#root_hub30#4&2188f0d&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[170]: USB\VID_046D&PID_C52B\5&3AC2BAEE&0&9
Processing Hub 2:
Processing Hub 3:
Processing Hub 4:
  Hub[2] = '\\?\usb#root_hub20#4&1c476887&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[026]: USB\VID_8087&PID_8001\5&27B27D6B&0&1
Processing Hub 5:
  Hub[3] = '\\?\usb#root_hub20#4&245fdec6&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[202]: USB\VID_8087&PID_8009\5&255CBB93&0&1
Processing 'SCSI' device:
  Hardware ID: 'SCSI\DiskNVMe____Samsung_SSD_970_EXM7'
Found non-USB non-removable device 'Samsung SSD 970 EVO Plus 500GB' => Eliminated
0 devices found
0 devices found
Checking for Rufus updates...
Checking release channel...
No new release version found.

1TekPro, 2020-05-26:
Interesting discovery:
The iso used was for the desktop version of Ubuntu. I tried using a different version (Ubuntu Server) because the 1 drive that worked said the installer was corrupt. Rufus successfully makes the server version without failure every time. Is it possible that the iso is at fault here?

pbatard, 2020-05-26:
The only things I can think of:
* Flaky internal USB hub
* Security solution that lets devices it likes through while removing others.

I'd be very curious to know if you could see the device in File Explorer and list its content after you created the drive, because, as far as Rufus is concerned, the drive was accessible by the time it exited, and we do use the Windows APIs to query the drive (which is illustrated by the fact that we used the `D:` drive letter to copy the files). So the drive should have been accessible as `D:` from the moment Rufus started to copy files over, and we certainly don't do anything like unmounting or even querying the drive on Rufus exit, so I can't fathom how on earth a drive that the Windows APIs listed both as a USB device and as a mounted partition would disappear.

The only thing Rufus does on exit is:
* Destroy the UI window and its child elements (none of this can even remotely affect USB devices unless there is a major bug in Windows)
* Clean up created resources like the localization file or the check for update thread (none of which relate to USB devices)
* Restore the autorun policy to how it was when the app was started. We temporarily disable autorun because otherwise users would get annoying prompts about "What do you want to do with this drive" each time they plug one, and we consider that, if they use Rufus, they probably plugged a drive to use it there. But even for that, we use the documented Microsoft procedure to go through Local Group Policies (rather than edit the registry manually which is bad practice), and (unless Windows has a major bug) disabling autorun should never ever prevent the device from being listed as a USB device. Autorun is just about prompting the user to run stuff from the drive, NOT removing the drive from the system altogether.

And that is really it. That's all that happens between the time Rufus happily reports that both the device and the `D:` drive are available and the time the application terminates. Especially, there is not a single part of the code I can think of that is still going to poke at the device before or during exit, unless you plug and unplug USB drives (because then we will get an event notification to re-enumerate devices).

So really, if all you did after the drive was created was close Rufus, then there has to be something external that is effectively removing the device (not even the mounted partitions, but the USB device as a whole), and the only thing I can think of would be a security solution that only whitelists devices it likes.

Especially, one thing I'd really like to know is whether you can see and access that device on a Linux platform, because, unless you have a flaky USB or PSU hardware system, there's just no way drive after drive could become bricked like this, especially as there really isn't any application accessing them.

One thing that I know Windows does automatically (i.e. that's not something Rufus can control), after a drive has been created with a partition it can access, is create a `System Volume Information` folder, so that's really the only access I can see, post Rufus creation. But again, since Rufus does no access the drive at all after creation, if the mere creation of `System Volume Information` is enough to make a drive go missing, there has to be something very weird about your system.

At any rate, again, if this was a major bug from Rufus, this issue tracker would be absolutely swamped from people reporting that they are losing drive after drive (Rufus gets downloaded more than 1 million times a month and runs only on Windows, so if there was such a widespread issue, it would be super obvious), which is clearly not the case. So there has to be an environmental component to your issue, and this may be very tricky to identify.

Unless you can recover them from a Linux system, I don't really want to ask you to try more drives, but I'd be really curious to see what happens if you try the same thing from a completely different Windows system (the closer to a vanilla Windows installation the better) and, as I mentioned above, if you can access the D: drive after Rufus is done (while leaving Rufus open) and if you can copy files to and from that drive.

pbatard, 2020-05-26:
Oh and you want to monitor what happens in Device Manager as well, as if Rufus can't list the device then Device Manager won't either, so the device has to disappear from Device Manager at some point...

1TekPro, 2020-05-26:
It only disappears when you manually scan for changes in device manager if it is open already.

pbatard, 2020-05-26:
> It only disappears when you manually scan for changes in device manager if it is open already.

Okay.

> Ubuntu recognizes all of the failed drives

Then we already have a difference here, because Windows does not recognize the hardware at all whereas Ubuntu does. Usually, querying of a USB device is pretty much the same in Windows and Linux, as it boils down to sending a few USB request for the same descriptors, but maybe Linux is a bit more forgiving when it gets unexpected descriptors.

But really, for a USB device to fail detection, it would need to have it's descriptors screwed up, and these only exist in a protected area of the drive that neither Windows nor any application can write to, let alone access __unless__ you happen to have a "fake drive", because then, when write operations start to go over the lower actual capacity of the drive, you will start writing into flash cells that you were never supposed to be able to write to, and in those are likely to be the ones holding the descriptor data.

In fact, what you describe is pretty much the exact process in which I expect a "fake drive" to fail:
1. Application writes data that goes over the actual flash size (e.g. actual flash size is 2 GB but a 4 GB image is being written)
2. The "fake" firmware of the drive, which is purposefully designed not to check for limits regarding the actual flash capacity (so that users don't discover that it's a fake drive) ends up writing internal firmware areas.
3. As long as the computer doesn't query the descriptors (which really only happens when you replug or force a hardware scan), everything looks good, because the cached USB descriptors that Windows retrieved are being used.
4. After user re-plugs the drive or re-scans peripherals, the USB descriptors are being queries and return garbage, since they have now been overwritten.
5. Because the descriptors could not be retrieved, the device being declared as missing.

And once again, I have to be absolutely explicit that, for a properly designed drive, there is just no way for the OS or any application, to mistakenly access the internal firmware area instead of the general data area. As you may expect, if the firmware reserved area was left open to access, and it was let to the application to ensure that they don't overwrite it, you'd get bricked device after bricked device from buggy applications, so, of course, it's the internal flash controller's job to only ever let the USB host (i.e. the computer) access the general purpose area and hide it's internal area.

Now, that's not to say that there doesn't exist __proprietary__ USB commands that manufacturers can add to write to that area, but it also needs to be pointed out that tese are:
* Completely different from standard USB commands 
* Completely different from one manufacturer to the next (because they are internal commands)
* Impossible for an application like Rufus __that doesn't actually build its own USB commands but uses the Windows Hardware Abstraction Layer__ (i.e. the layer that makes Windows present all disks a mere block devices, regardless of whether they are SATA, NVMe, SCSI, USB, Thunderbolt, etc.) to send on the USB bus, even by mistake, unless there is a __MAJOR__ bug in the Operating System itself.

So the explanation about using a fake drive would fit exactly what you are observing. But of course it is really puzzling that it would happen to drive after drive from different vendors, and also it is puzzling that the bad blocks check from Rufus, which is designed to check for fake drives, wouldn't have detected it (though it is possible that, due to its popularity, fake drive vendors have started to produce firmwares that work around our detection, but I somehow doubt that).

Oh and that might also explain why one ISO will work and the other fail, as, depending on the size of the ISO, you may have one that goes over the effective size of the fake drive and another that doesn't.

In other words, what you see if exactly what I'd expect to happen if you were using a bunch of fake drives, and it therefore super strange that all the tests to validate that this is indeed a fake drive issue (run a fake drive check, use different brands of drive) seem to be negative... However, I am still 100% positive that, if you are using standard hardware, Rufus on its own does not have the means to brick drives like this, so there must be some hardware component that we are still missing.

1TekPro, 2020-05-26:
Thank you for your detailed response. From everything that you have said, and what I have researched, it would appear like they were fake drives. But no tests point to that conclusion. After running a flash size testing program, the size check comes back valid to the claimed size.
I tested the size myself also, by writing 7,416mb of data to an 8gb usb drive and then read the last bits of data to make sure there wasn't any "trimming" used to make it look like it all fit (8gb on a 4gb drive for example).

pbatard, 2020-05-26:
Yes, as I said, if your results are to be believe, there is something super weird going on. The next best thing to do then is to see if you observe the same thing on a completely different computer, with as close to a vanilla installation of Windows as you can (i.e. no custom security solutions or customization policies), because, as I pointed out, we're definitely not seeing the kind of reports one would expect if there was a bug in Rufus that was at the root of your issue (else, user after user would be reporting the same results as yours, which they aren't, and it's exceedingly difficult to troubleshoot something that, from what I can estimate, and what this very issue tracker would show otherwise, is not even happening for one in one million Rufus users).

There has to be something in your environment that triggers this behaviour.