pbatard/rufus #1549: Installer is partially operational when size persistence is equivalent to full

Ricky-Tigg, 2020-05-28:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.
- [x] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] Using a different USB drive.
- [x] Plugging the USB into a different port.
- [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the SHA-256 checksum, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Last official ISO image for Bliss-OS (x86_64) was written successfullly for each two cases tested. In each case the _size persistence_ was the only  parameter whose value changed, either **null** or equivalent to **full**. 

Each case was conducted on BIOS and EFI systems.
- Case 1 (here as comparison mean) - persistence: **null**
Installer is fully operational.
- Case 2 - persistence: **full**
Installer is partially operational. While installation has reached the partition section, no key respond to invocations, thus installation cannot be operated further.

Log
---
[rufus_x86..v3.10_bliss-os_x86_64.v11.11.log](https://github.com/pbatard/rufus/files/4693854/rufus_x86.v3.10_bliss-os_x86_64.v11.11.log)

pbatard, 2020-05-28:
1. You removed data from the log. All the data about the ISO you opened is missing, so I don't know what was the actual ISO you used. Which means I'm not going to bother trying to replicate your issue. Do __NOT__ remove data from the log.
2. `Write error [0x000003EE] Tiedoston levyä on muutettu ulkoisesti, niin että avattu tiedosto ei ole enää kelvollinen.` indicates that something __EXTERNAL__ to Rufus is modifying files after they are being written. You will need to find what that is, because that's the most likely cause of your problem.

closed 2020-05-28 by pbatard

Ricky-Tigg, 2020-05-28:
What a provocative asssumption. You had to decrete all at once in order to choose that an external factor has to be a physical person and thus me.  That kind of personn then you are. There are ways to speak and i won't guess how many ennemies you have already made with that deliberate provocative language. Nevertheless i reformatted the disk, that time without the irrelevant checking of bad sectors; Logs -[rufus.log](https://github.com/pbatard/rufus/files/4696671/rufus.log)- are not identical to the previous ones. You do understand what it means. A program whose logs may vary randomly unexpectedly will make soon or later vanished users' trust in its presumed reliability.

pbatard, 2020-05-28:
> What a provocative assumption.

I am not assuming. I am just reporting what I'm seeing.
Your log was both missing data that was removed for one reason or another, and displaying data that indicates an underlying problem.

It is you who are assuming that the problems being reported by the original log come from Rufus and are a statement of the program being unreliable, where it is just a report of external issues being observed.

So you are free to accept facts, or reject them, but that that doesn't make them any less valid. When Rufus repeatedly gets an error from Windows that states `[0x000003EE] The volume for a file has been externally altered so that the opened file is no longer valid` when trying to write a file, it's pretty safe to conclude that there is something on your system or your hardware that is causing the issue and that the problem doesn't originate from Rufus, especially when the couple of other instances where I have ever seen `[0x000003EE]` being reported had to do with something completely external to Rufus (such as a failing flash drive).

If you are unwilling to accept that logs reporting a certain kind of error are unlikely to indicate a software bug with the application but rather an external problem that it is up to you to properly investigate, and since you also seem to be unhappy about the level of support you are getting, then by all means please use a different application.

Ricky-Tigg, 2020-05-29:
I though your speak was just for me. A confirmation that you apply the same method with all is just there at your Git project report channel you then use to mess with users bringing you legitime reports. So its not just with me. Good to know. You must take pleasure from all that. Anyway, since the _case 2_ had evolved  as last log showed, it resulted in new observation while booting through BIOS:
```
Syslinux 6.03 EDD 2014-10-06 (...)
No DEFAULT or UI configuration directive found!
boot:
```

pbatard, 2020-05-29:
> bringing you legitime reports. 

You mean reports where __essential__ data has been forcibly removed, like the exact name of the ISO being used, or where major errors are being dismissed as something non important?

The check list is very explicit about what I require to provide support to users. If a user decides that some of the elements I asked them to do (such as not removing any data from the log or checking elements that they see as an annoyance without effectively performing the requested action) are not important enough to be followed, then, because they have demonstrated right off the bat that they are unable or unwilling to follow guidelines, I will 

Let me let you in on a little secret here: The propensity for a user to follow the check list is my "brown M&M's" test". If a user can't follow it, it tells me that they are not actually serious about getting help and are more likely to become an annoyance than anything else. Therefore, I will immediately treat them as such, because I genuinely don't have time to try to second guess data that should have been provided or acted upon, and that the user chose to withhold or ignore.

From the replies I am seeing, it looks like trying to get involved with you as little as possible is indeed the right choice, as you appear to be unwilling to take any direction and your second log shows even more write errors, pointing to what is likely to be a defective flash drive rather than anything else. So I will continue to do that. And since you are happily ignoring the many, many issues where I have been helping folks who __do__ properly follow the check list or the advice I give them, and instead try to imply that I _mess with all users bringing reports_, then, because I genuinely don't have time for this, you have also earned yourself a ban.