pbatard/rufus #1552: Few problems with computer after trying to flash iso to usb

CL0SH4REX, 2020-05-31:
First of all, i tried to flash kali linux iso on my usb. When it got about 30% i got error and rufus made usb write-protected. Secondly, my pc won't shut down/restart. When i try to do it it just stays on shutting down/restarting screen and i have to force shut it down with buttons on pc. Third and last of all, my usb wifi adapter can't be recognized by pc. I use cudy wu700. It worked perfectly, but as i tried to flash iso on my usb, my network stopped working. Adapter doesn't work anymore, i get message that it's not recognized and etc. When i reinstall drivers for adapter, it works but when i try to use internet it just blocks it, but every other usb device works fine. I tested same adapter on other pc and it works well. What can i do, worst problem is for wifi adapter.

JonnyTech, 2020-05-31:
Sounds like hardware failure - not a Rufus issue

pbatard, 2020-05-31:
> When it got about 30% i got error and rufus made usb write-protected. 

https://github.com/pbatard/rufus/wiki/FAQ#Rufus_cant_detectformat_my_drive_and_reports_that_it_is_readonly_or_contains_no_media

>  Secondly, my pc won't shut down/restart.

How's that a Rufus issue? Sorry but this is not a general tech support forum, especially if you consider that the check list should not apply to you.

> Third and last of all, my usb wifi adapter can't be recognized by pc.

Same.

Please ask for help in a dedicated forum, because this is not the proper place.

closed 2020-05-31 by pbatard