pbatard/rufus #1558: The problem here (and this is one I am struggling with a lot as I am introducing ext2/ext3 formatting) is that Windows, and especially Windows 10, appears to be aggressively caching the partition data from drives, and besides ejecting the media, there doesn't seem to exist an efficient way to make it refresh its view of the partition layout after it's been modified. The partition caching is problematic because as long as Windows thinks there is a partition occupying this or that range of sectors, it may prevent access to said sectors being written to even by its own APIs.

Cro-Magnon28, 2020-06-10:
The problem here (and this is one I am struggling with a lot as I am introducing ext2/ext3 formatting) is that Windows, and especially Windows 10, appears to be aggressively caching the partition data from drives, and besides ejecting the media, there doesn't seem to exist an efficient way to make it refresh its view of the partition layout after it's been modified. The partition caching is problematic because as long as Windows thinks there is a partition occupying this or that range of sectors, it may prevent access to said sectors being written to even by its own APIs.

None of the things I've tried to implement so far (calling on VDS refresh API calls, simulating a disable + enable of the disk as you would do in Device Manager) have yielded much success. For one thing, VDS seems to be completely hopeless as clearing the partition data cache, even as it is supposed do so with calls such as [`IVdsService_Refresh()`](https://docs.microsoft.com/en-us/windows/desktop/api/vds/nf-vds-ivdsservice-refresh) or [`IVdsService_Reenumerate()`](https://docs.microsoft.com/en-us/windows/desktop/api/vds/nf-vds-ivdsservice-reenumerate), and thanks to Windows' weirdness, the device disable+enable trick may appear to work fine, but can leave the device disabled the next time you replug it...

So what you want, really, is Windows to acknowledge that the partition layout it is trying to use is obsolete and should be discarded. IMO, the most efficient way to do that is to use the 2 following cheat modes in Rufus:

1. First, you should issue <kbd>Alt</kbd>-<kbd>Z</kbd>. This writes zeroes to the whole drive which should ensure that the partition table gets zapped (though one should know that, if a drive is GPT, there does exist a backup partition table near the end of the drive that could theoretically be used. However, from my experience, Windows does tend to ignore it if it finds there is no primary GPT table.
2. Let the zeroing run for a little while and then cancel it.
3. Issue <kbd>Alt</kbd>-<kbd>C</kbd>, which will cycle the USB port for the currently selected device. Of course, this only works for USB devices, and what's more, only on Windows 8 or later, as Microsoft decided to remove USB port reset, which they introduced in Windows XP, from Windows 7, only to reintroduce it back in 8... Or if you can't cycle the port just eject and reinsert the device.

If you do the steps above, then you will usually find that whatever formatting issue you were having goes away.

It should be pointed out that, obviously, Rufus does zero the partition tables (and a few more sectors just to be safe) before it repartitions and reformats a drive, but as I pointed out, it does appear that Windows was never really designed for on-the-fly changes to partition data, so, even though we apply this and pretty much all the tricks in the book we can think of to tell Windows that it should refresh its vision of the drive layout, the only consistent way we found to make it do that is to make it actively disconnect and reconnect the device (which we can't really automate, even on Windows 8 or later, as a USB port cycle can take up to 30 seconds, and, as I mentioned, disable + enable of a device can lead to that device remaining in a "zombie state").

So if you are getting errors, try the zeroing + cycle trick described above, as it will probably help.

_Originally posted by @pbatard in https://github.com/pbatard/rufus/issues/1319#issuecomment-496481875_

Cro-Magnon28, 2020-06-10:
Sir?
How exactly this step to be applied?

Cro-Magnon28, 2020-06-10:
Sir @pbatard can you help me?

Rufus x86 v3.10.1647
Windows version: Windows 7 SP1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 2.0 device 'USB2.0 Flash Disk USB Device' (1221:3234)
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 2039, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00193820
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 15.6 GB (16776167424 bytes)
  Start Sector: 2048, Boot: Yes
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: '02_17_2019'
  Size: 10.6 GB (Projected)
  Has a >4GB file
  Uses: EFI
  Uses: Bootmgr (BIOS and UEFI)
  Uses: Install.esd (version 0.14.0+)
Using image: Windows AllinOne.iso (10.6 GB)
Checking for Rufus updates...
Network is unavailable: Internet has not be initialized.
Updates: Unable to connect to the internet

Format operation started
Requesting disk access...
Will use 'E:' as volume mountpoint
Deleting partitions...
Deleting ALL partition(s) from disk '\\?\PhysicalDrive1':
● Partition 1 (offset: 1048576, size: 15.6 GB)
Opened \\.\PhysicalDrive1 for exclusive write access
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Bad Blocks: Checking from block 0 to 127999 (1 block = 128 KB)
Bad Blocks: Using offset 58702 for fake device check
Bad Blocks: Writing test pattern 0x00
Bad Blocks: Reading and comparing
Bad Blocks: 8115
Bad Blocks: 8116
Bad Blocks: 8117
Bad Blocks: 8118
Bad Blocks: 8119
Bad Blocks: 8120
Bad Blocks: 8121
Bad Blocks: 8122
Bad Blocks: 8123
Bad Blocks: 8124
Bad Blocks: 8125
Bad Blocks: 8126
Bad Blocks: 8127
Bad Blocks: 8128
Bad Blocks: 8129
Bad Blocks: 8130
Bad Blocks: 8131
[... 271 lines cut ...]

Found USB 2.0 device 'USB2.0 Flash Disk USB Device' (1221:3234)
1 device found
No volume information for drive 0x81
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 2039, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00893ED6
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: NTFS/exFAT/UDF (0x07)
  Size: 15.6 GB (16776167424 bytes)
  Start Sector: 2048, Boot: Yes
Found USB 2.0 device 'USB2.0 Flash Disk USB Device' (1221:3234)
1 device found
No volume information for drive 0x81
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 2039, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00893ED6
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: NTFS/exFAT/UDF (0x07)
  Size: 15.6 GB (16776167424 bytes)
  Start Sector: 2048, Boot: Yes

Cro-Magnon28, 2020-06-10:
Rufus x86 v3.10.1647
Windows version: Windows 7 SP1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
Found USB 2.0 device 'USB2.0 Flash Disk USB Device' (1221:3234)
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 2039, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00C06D66
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 15.6 GB (16776167424 bytes)
  Start Sector: 2048, Boot: Yes
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: '02_17_2019'
  Size: 10.6 GB (Projected)
  Has a >4GB file
  Uses: EFI
  Uses: Bootmgr (BIOS and UEFI)
  Uses: Install.esd (version 0.14.0+)
Using image: Windows AllinOne.iso (10.6 GB)

Format operation started
Requesting disk access...
Will use 'E:' as volume mountpoint
Deleting partitions...
Deleting ALL partition(s) from disk '\\?\PhysicalDrive1':
● Partition 1 (offset: 1048576, size: 15.6 GB)
Opened \\.\PhysicalDrive1 for exclusive write access
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (MBR)...
● Creating Main Data Partition (offset: 1048576, size: 15.6 GB)
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (NTFS)...
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Error while formatting
[0x0000007E] The specified module could not be found. - Retrying...
[0x0000007E] The specified module could not be found. - Retrying...
[0x0000007E] The specified module could not be found. - Retrying...
[0x0000007E] The specified module could not be found. - Retrying...
Format error: Undetermined error while formatting.
Re-mounted volume as E: after error

Found USB 2.0 device 'USB2.0 Flash Disk USB Device' (1221:3234)
1 device found
No volume information for drive 0x81
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 2039, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00D5A8B1
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: NTFS/exFAT/UDF (0x07)
  Size: 15.6 GB (16776167424 bytes)
  Start Sector: 2048, Boot: Yes
Found USB 2.0 device 'USB2.0 Flash Disk USB Device' (1221:3234)
1 device found
No volume information for drive 0x81
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 2039, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00D5A8B1
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: NTFS/exFAT/UDF (0x07)
  Size: 15.6 GB (16776167424 bytes)
  Start Sector: 2048, Boot: Yes

pbatard, 2020-06-10:
Your device has bad blocks, which means its flash memory is failing. It's a pure hardware issue. You need to use a different flash drive because this one is dead.

Also next time, please follow the check list.

closed 2020-06-10 by pbatard