pbatard/rufus #1559: Hey Man, URGENT problem please answer me

DoodMann, 2020-06-11:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>
Hey creator nice app but, I know you already said this to others but. IT KILLED MY DRIVE! Apparently, I was flashing a Windows 10 iso file when we had a blackout here. When the electricity came back, Windows can't find my USB. I already tried disk management, diskpart, even a temporary linux version but no luck. Please can you help me?
Log
---
```
<FULL LOG>
```

pbatard, 2020-06-11:
Sorry but if your computer can't find your drive then it's the blackout that killed it not Rufus.

Rufus (and Windows) just sent regular write commands to the drive. If a USB drive can't handle a power shutdown during a regular write, then that's a problem with the firmware of that USB drive and not with the software that sent the write commands (which, as I explained __over and over again__ are absolutely nothing special and are not "special" USB commands that Rufus builds on its own).

If the drive can't be detected any longer, then you should RMA that drive and let the manufacturer know that a power failure during a standard write operation may kill the drive, so that they may try to improve their firmware.

closed 2020-06-11 by pbatard

blackcrack, 2020-06-12:
@pbatard 

"Hey Man, URGENT problem please answer me"   ... \*lol*

DoodMann, 2020-06-14:
Hey man, thanks for the answer. Damn, Kingston doesn't make these kind of
failsafes. Oh, and what is RMA?

pbatard, 2020-06-14:
https://en.wikipedia.org/wiki/Return_merchandise_authorization

DoodMann, 2020-06-16:
oh hey man, it's me again. Can you make a ubuntu/debian version of Rufus?
That's a big help!

pbatard, 2020-06-16:
https://github.com/pbatard/rufus/wiki/FAQ#Do_you_plan_to_port_Rufus_to_LinuxMac_OSSome_other_OS

blackcrack, 2020-06-17:
hey @DoodMann ,

=> :heart: 
*thumps up* :+1: , you be really nice !  (this i call, "Elite" in "Whitehat scene")
i wish you good compiling of Computers and tinkering in Software !

best regards
Blacky

blackysgate.de
Hug the Penguin ! ;)