pbatard/rufus #1562: exFAT is Now Included in Linux Kernel - Time to Reconsider exFAT Support with UEFI/GPT?

Musafir86, 2020-06-13:
Hello & good day sir,

-Firstly, thank you very much for dedicating your precious time developing this excellent piece of Free Software.

-Now, I hope that I'm not beating a dead horse here, but situation has changed significantly since the last discussions about exFAT support in Rufus, so I think it's already time to add exFAT support when choosing GPT & UEFI (non-CSM).

-In case you (unlikely) missed the news, here I provide the relevant links:
"exFAT in the Linux kernel? Yes!"
https://cloudblogs.microsoft.com/opensource/2019/08/28/exfat-linux-kernel/

"Microsoft Publishes exFAT Specification, Encourages Linux Support"
https://www.phoronix.com/scan.php?page=news_item&px=Microsoft-exFAT-Specification

"Samsung Sends In Improvements To The exFAT File-System Driver For Linux 5.8"
https://www.phoronix.com/scan.php?page=news_item&px=exFAT-Improvements-Linux-5.8

More articles/news here: https://www.phoronix.com/scan.php?page=search&q=exFAT

Thanks & regards,
-Musafir86.

pbatard, 2020-06-13:
https://github.com/pbatard/uefi-ntfs/commit/a30338b7a9803ae5612e17f9b8ecb2733c526d07

pbatard, 2020-06-13:
Also https://github.com/pbatard/rufus/blob/master/ChangeLog.txt#L24

closed 2020-06-13 by pbatard

Musafir86, 2020-06-13:
Thanks for the (very) quick reply!

Yes, I did read the changelog and searched for exFAT-relevant details in it before posting the issue. However, when I tried the latest Rufus 3.10.1647, selected the Windows 10 1909 x64 (en-US) ISO image - which has install.wim that's over 4GB - Rufus defaulted to NTFS with no option to change to exFAT (only FAT32).

So, is there any specific parameter need to be passed to enable exFAT support?

pbatard, 2020-06-13:
Now, to elaborate a little bit on that:
1. Just like I don't enable FAT32 for Windows MBR drives (even when the ISO doesn't have a file that is smaller than  GB), when NTFS does __just fine__, I don't see much point of formally enabling exFAT for Windows GPT drives where their too NTFS does __just fine__ (and please spare me the _"But exFAT doesn't wear out the drive as much as NTFS"_ which is a complete bullshit argument when creating an installation drive).
2. However, for people who really want to use exFAT for Windows GPT boot, I offer the option to create a blank UEFI:NTFS drive with exFAT as the file system, so that they can just copy the ISO content if they want.

Unless someone can demonstrate, with actual factual evidence (rather than hear and say) that exFAT offers major advantages over NTFS when it comes to running Windows (no, interoperability with Linux or Mac systems does not count, because, again, it's not something that matters for a Windows installation drive).

In other words, the only push I see for exFAT usage comes from pure corner case that few people are expected to run into, which, as per my usual burb from [here](https://github.com/pbatard/rufus/wiki/FAQ#Why_cant_you_just_add_the_feature_I_want) makes me see __not__ having to support exFAT (which means not having to test exFAT support for breakage and regression, or having to answer irrelevant questions like _"But should I use exFAT or NTFS?"_) as the better option that adding support for it during Windows drive creation.

I __already__ made a concession for people who are adamant about using exFAT through providing the ability of creating UEFI:NTFS exFAT based drives. So please don't make me regret that decision by insisting that I add formal exFAT support for Windows, when there really is no clear advantage to be gained in doing so (for the record, Windows 7 does __NOT__ support exFAT boot at all, so I'd either have to detect Win7 or answer queries about why Windows 7 doesn't boot, which is less than ideal).

Hope that answers your question.

Musafir86, 2020-06-13:
Okay, it's really clear to me now (I didn't understood that "create blank exFAT" choice before).

Thanks again for your time.

Regards.