pbatard/rufus #1567: RUFUS MAY have messed up the access permissions for two of my thumb drives

ZMG0021, 2020-06-20:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log. (Showed another conversation/issue that had turned sour and was old.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [x] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I am not including the Log as this is merely a statement at the moment but will if required

I used the RUFUS application to attempt to write and image, The first time I attempted it it split my drive into multiple partitions which I assumed was a minor glitch, Used RUFUS to format the drive and it made the drive in accessible from either of my laptops or my desktop showing the windows "You need permission to view this file" error.

Checked using Windows Disk Management software, saw no errors in the formatting and ran a full Reset, I have attempted replicating the error with two different drives other than the first one. The error only occurred on one other drive that was fresh out of the box. I am still learning some of the stuff that I need to do so please do not be rude. I have been able to restore full functionality but was wanting to see if this was an error on my part or an error on the programs part

Log
---
```
Rufus x86 v3.10.1647
Windows version: Windows 10 64-bit (Build 18362.900)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
Found USB 2.0 device 'SanDisk Cruzer Glide USB Device' (0781:5575)
Could not open GUID volume '\\?\Volume{1dc242a9-84b5-1342-9ca4-cec37a89c21f}': [0x00000005] Access is denied.
No logical drive found (unpartitioned?)
1 device found
Could not open GUID volume '\\?\Volume{1dc242a9-84b5-1342-9ca4-cec37a89c21f}': [0x00000005] Access is denied.
No volume information for drive 0x81
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1869, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x2B414B0B
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 14.3 GB (15374221312 bytes)
  Start Sector: 2048, Boot: No

```

pbatard, 2020-06-20:
> The first time I attempted it it split my drive into multiple partitions which I assumed was a minor glitch

Not a glitch. If you write an image, you write what the creators of that image decided to produce, including multiple partitions, which are __very common__ to handle UEFI boot (usually you will have a small FAT32 EFI System Partition and a larger partition, that may not be accessible at all from Windows, with the actual image content).

> Used RUFUS to format the drive and it made the drive in accessible from either of my laptops or my desktop showing the windows "You need permission to view this file" error.

Never had any other report of `You need permission to view this file` issue, be it by e-mail, on user forums or on this issue tracker. This would tend to point at an issue with your environment, not Rufus, as I hope you can agree that an error that only happens one in a million times (if not smaller) is unlikely to be a software error.

> Checked using Windows Disk Management software, saw no errors in the formatting and ran a full Reset,

So what exactly did you do to "Reset" the drive if Rufus didn't work?

> so please do not be rude. 

Please don't conflate stating facts that may not go in the direction that someone wants with being rude. A lot of people who happen to lose their drive due to external reasons seem to come to this issue tracker with the expectation that I, as a developer, will say something like _"Oh yeah, this is totally Rufus' fault - here's a refund to buy a new drive"_ and seem to take it very very personally when I explain, at length, that it's next to impossible for an application like Rufus to actually kill a drive, even if it wanted to or was riddled with bug. If that counts as "being rude", so be it, but at one stage one has to differentiate looking at facts that don't go the way you expected with someone being malicious with you. And I'm pretty sure some people too will see that explanation as "rude", where, there again, I am just trying to state facts.

>  I have attempted replicating the error with two different drives other than the first one. The error only occurred on one other drive that was fresh out of the box. 

Okay, then please provide the log of how you created the drive that messed the access permission.

You do understand that you provided way too few details for me to provide much of any insight about your issue, such as not revealing the image you used at all (which is __critical__ information) or the brand and model of your drives.

Call me "rude" by there too, I really can't fathom why people who open an issue to report about Rufus potentially "damaging" their drive always seem to choose to withhold crucial data like the above. This is quite frustrating, because you really shouldn't go around pointing the finger at an application without also providing as many details as you can to try to back up you claim. If you decide to withhold evidence, such as simply stating that you wrote "an image", without indicating what it was, or not providing a full log of you created the failed drive, please don't be surprised to find the developer seemingly being dismissive of your claims, because you don't appear to want to back them up with information that is critical to analysing the validity of your claims.

ZMG0021, 2020-06-20:
I am currently working on another comment to this, But if you read the LOG it even specifies the drive type. I will have the new comment up soon with the appropriate log referenced

ZMG0021, 2020-06-20:
For your information I was attempting to create a bootable ISO for Chromium OS, I have since created it using other software and guess what. It didn't have to split the drive into multiple partitions. I am more than willing to make a screen recording detailing the entire process that I used to make the image bootable using RUFUS.

Now I see why the last person who called you out on the crap got angry with you, I was hoping to actually have a question answered, I even stated in the thing I was not blaming I merely thought it could have been one of **MANY** issues, And I accessed the logs that were available to me.

And if my claims of RUFUS **POSSIBLY** screwing with the permissions are one in a million how do you explain the countless others that have reported issues where their drives wouldn't work anymore. 

I had no issue with beleiving that the ISO needed multiple partitions but when it won't boot and only shows as having 16 MB when plugged in I start to think there is an issue. Now the log i am uploading will have a time that says around 7:00 AM as I am unable to locate where logs are saved as i typically do not need to access them.

I used the windows disk management utility to try and fix it, **THANKFULLY** that was able to fix the errors, I even attempted using RUFUS again to see if it was a one time glitch. I am more than willing to screen record and show the issue happening again. 

Now if you will stop being so pretentious because you refuse to believe your software may have errors please feel free to read the log I am submitting below. And for reference I have been
[image]
[image]
[Rufus log.txt](https://github.com/pbatard/rufus/files/4807847/Rufus.log.txt)
[image]
[image]

I was able to prevent the error on this  attempt, Please review all logs attached and all images/screenshots that show the errors after creating the bootable drive. I was referring to the reset error that revoked my permissions. I have managed to avoid that error this time.

In the future try not to automatically assume that someone is blaming your software. I was not I simply state it **MAY** have caused the errors.

pbatard, 2020-06-20:
Sorry, but your log only provides data about one of your drives, not two, and as I pointed out, you did not indicate the image you used at all.

Also, did you read [this entry](https://github.com/pbatard/rufus/wiki/FAQ#Error_0x00000005_Access_is_denied) from the FAQ? One possibility is that whatever image you were writing wrote a file that security software could consider as malicious and it may be the reason why it prevents most applications from accessing the drive.

I have to stress out that `[0x00000005] Access is denied.` is an error that is completely external to Rufus (even if it started to happen after you wrote the drive with Rufus), so there's not much that can be done to prevent it.

ZMG0021, 2020-06-20:
Well as i did not realize I would be posting the issue here i did not save logs and only saved the log right before I posted

pbatard, 2020-06-20:
> a bootable ISO for Chromium OS

Aha! there you go.

Windows has an issue with Chromium images. This is detailed in #753 

The problem is that Windows doesn't like GPT drives with tons of partitions, and Chromium images are built like that.

A search for Chromium in this issue tracker would have helped you, hence the precise reason why __NOT__ disclosing the image you used was a bad idea.

ZMG0021, 2020-06-20:
In that case I do apologize for thinking RUFUS had an issue, however it still shouldn't have caused my drive to revoke my permissions.

And when will RUFUS be available for linux

pbatard, 2020-06-20:
Can you please provide the download URL for the image you used? I'm sorry but your report is still lacking important data, such as the precise image you used.

Also, your second question is answered [in the FAQ](https://github.com/pbatard/rufus/wiki/FAQ#Do_you_plan_to_port_Rufus_to_LinuxMac_OSSome_other_OS).

ZMG0021, 2020-06-20:
Allow me just a moment to retrieve the link for you

ZMG0021, 2020-06-20:
https://chromium.arnoldthebat.co.uk/special//Cx86OS_R58-9334.B-Vanilla.7z

That is the direct download link to the image that I used on the drive

pbatard, 2020-06-20:
> it still shouldn't have caused my drive to revoke my permissions.

Again, that is a Windows bug, so you should report that issue to Microsoft. If Windows has a bug when it refreshes partition after Rufus wrote a drive, I hope you can agree that there's not much Rufus can do. I tried to add some mitigation for that issue, but, basically, Windows doesn't like drives with lots of GPT partitions, and, as shown by your log, Chromium OS images are like this.

ZMG0021, 2020-06-20:
https://chromium.arnoldthebat.co.uk/special/
The one highlighted in the image is the one i downloaded 
[image]

ZMG0021, 2020-06-20:
I will be contacting MS about the issue as I **HATE** dealing with crap like that especially when its something I needed to test quickly as I had someone wanting to pay money if I would be able to install it on their current computers.

pbatard, 2020-06-20:
Thanks, I'll try to take a look when I get a chance, but again, the analysis I conducted in #759 was pretty explicit that Windows has a bug when you use certain APIs to try to repartition/reformat a drive that contains many (>10?) GPT partitions, which actually makes the drive looks like it completely failed (even though you can easily repartition/reformat that drive on Linux).

I'll see if there's anything I can add to Rufus to mitigate this issue further, but I'm not overtly optimistic. And please please please, next time you log a report, do provide the image you used. If I had seen Chrome/Chromium OS right from the start, we would have avoided a lot of pointless back and forth.

ZMG0021, 2020-06-20:
I Most certainly agree. Again I am still learning how to do a lot of stuff like this as I am currently doing a side hustle of setting computers up for people

ZMG0021, 2020-06-20:
Looks like it is now time for Linux to go on my backup computer as I am currently using Windows only

pbatard, 2020-07-23:
Okay, I thought I'd manage to get a chance to investigate the mitigation Rufus currently applies for Chromium images further, but I'm afraid I just don't have the time for that, and since it's really a Windows bug, that we already tried to mitigate, our current mitigation will have to do. &rarr; Closing.

closed 2020-07-23 by pbatard