pbatard/rufus #1568: Version numbering

Hawkwala, 2020-06-23:
﻿

Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>

Log
---
```
<FULL LOG>
```

pbatard, 2020-06-23:
?

closed 2020-06-23 by pbatard

Hawkwala, 2020-06-23:
May I suggest a change in the version numbering system. I was using Rufus Version 3.9 and when I searched the website for the latest version it showed that 3.11 is the latest version. Version 3.9 should actually be version 3.09 to maintain the proper sequence. With the current numbering system, 3.9 is higher than 3.11. If the current version is labeled as 3.09 then 3.11 is higher in the sequence.

pbatard, 2020-06-23:
Sorry but that's how most software operates. It's not a mathematical fraction, and you never know if a software may end up with a 3.100 version (in which case you may want to request that I number 3.1 as 3.001 rather than 3.01, which would be super confusing).

There are some __established conventions__ you just have to learn to live with. This is one of them.