pbatard/rufus #1570: No persistance for latest Calculate Linux

enigma131, 2020-06-28:
I've made a USB Bootable Pen with latest Rufus and latest Calculate Linux:
http://miroir.linuxtricks.fr/calculate/release/20.6/cldc-20.6-x86_64.iso
Persistance is proposed, I made it with , Pen is starting and working but without persistance.
How to debug the problem?

pbatard, 2020-06-28:
> How to debug the problem?

Persistence currently only works with Ubuntu and Debian Live. For persistence to work, the distro maintainers need to properly support it in their boot script, and few distros currently do (but Rufus does propose the option just in case, because it's impossible to tell if a distro does support persistence or not).

Also, distributions that are derivatives of Ubuntu need to make sure that they have applied the bugfix for https://bugs.launchpad.net/ubuntu/+source/casper/+bug/1489855 which was only applied recently in Ubuntu, else persistence will not work.

In other words, please feel free to contact your specific Linux distro maintainers and ask them to follow either what Ubuntu does or what Debian Live does to enable persistence, because at this stage, I am [not planning to add support in Rufus for any other methods](https://github.com/pbatard/rufus/blob/master/src/iso.c#L313-L339).

closed 2020-06-28 by pbatard

enigma131, 2020-06-28:
OK that's more clear now
Thanks