pbatard/rufus #1573: Not offered BIOS for MBR Kaspersky Rescue Disk

B00ze64, 2020-07-04:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
When burning Kaspersky Rescue Disk 2018 (downloaded today) with Rufus 3.04 I am offered the option of BIOS or UEFI, but with Rufus 3.11, the only option offered is UEFI. The KRD ISO is of course compatible with BIOS/CSM. The FAQ (from 2015) mentions Kaspersky are bad boys, and break GRUB identification, and it appears they are still doing it, but I opened the issue because Rufus 3.04 still offered a choice of BIOS or UEFI; Rufus 3.11 does not anymore.

Log
---
```
Rufus x86 v3.11.1678 (Portable)
Windows version: Windows 7 SP1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
0 devices found
Scanning image...
ISO analysis:
  Image is an ISO9660 image
Could not get ISO-9660 file information for file boot/grub/i386-pc/normal.mod
  Could not read Grub version from 'boot/grub/i386-pc/normal.mod'
  Could not detect Grub version
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'KRD'
  Size: 587.8 MB (Projected)
  Uses: EFI
Using image: Kaspersky Rescue Disk 2018-2020-07.iso (591.5 MB)
```

pbatard, 2020-07-04:
Thanks for the report. This looks like a regression indeed. I'll look into it when I get a chance.

EugeneJarecki, 2020-07-21:
I'd like to piggyback on this. Rufus 3.11 seems to be the latest version, and now the target system HAS to be UEFI, not BIOS. I've been happily booting from USB sticks with iso's burned via Rufus for months on an ancient HP Compaq with an AMD 64 Athlon X2. (Some of us are poor, yo!) 

Yesterday it didn't work. I urge the developer to put back the BIOS functionality so those of us who can't afford new UEFI systems can be spared the mess that is UNetbootin.

pbatard, 2020-07-21:
As indicated above, I am planning to fix this in version 3.12.

EugeneJarecki, 2020-07-22:
Thanks much! Rufus rocks as hard as the old band of the same name.
Shout-out to Chaka Khan.

commit a8f3c6c5 (2020-09-11): [iso] fix detection of GRUB version

* Commit 77d319267f869d0d9eb9ac5a831164add8a5aa8d broke lookup of ISO filenames since iso9660_open()
  enabled the Rock Ridge extensions by default, despite using ISO_EXTENSION_NONE
  for the internal call, and we addressed a FIXME related to this.
* This resulted in Rufus not being able to lookup 'boot/grub/i386-pc/normal.mod' to parse GRUB's
  version, since without Rock Ridge, 'i386-pc/' is unable to match the ISO-9660 'I386_PC/' dir.
* Closes #1573 and addresses part of #1616.
* Also fix a MinGW compilation warning.
closed 2020-09-11 by pbatard