pbatard/rufus #1576: [error report] Rufus always fails on if persistence size too small

midaspt, 2020-07-08:
﻿

Checklist
---------
- [X] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [X] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Issue description
-----------------

First of all, I am very sorry to be unable to provide any logs, but I was using a **Portable Rufus** (both v3.8 and v3.11) and didn't realize then I needed to keep them to post here. Anyway this is a simple error report, feel free to dismiss at your discretion -- although I think **Rufus** users could benefit by being made aware of the issue.

Following solutions posted to <https://askubuntu.com/questions/882979/> on a cheap Staples 4GB drive, I repeatedly tried to set my persistence to 8, 16, and 24MB, but always failed because Rufus encountered an internal error while creating the ext3 structures, which, although they didn't prevent **Rufus** from continuing with the process, would always trigger the Windows drive format prompt and that would then cause **Rufus** to throw a write error and stop.

It took me sometime to realize what the problem was. I was then finally able to complete the procedure by setting the persistence size to 256MB, so I don't really know what the minimum size must be, but believe it would be valuable to make that information public.

Thanks for your great work on Rufus, I really appreciate it.

pbatard, 2020-07-08:
> I repeatedly tried to set my persistence to 8, 16, and 24MB

That's very small. A lot of file systems do have a minimum file size, so I guess I'll need to make sure I prevent Rufus users from entering a size that is too small.

I'll see if I can do that for 3.12. Thanks for the report.

midaspt, 2020-07-10:
Thanks for your consideration, **pbatard**.

One last remark on something that might not be entirely clear: the reason I was trying to create such a small _persistence_ in this particular use case is because it's meant to be deleted right after creation, so I didn't want to waste too much drive space (in other words, what I wanted was to create a bootable drive with _persistence_ but with no _persistence space_ allocated on the drive itself). I'm perfectly aware this may just be too marginal a case to deserve any attention.

Thanks again.

commit e6ac3782 (2020-10-02): [ext2fs] set a minimum ext partition size of 256 MB

* Not gonna bother with computing the effective minimum size so 256 MB will have to do.
* Closes #1576
closed 2020-10-02 by pbatard