pbatard/rufus #1578: MBR-BIOS-single NTFS flash drive boots UEFI

tockschmock, 2020-07-11:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [x] Running Rufus on a different computer.
- [x] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Please apologize for mistakes. I am not very experienced with forums and github.

Thanks for Rufus and especially for the clear explanations in the documentation.
I learned there, that it is an important goal for Rufus to avoid confusion about the mode, Rufus created flash drives finally boot (Legacy or UEFI).
The more I was suprised to see, that a MBR-BIOS-NTFS stick created from a W-10-2004 ISO (see log) boots (also) UEFI on 2 of my 3 PCs. At least I interpret the boots as UEFI, since the boot list entries on both PCs show UEFI: , the blue tile "UEFI Firmware Settings" (works) shows up and the W-10 installations start with the UEFI type partitioning of a clean disk (GPT, ESP, MSR, Boot). 
Another indication: running an detectefix64.exe (here on Github) shows also UEFI, but I dropped this from my "list of reliable evidences", since it shows UEFI also on an 2007 PC, that definitely has no UEFI capabilities. Other suggestions to find out, whether an WinPE runs in UEFI mode or not, did not work (reg query HKLM\System\CurrentControlSet\Control /v PEFirmwareType ???) for me.

I assume, that both PC's UEFI Firmware can read NTFS and therfore boot the stick from the single NTFS partition that (for reasons I do not understand) contains a functional efi directory for an UEFI mode boot.

The legacy mode also works, if I chose the legacy boot entry (without the UEFI: ...).
detectefix64.exe shows (here) correctly Legacy, the blue tile UEFI "Firmware Settings" is gone and the installation does the legacy partitioning (MBR, one small system partition first).

Therefore it seems clear, that contrary to many information, Windows(PE) can boot in an MBR partitioning scheme and from a single partition. Most information state in contrary, that Windows enforces GPT for UEFI and requires a separate (FAT32) ESP.

I am very interested in achieving a single partition UEFI-mode Windows installations, like the WinPE one on the stick, 
because I prefer simple partition backups for recovery purposes and don't want to deal with hidden (ESP), recovery (not necessary since can be integrated in the system partition or booted from a stick) or obscure unformatted partitons "for later use" (MSR).
However I could nopt repeat the scheme (single partition UEFI-mode) for a W-10 installation.
I managed to install W-10 in a single NTFS partition on MBR (by pre-partitioning the disk).
However my (amatuer) trials to "inject" the efi features with bcdboot ... did not work.

Could you please comment, if my stick really boots UEFI and whether it is technically (im)possible to replicate this into an UEFI  Windows installation on a single partition?
I believe I saw hints that this can be done (registry changes, bcdboot, ???) and the UEFI WindowsPE NTFS partition Rufus created, supports this.

Less but also interesting:
Is there a clean way to determine, which mode a WinPE is running?
Do you have an idea, what can go wrong with the detectefix64.exe?

Thank You

[log.txt](https://github.com/pbatard/rufus/files/4907553/log.txt)

pbatard, 2020-07-11:
> I was suprised to see, that a MBR-BIOS-NTFS stick created from a W-10-2004 ISO (see log) boots (also) UEFI on 2 of my 3 PCs.

Well, first of all MBR does not equate BIOS. Be very minful that UEFI supports __both__ MBR and GPT partition schemes (it's in the specs that it must support both), so using MBR is not at all a guarantee that only BIOS/Legacy can apply.

I think the confusion comes from the fact that, on the other hand, BIOS supports booting from MBR only and so, because GPT happens to have been introduced with UEFI, people wrongly assume that it must mean that UEFI supports GPT only, which is not at all the case.

This being said, yes, more and more systems support NTFS as a UEFI file system, which is the one major roadblock to booting Windows from UEFI, so yes, if you have a MBR + NTFS drive containing files extracted from a Windows ISO (which is what Rufus will produce), and if your UEFI firmware natively supports NTFS, then even if you create a drive that is targeted at BIOS systems, it's going to be bootable as NTFS.

I did consider forcefully removing the UEFI bootloaders when creating a BIOS bootable drive, so that only BIOS can boot, but decided against it, as I don't think it's a good idea to have an application that converts an ISO to USB to semi-randomly remove files from the original media.

The end result of that is that, in some circumstances, a media that was created to be BIOS bootable will happen to also be UEFI bootable.

Also, you may want to double check whether you enabled dual BIOS+UEFI boot (<kbd>Alt</kbd>-<kbd>E</kbd>), which is a __persistent__ setting (meaning that it gets preserved between Rufus sessions) that one can apply to create a Windows USB media that can boot in either BIOS or UEFI mode.

> I assume, that both PC's UEFI Firmware can read NTFS and therfore boot the stick from the single NTFS partition that (for reasons I do not understand) contains a functional efi directory for an UEFI mode boot.

Either that or you enabled dual BIOS+UEFI boot, in which case Rufus will have added a small 512 KB partition to enable NTFS boot from UEFI (which you may not see at all in Windows because Disk Manager for instance does not display partitions that small).

> Therefore it seems clear, that contrary to many information, Windows(PE) can boot in an MBR partitioning scheme and from a single partition.

Yes, that is well know. Windows does not really need a separate ESP to boot, and, as far as I know, it's mostly being recommended in case users want to use bitlocker to encrypt their system partition (because then an ESP __is__ necessary).

> Most information state in contrary, that Windows enforces GPT for UEFI and requires a separate (FAT32) ESP.

Once again, you have to be mindful that people on the internet will interpret something that is commonly done (such as using GPT rather than MBR for booting UEFI or creating both an ESP and a system partition) into something that is mandatory, when this is absolutely not the case. It's the same people who are going to be adamant to tell you that UEFI can only boot from FAT32, when this is certainly not the case either. So always be critical of the information you get (including this one, because there might be elements that I also could either be misinterpreting or reporting wrong).

> I am very interested in achieving a single partition UEFI-mode Windows installations, like the WinPE one on the stick,

Then I'm afraid you'll have to ask that question to Microsoft or Microsoft user forums.

I know that for BIOS installations, you could easily tell Windows to use a single partition by simply setting up the partition scheme yourself during the setup, and who knows if Microsoft might have an installer that's able to detect if the UEFI firmware supports NTFS and let you use a single partition then. But if not, then I'm afraid this is a pure Microsoft issue, on which I have no insight at this stage.

> Could you please comment, if my stick really boots UEFI

That's not something I can do. You're the only one who can find out. Usually, if the boot option you selected was prefixed by `UEFI:` and it didn't fail, then it has to be "real" UEFI boot. But there's no way I can give you the assurance you want about a system that was designed by someone else, and that I no knowledge about. The only way to tell is to have whatever you boot report whether it sees its environment as BIOS or UEFI, which Windows most certainly can do once it's been installed.

Or you can try to replace the UEFI bootloader you expect to be executed if it's real UEFI boot with something like a UEFI shell, and see if you actually get to the shell, coz that's a pretty sure way to know.

> and whether it is technically (im)possible to replicate this into an UEFI Windows installation on a single partition?

No idea. That's a Microsoft question. Only Microsoft knows for sure about what Windows and/or the Windows bootloaders can or cannot achieve...

> Is there a clean way to determine, which mode a WinPE is running?

Sorry but I have no idea.

> Do you have an idea, what can go wrong with the detectefix64.exe?

I had never heard about `detectefix64.exe` before you mentioned it, and as much as I'd like to help, there's only so much time I can invest playing with third party tools, unless a lot of users start to report that they are using such tools in conjunction with Rufus. So I'm going to have to leave that investigation to you.

[... cut ...]  of elements that I want to be or am already in the process of investigating for Rufus. So, as much as I wish it could be otherwise, I don't actually have the luxury to drop whatever I'm already working on, to take on another item, even if it might seem tangentially relevant.

On the other hand, I am always interested to hear from people who took these kind of investigations into their own hands, and can provide a digest of their findings... :wink:

I hope that more or less answers your question

tockschmock, 2020-07-12:
> (Alt-E), which is a *persistent* setting (meaning that it gets preserved

... that was quick!
Thank you from a Leipzig, Germany breakfast Cafe.
I will send some more details later.
Ralf

tockschmock, 2020-07-12:
Thanks agin for your quick response.

Yes, I am aware that MBR dos not mean BIOS - I just wanted to list the parameters for the stick creation quickly.
I was also taking care that the dual mode (Alt-E) is off.

So the result remains: UEFI:NTFS capable PCs (2 out of 3 in my home) can boot the stick. intended to be Legacy, also UEFI-style.
I read somewhere, that UEFI:NTFS is nowadays more the rule than the exception (all Surface Books can do this, if I remember the citation correctly).
Therefore it would be helpful to show the type of boot for a stick after booting from it (autorun?).

I used the blue tile UEFI "Firmware Settings" as "evidence" for an UEFI boot and its absence as evidence for a legacy boot.
I am not sure however, wether this is reliable - it is also not exactly elegant.

My research to detect the type of boot for WinPE resulted in:
reg query HKLM\System\CurrentControlSet\Control /v PEFirmwareType ???   did not work for me
and this (old) link:
https://social.technet.microsoft.com/Forums/ie/en-US/6cbb488d-3062-4aad-b712-7a9e4d045b13/how-to-use-wmi-to-query-to-distinguish-legacy-and-uefi-mode-windows?forum=winserverManagement
Somebody apperently wrote a few lines of code evaluting a "GetFirmwareEnvironmentVariable".
https://github.com/xcat2/xcat-core/blob/master/xCAT-server/share/xcat/netboot/windows/detectefi.cpp
The detectUefi.exe apparently did not work correctly for x86/x64 (thats what I understand from the discussion ) and therefore it was improved to an:
DetectEFIx86.exe and 
DetectEFIx64.exe
available via the dropbox links.

I used this files and they are correct in most cases.
However since the DetectEFIx86.exe also failed on an old PC, that definetly had no UEFI (detecting UEFI), I lost faith in this EXEs.
Probably you better understand what was done here and might be able to use it for displaying the actual boot mode of a stick.

As far as my real goal is concerned (achieving a single partition UEFI Windows-10), there is surprisingly little information available. Moving and rebuilding the ESP seems easy and is explained on various websites. Applying this steps to a move the ESP into C:\ and then deleting it does not work for me however...

There was a hot debate here for Linux (IMPOSSIBLE !!!)
https://askubuntu.com/questions/827491/is-separate-efi-boot-partition-required

So apparently I am the first guy in the world who (accidentially) demonstrated a single patition UEFI boot (though only into WinPE  ;-)  I should apply with the Guinness World Records.

I found this article interesting, but could not read it yet in detail.

Please let me know, if you have any ideas or links whether a single partition UEFI Windows-10 installation is possible (or impossible).
So far I did not find a clear explanation that (let alone why) it should be impossible.

Thanks

pbatard, 2020-07-12:
> Therefore it would be helpful to show the type of boot for a stick after booting from it (autorun?).

Then you'll have to ask Microsoft to do that.

Rufus does __NOT__ alter the bootloaders, because its mandate is to create a bootable drive that is as close a possible as the original. Especially, if you take Secure Boot into account, you really don't want to do that.

So if you are expecting that Rufus should modify the bootloaders to indicate the boot mode, I can tell that I'd start receiving a lot of complaints for doing something that is unnecessary and potentially dangerous, as it can be seen as compromising existing bootloaders.

Oh and autorun has no relation to that at all.

If you want bootloaders to perform a specific thing, then you need to ask the people who create said bootloaders, and I am not them.

> Probably you better understand what was done here and might be able to use it for displaying the actual boot mode of a stick.

As I said I am not interested in looking at what somebody else's utility is doing to solve a question that you are the only person to ask. I just don't have the time for that. And again, I am not going to __inject__ a foreign utility into the Windows installation files. That is just very bad practice.

> So apparently I am the first guy in the world who (accidentially) demonstrated a single patition UEFI boot

Sorry to burst your bubble but you're not. As I tried to bring to your attention earlier, upvotes != factual, and you should be weary of information you find on the internet. That answer from askubuntu is 100% wrong, even for booting ext4. For instance, one can easily create a UEFI firmware with ext4 file system drivers, in which case you are able to boot Linux from a single ext4 partition... And as I told you, if you do have a UEFI firmware with NTFS support, you can create a single NTFS partition UEFI bootable drive.

> Please let me know, if you have any ideas or links whether a single partition UEFI Windows-10 installation is possible (or impossible).

I'm afraid I'm not going to do that work for you. It's up to you to find out if that is possible. You may want to see if forums such as [reboot.pro](http://reboot.pro/) offer any insight on that, which, IMO, is the right place to ask your questions and get them answered rather than an issue tracker that is specific to Rufus issue, and not WinPE boot.

tockschmock, 2020-07-13:
> Rufus does **NOT** alter the bootloaders, because its mandate is to create a bootable drive that is as close a possible as the original.
>
I did not mean altering the original, but adding a feature to the stick, allowing to verify the mode, the stick booted up. Similar to my attempts to find a way/exe to verify this in WinPE before running the Windows installation.
Otherwise, what you want to avoid here: https://github.com/pbatard/rufus/wiki/FAQ#Why_doesnt_Rufus_create_a_Windows_installation_USB_that_can_be_booted_in_dual_BIOSUEFI_mode 
(installing Windows in the wrong mode), can just happen on many modern (UEFI:NTFS cabable) PCs, since an Legacy meant stick could boot UEFI.
> you should be weary of information you find on the internet. That answer from askubuntu is 100% wrong
>
I am not an expert, but these guys seemed to know what they are talking about.
Check Rod Smith's page and link.
> You may want to see if forums such as [reboot.pro](http://reboot.pro/) offer any insight on that (single-partition W-10 UEFI installation)
>
Thanks for the link - I will continue my search there (or rather give up on UEFI).
I also withdrew my application for a Guinness World Record ;-(

pbatard, 2020-07-13:
> but adding a feature to the stick

Adding files is the same as altering the original content. I want to alter (by modifying, removing or adding files) as little as possible. The only elements Rufus ever adds are:
- USB compatible versions of Syslinux or GRUB (because the files you find on an ISO cannot be used for USB boot)
- `autorun.inf` and incon, to make a newly created drive easier to differentiate from other ones and more importantly, to make Windows display the actual name that the user entered for the drive, even if it contains Unicode characters.

That's it. And that's how it should be.
You may also want to read [this](https://github.com/pbatard/rufus/wiki/FAQ#Why_cant_you_just_add_the_feature_I_want), because what you are requesting is for a usage scenario that you believe a lot of people will want, but that I see __very__ differently.

>these guys seemed to know what they are talking about. Check Rod Smith's page and link.

I know _Rod Smith_. He knows what he is talking about. The problem is that the folks on the thread you pointed to completely __misinterpreted__ what Rod was alluding to (or rather failed to understand that _Rod_ was talking about how UEFI can pick up boot files from a non ESP partition, and not about whether ESP is required or not), and ran with the idea that he was substantiating _oldfred_'s claim, which he isn't. I mean, for crying out loud, you found out yourself that an ESP is not a requirement. So why point to a thread that says ESP is an absolute requirement for UEFI boot, when you have seen actual evidence that it isn't?

And this is exactly the point I am trying to make. Do assume that upvotes equate facts. Or assume __anything__ as to the reality of information you find on the internet, for that matter, unless you have validated it yourself.

At any rate, I don't think there's more that I can provide here, so I will close this issue.

closed 2020-07-13 by pbatard