pbatard/rufus #1582: Undetermined error while formatting

baderalghamdi, 2020-07-22:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I screwed up my partitions and added some NTFS space by a partition manager free program and I want to reinstall windows because Reset this PC didn't fix anything and now I'm encountering this problem.

Log
---
```
Rufus x86 v3.11.1678
Windows version: Windows 10 64-bit (Build 19041.388)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 2.0 device 'Generic Flash-Disk USB Device' (1908:1320)
1 device found
Disk type: Removable, Disk size: 8 GB, Sector size: 512 bytes
Cylinders: 979, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x01640BF3
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 7.5 GB (8052015104 bytes)
  Start Sector: 2048, Boot: Yes
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'CCCOMA_X64FRE_EN-GB_DV9'
  Size: 4.9 GB (Projected)
  Has a >4GB file
  Uses: EFI
  Uses: Bootmgr (BIOS and UEFI)
  Uses: Install.wim (version 0.13.1)
Using image: Win10_2004_EnglishInternational_x64.iso (4.9 GB)

Format operation started
Requesting disk access...
Will use 'F:' as volume mountpoint
Deleting partitions (This may take a while)...
Deleting ALL partition(s) from disk '\\?\PhysicalDrive1':
● Partition 2 (offset: 1048576, size: 7.5 GB)
Opened \\.\PhysicalDrive1 for exclusive write access
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Partitioning (MBR)...
● Creating Main Data Partition (offset: 1048576, size: 7.5 GB)
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (NTFS)...
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Error while formatting
[0x00000032] The request is not supported. - Retrying...
[0x00000032] The request is not supported. - Retrying...
[0x00000032] The request is not supported. - Retrying...
[0x00000032] The request is not supported. - Retrying...
Format error: Undetermined error while formatting.
Re-mounted volume as F: after error

Found USB 2.0 device 'Generic Flash-Disk USB Device' (1908:1320)
1 device found
No volume information for drive 0x81
Disk type: Removable, Disk size: 8 GB, Sector size: 512 bytes
Cylinders: 979, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0164C252
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: NTFS/exFAT/UDF (0x07)
  Size: 7.5 GB (8052015104 bytes)
  Start Sector: 2048, Boot: Yes
Found USB 2.0 device 'Generic Flash-Disk USB Device' (1908:1320)
1 device found
No volume information for drive 0x81
Disk type: Removable, Disk size: 8 GB, Sector size: 512 bytes
Cylinders: 979, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0164C252
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: NTFS/exFAT/UDF (0x07)
  Size: 7.5 GB (8052015104 bytes)
  Start Sector: 2048, Boot: Yes

```

pbatard, 2020-07-22:
Please try to zero the drive (<kbd>Alt</kbd>-</kbd>Z</kbd>) or at the very least run a bad blocks check and retry again.

Also please follow as much of check-list items as you can (such as plugging your USB in a different port, or trying from a different computer).

You may also want to see what happens when you try formatting the drive in Windows, as the error you get comes from the Windows APIs, so my expectation is that you will also see an error if you try to format your drive to NTFS in Windows.

baderalghamdi, 2020-07-25:
> 
> 
> Please try to zero the drive (Alt-Z) or at the very least run a bad blocks check and retry again.
> 
> Also please follow as much of check-list items as you can (such as plugging your USB in a different port, or trying from a different computer).
> 
> You may also want to see what happens when you try formatting the drive in Windows, as the error you get comes from the Windows APIs, so my expectation is that you will also see an error if you try to format your drive to NTFS in Windows.

I tried formatting my drive in Windows using NTFS and it didn't work but using FAT32 and it worked, I also checked for bad blocks and found 313 corruption error(s) so how can I fix this? buying a new drive or formatting the PC another way? I can't switch ports in my USB drive because when I try to plug it it won't go in so I can only plug it in the left USB port.

pbatard, 2020-07-25:
> I also checked for bad blocks and found 313 corruption error(s)

There's your problem. It has nothing to do with Rufus. Your flash drive is defective.

> so how can I fix this?

Purchase a new one. Attempts to salvage this one is a waste of time, because the problem, really, is that the flash memory cells used by your drive have started to become defective, and this is not something that can be reverted.

Next time, please follow the check list in its entirety, as it did ask you to perform a bad blocks check before reporting your issue, and your issue is the result of your drive having bad blocks.

Since there's nothing Rufus can do, I will close this issue.

closed 2020-07-25 by pbatard