pbatard/rufus #1585: "A required device isn't connected or cannot be accessed" error

coolrecep, 2020-07-25:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
USB, created with Rufus fails to install Windows 10 2004. Getting error: A required device isn't connected or cannot be accessed.

[image]

Windows USB/DVD Download Tool solved the issue. This is the first time a user had a problem with Rufus. I'll try to get the log soon.

Log
---
```
<FULL LOG>
```

pbatard, 2020-07-25:
I'll be waiting for the log. Please also make sure that you compute the SHA-1 of your ISO by clicking the `(✓)` button after you selected it, as the first thing to check is that your ISO is not corrupted and that its SHA-1 matches the one from an official Microsoft retail ISO.

coolrecep, 2020-07-25:
I did ask for the SHA1 and it was correct for the Turkish ISO:

[image]

pbatard, 2020-07-25:
Please avoid screenshots. A log works much better, and I'm still waiting to see the one of how you created your drive.

For what is worth, nobody else is reporting issues with Windows 10 2004 install (and I have performed it myself quite a few times), though it's always possible Microsoft botched the Turkish ISO (Yes, even Microsoft managed to improperly master retail ISOs in the past), so chances are that it's your target system or your flash drive that are at fault. On errors like this, the first things you want to do is:
1. Try with a different flash drive if possible
2. If not possible, run a bad blocks check
3. See if it boots on a different computer

coolrecep, 2020-07-25:
It's not me, it's one of our users. If it was me, I would have provided the log in the first place. I'm here to help you develop a better software. **I'll share the log as soon as I get it.**

He used the same flash drive but this time with Windows USB tool and got success. I've never had a problem with Rufus either.

pbatard, 2020-07-25:
Ah okay, I didn't catch that.

I'll see if I can get my hand on that ISO to test it then as, if it's an issue with Rufus, then it has to do with using this specific `tr_windows_10_consumer_editions_version_2004_x64_dvd_e198bbf6.iso` image, as other Windows 10 2004 ISOs work fine with Rufus. But I can't promise anything.

Can you at least indicate if they tested that USB drive on more than one machine, and saw the same outcome? Or if it was just with one machine?

coolrecep, 2020-07-25:
I and thousand of peope used that ISO and had no issue at all. For the record, this is his laptop:

https://www.lenovo.com/us/en/laptops/thinkpad/thinkpad-e-series/E495/p/22TP2TEE495

(Up to date BIOS)

And this is his USB: SanDisk Ultra Dual Drive 64 GB Micro USB

And this was the thread (Turksih): https://www.technopat.net/sosyal/konu/coezueldue-a-required-device-isnt-connected-or-cannot-be-accessed-hatasi.1008670/

pbatard, 2020-07-25:
> I and thousand of peope used that ISO and had no issue at all.

You should have started with that.

I have tested the image on a few machines (mostly UEFI ones) and saw no issue at all.

<details>
    <summary>Here's the log of how I created it (click the arrow to display it)</summary>

```
Rufus x86 v3.11.1678
Windows version: Windows 10 64-bit (Build 19041.388)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0809 (en-GB)
Will use default UI locale 0x0809
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 3.0 device 'SanDisk Extreme USB Device' (0781:5580)
1 device found
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3818, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 3
Disk GUID: {0326AE05-1F8B-4B8A-9E41-AEA1F19D0DE2}
Max parts: 128, Start Offset: 17408, Usable = 31406914048 bytes
Partition 1:
  Type: EFI System Partition
  Name: 'EFI System Partition'
  ID: {07FDA032-CCE5-4701-B6AC-CAFF869AD718}
  Size: 383 MB (401604608 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2:
  Type: Linux Data Partition
  ID: {991603FC-0A4E-49B3-B6D9-31796B783A8E}
  Size: 27.9 GB (29979836416 bytes)
  Start Sector: 786432, Attributes: 0x0000000000000000
Partition 3:
  Type: Linux Swap Partition
  ID: {9C8C7A43-65FF-434F-9030-3E4E010D6EFB}
  Size: 976 MB (1023410176 bytes)
  Start Sector: 59340800, Attributes: 0x0000000000000000
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'CCCOMA_X64FRE_TR-TR_DV9'
  Size: 4.9 GB (Projected)
  Has a >4GB file
  Uses: EFI
  Uses: Bootmgr (BIOS and UEFI)
  Uses: Install.wim (version 0.13.1)
Using image: tr_windows_10_consumer_editions_version_2004_x64_dvd_e198bbf6.iso (4.9 GB)

Computing checksum for 'D:\Incoming\tr_windows_10_consumer_editions_version_2004_x64_dvd_e198bbf6.iso'...
  MD5:	 404dd74b28fb850ef5adb4024f934358
  SHA1:	 40e76c8dcdf55c5bcafe7e146a06e2c74c5da66d
  SHA256: 44a8f0864a9b230c71e06b78a87c77ef2491885adfab56b859586d81a40a5a32

Format operation started
Requesting disk access...
Will use 'F:' as volume mountpoint
Deleting partitions (This may take a while)...
Deleting ALL partition(s) from disk '\\?\PhysicalDrive5':
● Partition 1 (offset: 1048576, size: 383 MB)
● Partition 2 (offset: 402653184, size: 27.9 GB)
● Partition 3 (offset: 30382489600, size: 976 MB)
Opened \\.\PhysicalDrive5 for exclusive write access
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
[... 976 lines cut ...]
  Size: 29.2 GB (31404796416 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2 (UEFI:NTFS):
  Type: Microsoft Basic Data Partition
  Name: 'UEFI:NTFS'
  ID: {F5F50067-3A77-49D9-AC5A-ECAD76277EFE}
  Size: 512 KB (524288 bytes)
  Start Sector: 61339541, Attributes: 0x0000000000000000
Checking for Rufus updates...
Checking release channel...
No new release version found.
Checking beta channel...
No new beta version found.
```
</details>

I'm all for _"develop(ing) a better software"_, but I don't know what makes you expect that I'm going to go out of my way to investigate an issue that is clearly Lenovo's or Microsoft's to fix, or at the very least, investigate. If something works on 99 machines but doesn't work on 1, the first thing you do is contact the manufacturer of the machine where it doesn't work, or the developer of the software that actually produces the error (Windows setup boot loader) and ask them to investigate, instead of asking the developer of a software application (which you get for free and which, __until proven otherwise__, is unlikely to be the culprit), to invest their own time (which, as opposed to the application, isn't free).

As it stands, I don't have access to a single Lenovo machine I could try to replicate this issue on, and, even if I did consider that it's Rufus' job to try to add __workarounds for every single flawed hardware under the sun__ (which I don't, unless a large enough number of people start to use said flawed hardware, because there's just too much of it) there's simply not much I can investigate, nor is there much advice I can provide without having access to a machine that can replicate the issue. 

I'm not Microsoft, or Lenovo. I don't have a comprehensive test lab, or a bunch of employees that I can boss around to investigate stuff.

Thus, unless you can provide me comprehensive details, originating from Microsoft or Lenovo, indicating that, as opposed to what is the logical assumption here, it's not actuall a Microsoft or Lenovo issue, and that the problem is an actual consequence of the manner in which Rufus created the drive, there's very little I can do about this issue.

I will therefore close it.

closed 2020-07-25 by pbatard

coolrecep, 2020-07-26:
Here is the log, if you are still interested:

`
Rufus x86 v3.11.1678 (Portable)
Windows version: Windows 7 SP1 32 bit
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.04
System locale ID: 0x041F (tr-TR)
Will use default UI locale 0x041F
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'tr-TR'
Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
Found USB 2.0 device 'SanDisk Ultra USB Device' (0781:558A)
1 device found
No volume information for drive 0x81
Disk type: Removable, Disk size: 64 GB, Sector size: 512 bytes
Cylinders: 7480, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {F421FDF1-C3E0-4F43-BBA4-77FAE6FBBB7C}
Max parts: 128, Start Offset: 17408, Usable = 61530405376 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  ID: {B8976A63-F86E-4871-9CB1-54D68F44B3B3}
  Size: 57.3 GB (61528287744 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2 (UEFI:NTFS):
  Type: Microsoft Basic Data Partition
  Name: 'UEFI:NTFS'
  ID: {E096C4A5-C11C-4E6E-974E-4785027E6AA7}
  Size: 512 KB (524288 bytes)
  Start Sector: 120174485, Attributes: 0x0000000000000000
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'CCCOMA_X64FRE_TR-TR_DV9'
  Size: 4.9 GB (Projected)
  Has a >4GB file
  Uses: EFI
  Uses: Bootmgr (BIOS and UEFI)
  Uses: Install.wim (version 0.13.1)
Using image: tr_windows_10_consumer_editions_version_2004_x64_dvd_e198bbf6.iso (4.9 GB)

Format operation started
Requesting disk access...
Will use 'F:' as volume mountpoint
Deleting partitions (Biraz zaman alabilir)...
Deleting ALL partition(s) from disk '\\?\PhysicalDrive1':
● Partition 1 (offset: 1048576, size: 57.3 GB)
● Partition 0 (offset: 61529336320, size: 512 KB)
Opened \\.\PhysicalDrive1 for exclusive write access
No logical drive found (unpartitioned?)
Drive does not appear to be partitioned
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Partitioning (GPT)...
● Creating Main Data Partition (offset: 1048576, size: 57.3 GB)
● Creating UEFI:NTFS Partition (offset: 61529336320, size: 512 KB)
Writing UEFI:NTFS data...
Waiting for logical drive to reappear...
Formatting (NTFS)...
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Format completed.
[... 910 lines cut ...]
CHKDSK dosyalar  1024 dosya kaydDosya do  0 b�y�k dosya kayd  0 bozuk dosya kayd  0 EA kayd  0 yeniden ayrCHKDSK dizinleri do  1210 dizin girdisi iDizin doCHKDSK g�venlik tan  1024 dosya SD'leri/SID'leri iG�venlik tan  93 veri dosyasWindows dosya sistemini g�zden ge�irdi ve hi�bir sorun bulamad
  60086218 KB toplam disk alan902 dosyada    5192896 KB.
       432 KB 95 dizinde.
Sistem taraf     65536 KB G�nl�k dosyasDiskteki kullan      4096 bayt her ay  15021554 ay  13706020 ayNTFS Fixup completed.

Found USB 2.0 device 'SanDisk Ultra USB Device' (0781:558A)
1 device found
Disk type: Removable, Disk size: 64 GB, Sector size: 512 bytes
Cylinders: 7480, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {93CD3EA9-B23D-4624-8755-2A3607B28645}
Max parts: 128, Start Offset: 17408, Usable = 61530405376 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  ID: {ACEB76C9-B8FD-45BF-8D77-6EDF8CF841AA}
  Size: 57.3 GB (61528287744 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2 (UEFI:NTFS):
  Type: Microsoft Basic Data Partition
  Name: 'UEFI:NTFS'
  ID: {BF6CEFC7-31AA-498A-BBB6-54547783E472}
  Size: 512 KB (524288 bytes)
  Start Sector: 120174485, Attributes: 0x0000000000000000
`

pbatard, 2020-07-26:
I'm afraid I'm not, because it doesn't look like an issue that should be solved by Rufus. But thanks for providing the log anyway.