pbatard/rufus #1586: Quick help

BuckN0rris, 2020-07-25:
Hey, i wanted to ask hoe exactly this works because i found nothing on the Internet.

Is rufus Like Media creation tool that lets you make bootstick for Windows, or is it something, that needs to be plugged in you'r PC to be able to use it . I have massive problems with media creation tool because it just stops loading at around 18%, and i heard that rufus helps, but i don't know how this works, so does the pc need the rufus stick every time you boot up the pc or does it work easy as for example media creator tool that needs to install Windows once and then windows works and boots up without the USB drive.

pbatard, 2020-07-25:
> Is rufus Like Media creation tool that lets you make bootstick for Windows

Yes.

Hope that answers your question.

closed 2020-07-25 by pbatard

BuckN0rris, 2020-07-25:
Hey, so it works exactly like MCT or like a installation stick, or does it need to be plugged in every time, i haven't understood the answer.

BuckN0rris, 2020-07-25:
And if yes, do you have to do some specisl settings for it?

pbatard, 2020-07-25:
https://www.google.com/search?q=creating+a+windows+bootable+usb+with+rufus&tbm=vid