pbatard/rufus #1592: Rufus not visible

yeetmeintothevoid, 2020-08-03:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Rufus not visible>

pbatard, 2020-08-03:
There are multiple issues opened for this (#1379, #1230, #845), which I wouldn't mind trying to investigate.

The problem however is that not one single person has performed the steps I asked them to do.

So, once again:

> One thing that may help is if you have DebugView running while you launch Rufus, as Rufus will send its log data there, and it may give a better idea of the stage at which it seems stuck (though I'm afraid it would require more instrumentation to find out precisely where it is stuck). You can download and install DebugView from https://docs.microsoft.com/en-us/sysinternals/downloads/debugview.
>
> Of course, the most helpful thing one can do, if someone happens to see this issue systematically, is install Visual Studio 2019 (the Community Edition is completely free to install and run) on the computer where you are seeing the problem, then clone the github project and recompile/run Rufus from source, in Debug mode. Then you will then be able to find out precisely where in the code Rufus seems to be stuck on launch.
>
> Short of this, because those reports are exceedingly sporadic and appear to be tied to specific environment factors, it's next to impossible for me to troubleshoot the problem...

pbatard, 2020-08-03:
Also, at the very least, please indicate what version of Windows you are trying to run Rufus on.

pbatard, 2020-08-09:
Closing this issue.

Will reopen when/if someone ever decides to go through the steps I have now advised multiple people to perform (with not a single one of them reporting back), and without which I simply cannot investigate this issue.

closed 2020-08-09 by pbatard