pbatard/rufus #1594: Windows 10 (2004) create Win To Go fail - error at extract the ISO

ds2k5, 2020-08-07:
﻿
Tyed with: Rufus 3.11.1678 (Portable) and 3.10.1647 (Portable)
ISO: Win10_2004_German_x64.iso from: https://www.microsoft.com/de-de/software-download/windows10ISO
USB-Stick: 64GB ( had a working Win To Go 1909 on this Stick before )

Run the rufus.exe with local Admin rights.
Select the ISO
Select Windows To Go
Select Pro Version
Filesystem: NTFS
PartitonSchema: GPT
TargetSystem: UEFI (without CSM)

run until: ~23 % then fail with Error: Error at extract ISO

[image]

How to fix this ?

HASH of the ISO is OK

[image]

[image]

MG-96, 2020-08-09:
Had the same Problem, I got these Log Messages in Rufus 3.11.1678:
```
[...]
Extracting: D:\sources\install.wim (4.2 GB)
Write error [0x00000015] The device is not ready.
Retrying in 5 seconds...
Write error [0x00000015] The device is not ready.
Retrying in 5 seconds...
``` 
Also I used the `Win10_2004_German_x64.iso` .

However it worked with no problems on the second try...
Hope this helps if people keep getting this and if further investigation will be done.

pbatard, 2020-08-09:
@MG-96, if you get a `The device is not ready` in the middle of extraction, then this is clearly an issue that has nothing to do with Rufus. It means that either your USB Flash Drive has an issue or some external application (e.g. security solution) is interfering with the copy process.

@ds2k5, you **chose** not to follow the check list and not provide a log (which is further confirmed by the fact that you computed your SHA using an external application, whereas the check list tells you exactly how to do that in Rufus.

As such, and as indicated in the check list, I will close this issue, as I can't help people who choose not to follow the check list and I am missing the data I need to investigate.

The check list is __NOT__ optional.

closed 2020-08-09 by pbatard

sk7, 2021-03-07:
I had a similar error and the reason was that I was opening the ISO image from a network drive (Windows was in Parallels VM and i had the file on the host system). Copying the ISO image to a local file system fixed this for me.