pbatard/rufus #1598: [Feature Request] Support for FTP or URL iso source

tomtom90, 2020-08-12:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
If you find it suitable, could you add support for FTP sources? This would eliminate the need to download image files if publicly available. As for example in a good repository mirror one would have access to nearly every Linux/Unix iso ever needed.
[example can be named if necessary]

Maybe it would be more relevant/usable to just allow URLs as source?

P.S.: Many thanks for the tool and the work you invested. Hope you see potential in this request.
All the best
Thomas

Log
---
Unable to access file path: [0x80070057] The parameter is incorrect.

<FULL LOG>
Rufus x86 v3.11.1678
Windows version: Windows 10 64-bit (Build 19041.450)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0407 (de-DE)
Will use default UI locale 0x0407
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'de-DE'
Found non-USB removable device 'KINGSTON SV300S37A120G' => Eliminated
Found non-USB removable device 'Crucial_CT750MX300SSD1' => Eliminated
Found USB 2.0 device 'Intenso Alu Line USB Device' (058F:6387)
Using autorun.inf label for drive F: 'Ubuntu 20.04.1'
1 device found
Disk type: Removable, Disk size: 8 GB, Sector size: 512 bytes
Cylinders: 979, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {544C4D62-DA32-4EE7-BBE1-81977A509085}
Max parts: 128, Start Offset: 17408, Usable = 8053029376 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  ID: {5D724C24-DD1A-45F3-8964-4416A9FB7D1F}
  Size: 7.5 GB (8051998208 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Unable to access file path: [0x80070057] The parameter is incorrect.

pbatard, 2020-08-12:
Okay, so I guess you would like to use `http://some.server/some.iso` or `ftp://some.server/some.iso` instead of a local file.

I'm afraid I'm not planning to do that because, even as you will probably disagree, from where I stand, I genuinely don't see that many people using such a feature, and it's a lot more complex to add than you think when you factor in TLS, need for retry, need to display progress, protocols and so on. I can tell you with absolute certainty that the cost/benefit ration of such a feature, especially when it comes to maintaining it, is not worth it.

So, no, I am not planning to do that. If you want to access remote files, then please use one of the many solutions that exist to get Windows to map them for you __outside of Rufus__, as if they were local, in which case you'll have your solution, and it'll be a lot more flexible than Rufus, coz you'll be able to save credentials and everything. Also, please see [this FAQ entry](https://github.com/pbatard/rufus/wiki/FAQ#Why_cant_you_just_add_the_feature_I_want).

> [example can be named if necessary]

Please do, because I am not aware of any good permanent mirror that provides an extensive set of Linux distros. That won't change the answer, but I am interested in that mirror you found.

tomtom90, 2020-08-12:
Hey, thanks for your quick response. I somehow expected this.
I have added an FTP to the file explorer as network resource and hoped that windows may had a way to to access them without going around. But of course I see your point and agree.

I'll get my images from [ftp.gwgd.de/pub](ftp.gwgd.de/pub) (official mirror for many distros).
(A gui overview is given at [ftp.gwgd.de](ftp.gwgd.de)
It's a little mess, but once you figured out how the structure is you can find images quiet well.
Ubuntu as example [ftp.gwdg.de/pub/linux/debian/ubuntu/iso/](ftp.gwdg.de/pub/linux/debian/ubuntu/iso/)

pbatard, 2020-08-13:
Thanks for the mirror link. Maybe I haven't looked hard enough, but I'm not seeing ISOs for Manjaro, PopOS, Kali Linux, Fedora and other currently popular distros that I'd expect to see on a mirror that's supposed to provide access to "nearly every Linux/Unix iso ever needed". ;)

At any rate, since I am not planning to implement the feature you request, I will now close this issue.

closed 2020-08-13 by pbatard

tomtom90, 2020-08-13:
Well ok, maybe slightly exeaggurated :D (Considering the vast number of GNU/Linux derivatives I should have probably not said that xD.
Fedora I could find, the rest isn't up.
Should have left the first version. There I wrote "... I ever needed"
And apart from Kali and Manjaro I never used something that isn't available there xD.