pbatard/rufus #1599: ImgBurn may install malware

marcosfrm, 2020-08-14:
ImgBurn is recommended in Rufus site.

https://gitlab.com/fdupoux/sysresccd-src/-/issues/49

https://www.bleepingcomputer.com/forums/t/714642/is-imgburn-infected/

https://www.askwoody.com/forums/topic/warning-re-imgburn-2-5-8-0/

Older installers were clean. But now it is bundled with PUPs.

Possible alternatives:

https://www.burnaware.com/ (BUT use the link at botton "Also you can download free version without optional offers." in the download page)
http://www.anyburn.com/ (did not test it)

pbatard, 2020-08-14:
Goddammit, you're right. The people who own the site silently updated the installer and added 1 MB of whatever crap to it:

[image]

That's a very malicious thing to do. You just don't go around and change an existing installer like that.

Updating the Rufus website to mention other software requires a bit more work than it seems, because it requires editing about 40 translations (yeah, it's mostly search and replace but still), so for now I've just updated the link to point to a page that has older __clean__ versions of the software (https://www.videohelp.com/software/ImgBurn/old-versions), as that's a much simpler update. If you refresh the web page, you should see that it points to clean versions of the ImgBurn installer.

I think I'll take some time to evaluate what application I'd like to replace ImgBurn with, because I'm not happy about burnaware users having to jump through hoops in order not to get adware... Eventually, I may just remove explicit links to CD burning software and let users sort that themselves, since you can't even guarantee that an application that used to be fine will continue to be that way.

Oh, and this folks is exactly the reason why __CLOSED SOURCE SOFTWARE__ needs to be eradicated, even if it's "free/gratis" software like ImgBurn. __CLOSED SOURCE SOFTWARE SHOULD NOT EXIST, PERIOD!__

If ImgBurn wasn't closed source, the minute its developers started to ad malware/adware to its installer, someone on GitHub or wherever would have recompiled their own version without all this crap, and we wouldn't be in this situation.

And that is also the reason why you're never going to see something like that happen to Rufus, because you can rest assured that, if we started to produce a version that contains unwanted stuff like that and put it on our website, someone would be able to fork Rufus and produce their own version without it and it really wouldn't take long for users to figure out what version they should download...

blackcrack, 2020-08-15:
pbatard, calm down, imgburn exist a long time and was no troubles .. maybe be they're files compromised .. or so.. i have wrote something in they're forum.. lets see what's happends .. real freeware and cardware must not be "per se" bad..
best
Blacky

pbatard, 2020-08-15:
> imgburn exist a long time and was no troubles

Well, my concern is that it was last officially updated in 2013, and you just __DON'T__ silently add 1 MB to an installer, 7 years later, and pretend that it's innocuous...

The installer was __silently__ changed, presumably to add more adware or worse. Any software that does that is devious in my book, even more so as LigthningUK seems to indicate that he's fine with leaving that be ([_"The plugin handles what the user is offered, I have nothing to do with that side of things"_](https://forum.imgburn.com/index.php?/topic/25870-imgburn-may-install-malware/&do=findComment&comment=166312)) .

> maybe be they're files compromised

The answer you got is that [they weren't](https://forum.imgburn.com/index.php?/topic/25870-imgburn-may-install-malware/&do=findComment&comment=166312). This adding of uwanted crap is intentional.

> i guess the security of they're server is broken..

Nope. It's just another sad story of a formerly decent product succumbing to the appeal of quick $$$...

> real freeware and cardware must not be "per se" bad..

Closed source freeware __is__ bad. It denies people of the freedom to modify the application to:
1. Find out exactly what it does, and how it does it.
2. Create an use their own modified version if they choose to do so.

As a society, the amount of time we are collectively wasting on account of closed source software or software that doesn't respect user freedom is staggering. This is yet another example of that. As I pointed out, if ImgBurn wasn't closed source, I'd just have pointed to a fork of that project that doesn't install adware, and that would be the end of it. But instead, we find that we have to waste yet more time on this...

closed 2020-08-20 by pbatard

blackcrack, 2020-08-20:
https://forum.imgburn.com/index.php?/topic/25870-imgburn-may-install-malware/&tab=comments#comment-166312
need nothing to say..
you soo right..