pbatard/rufus #1600: suspicious malware reports

abubelinha, 2020-08-16:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I DID NOT click the 'Log' button because my issue just a question, and NOT a bug report.
- [x] The log I am copying is empty - I have NOT removed any part of it. It simply does not exist.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------

I downloaded both the standard **rufus-3.11.exe** and portable **rufus-3.11p.exe** versions, and both show a couple of negative community reports in virustotal.com

https://www.virustotal.com/gui/file/a43bf34285d393f843b05d80f74c4790ed7f6ac636393ca5f6375ae0bb02e719/community

https://www.joesandbox.com/analysis/157876/0/html

So I would like to know if there is some explanation for that, to avoid any risks before using these files.

BTW, I take the opportunity to ask about the differences between these two files, other than the name.
I ask this because the virustotal report provides the same hash, so I get the idea that both file contents are identical.

Thanks a lot.

Log
---
```
Empty log because I still did not use the program. So it did not fail at all.
```

pbatard, 2020-08-16:
First please see [this FAQ entry](https://github.com/pbatard/rufus/wiki/FAQ#Antivirus_X_reports_that_Rufus_contains_malware).

Then you may want to read our [extensive Security page](https://github.com/pbatard/rufus/wiki/Security).

Finally, since this is what Joe Sandbox analysis trips over:

1. We do modify one group policy on startup (and *revert* it on exit) because it is the Microsoft-recommended way to prevent annoying prompts of "Do you want to format this drive" when the user plugs a USB drive while Rufus is running. As careful examination of our code should demonstrate, there's nothing malicious about that.

2. We launch a 'rufus.com' commandline hogger when Rufus is launched from the commandline, on account that, if we don't do that, users of Far Manager will not be able to run the application. This is a trivial executable, the source and complete explanation of which can be found at https://github.com/pbatard/rufus/blob/master/res/hogger

3. We compress using UPX on account of being size conscious and in order to help our users spend less time on the download.

4. Because Rufus is a multi-language formatting application, it will of course deal with low level USB access as well as the creation of autorun.inf files, which is the *only* way one can make a drive with a user label containing extended characters to properly display in explorer. Obviously, we don't do anything when it comes to using autorun.inf to actually run an executable. This is only used to set the label and icon in Windows explorer.

Long story short, I'm growing tired of having to justify that Rufus is not malicious **over and over again**, just because a handful of security vendors do a very lousy job. If you are unhappy about this, please contact these vendors and get them to clean their act because I have much better things to do.

closed 2020-08-16 by pbatard

abubelinha, 2020-08-17:
Thanks @pbatard for the extensive explanation.
I had posted another question but it was [already answered](https://github.com/pbatard/rufus/wiki/FAQ#how-comes-the-portable-and-regular-versions-are-binary-identical) in the FAQs, sorry about it.

I wonder if it would be less confusing to provide a single .exe file, telling user to rename it in order to run as portable.

Like many others, I hitted the problem of "Rufus not being able" to download sys files from github (or my OS not letting Rufus access gbif network, as you explain).
AFAIK, my firewall is not blocking anything but the solution you provide [here](https://github.com/pbatard/rufus/wiki/Usage-Notes#Where_to_manually_copy_the_SyslinuxGRUB_files_if_Rufus_cant_download_them) works perfectly, thanks!

I guess my Win7 might be needing the updates you mention everywhere, so I'll try that before the next time:

https://github.com/pbatard/rufus/issues/487#issuecomment-93869671
https://github.com/pbatard/rufus/issues/1232#issuecomment-431503049
https://github.com/pbatard/rufus/issues/1440#issuecomment-578696762

Thanks a lot for the software and specially for the great support you provide!
Long life to lough Akeo   ;)

pbatard, 2020-08-17:
> I wonder if it would be less confusing to provide a single .exe file

No, that would be super inconvenient for people who want the portable version.

abubelinha, 2020-09-18:
It depends on which people.
In fact I wanted the portable version, and to me it was more inconvenient that both files were identical.

A short explanation in the download site would speed up my experience.
(i.e. "both files are identical except by ther names: that makes them portable or not")

pbatard, 2020-09-18:
> It depends on which people.

Which is precisely what I'm saying. You are assuming that the majority of people will be in your camp (i.e. looking at the binary data, wondering why it's identical and not bothered with having to rename a single download to add a `p` to make it portable). From the reports I get and the profiling of people who download Rufus, I am seeing it very differently.

I can assure you that your experience (checking the two downloads for identity) is a **minority** experience, and I am not going to inconvenience the **majority**, who are not going to look into that or, if they do, are going to head for the FAQ to see if that question is answered, just because **you** got puzzled by it.

> A short explanation in the download site would speed up my experience.

For the same reason as above, because it's a minority issue, the right place to have it clarified is in the FAQ, not on the homepage. If I were to put all the _"I couldn't find it straight away, so this should feature prominently on the homepage!"_ requests I get, even for one-liner clarifications, it would detract from the user experience I am trying to achieve. And I would need to get it translated in 40 languages, which means I'd have to spend an extensive amount of time (chasing after translators and updating the site is far from the 30-seconds people seem to imagine it is. If I were to add 40 translations for a new one liner, I estimate that, overall, I'd probably have to waste 10 hours of worktime doing that. And that's a best estimate).

I'm afraid that I am not interested in speeding up **your** experience as you put it. I am interested in providing a good experience for the majority, and that includes not confusing them with matters that can easily be found in the FAQ and that the vast majority won't care about, let alone notice. So that is why I won't do what you request, sorry.