pbatard/rufus #1601: Rufus made USB accessible only by the BIOS every OS fails to interact with it

Baroneimen, 2020-08-17:
﻿

Checklist
---------
- 

- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.

- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
   Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  Running Rufus on a different computer.
 If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>
Yesterday while creating a Windows 10 bootable USB Rufus failed to do so saying explorer.exe was interfeiring with the process, leaving the USB unreadable, I can't create partitions nor format the drive with Linux or Windows, every application that try to access the USB just crashes
Run different programs to see if the USB had bas sectors every program said that the USB was in a good state,

The BIOS is able to boot from the USB to then fail (missing files) because Rufus didn't finish its thing.

Today Windows is now able to detect the inserted USB but auto ejects it after 15 seconds of it being plugged.
Log
---
```
<FULL LOG>
```
Rufus x86 v3.11.1678
Windows version: Windows 10 64-bit (Build 19041.450)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0410 (it-IT)
Will use default UI locale 0x0410
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'it-IT'
0 devices found
0 devices found
Found USB 3.0 device 'Kingston DataTraveler 3.0 USB Device' (0951:1666)
1 device found
No volume information for drive 0x82
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1881, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 0
Disk ID: 0x68B4B818
Drive has a Zeroed Master Boot Record

Image provided: 'D:\ISOS\CentOS-7-x86_64-LiveGNOME-1708.iso'
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 4.05/0x581e199a (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'CentOS-7-x86_64-LiveGNOME-1708'
  Size: 1.2 GB (Projected)
  Note: File on disk is larger than reported ISO size by 618 KB...
  Uses: Syslinux/Isolinux v4.05
Using image: CentOS-7-x86_64-LiveGNOME-1708.iso (1.2 GB)
Search for conflicting processes was interrupted due to timeout

Format operation started
Requesting disk access...
No drive letter was assigned...
Will use 'F:' as volume mountpoint
Deleting partitions (potrebbe richiedere del tempo)...
No partition to delete on disk '\\?\PhysicalDrive2'
Opened \\.\PhysicalDrive2 for exclusive write access
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
write_sectors: Write error Windows error code 0x000001B1 (FormatMessage error code 0x00003AFC)
  StartSector: 0x00000000, nSectors: 0x1, SectorSize: 0x200
Retrying in 5 seconds...
write_sectors: Write error Windows error code 0x00000037 (FormatMessage error code 0x00003B01)
  StartSector: 0x00000000, nSectors: 0x1, SectorSize: 0x200
Retrying in 5 seconds...
write_sectors: Write error Windows error code 0x00000037 (FormatMessage error code 0x00003B01)
  StartSector: 0x00000000, nSectors: 0x1, SectorSize: 0x200
Retrying in 5 seconds...
write_sectors: Write error Windows error code 0x00000037 (FormatMessage error code 0x00003B01)
  StartSector: 0x00000000, nSectors: 0x1, SectorSize: 0x200
Could not reset partitions

0 devices found

pbatard, 2020-08-17:
> Yesterday while creating a Windows 10 bootable USB Rufus failed to do so saying explorer.exe was interfeiring with the process, leaving the USB unreadable

That's not exactly correct.

When Rufus encounters an error (such as a write error), it will try to report the processes that may **also** be interfering with the drive. But that does not mean that this is the actual cause of the error.

From what you report below, what most likely happen is:
* Some of tour drive flash cell failed
* This prevented Rufus to write to your drive
* Because it got a write error, Rufus reported it along with the **potential** applications that may interfere with writing to the drive

> I can't create partitions nor format the drive with Linux or Windows

That's usually a good indication that your drive is defective and that you will need to replace it.

> Run different programs to see if the USB had bas sectors every program said that the USB was in a good state

Sorry but that's not possible.
If you can't repartition/reformat the drive in Linux and Windows, there's just no way a bad blocks check will work, unless it is not doing its job properly.

So can you please install e2fsprogs on Linux, if it's not already installed, run the `badblocks` command against your device and provide the output. I am **very** interested to see what it reports...

> `write_sectors: Write error`

Constant write errors are usually a good indication that your USB is defective.
Again, what most likely happened is that the flash memory from your drive failed, and therefore Rufus encountered an error while it was trying to write your image to it. You will have to recycle this drive or RMA it to its manufacturer.

Baroneimen, 2020-08-17:
e2fsprogos found the device being defective
Pass completed, 15109421 bad blocks found. (15109421/0/0 errors)

even so, since the USB is stil bootable from the BIOS is there anyway to save this USB or should I just give up and buy a new one?

pbatard, 2020-08-17:
> e2fsprogos found the device being defective
> Pass completed, 15109421 bad blocks found. (15109421/0/0 errors)

That was expected. Your flash memory was already in the process of failing when you used Rufus to write to that drive, and that is why you got write errors in Rufus.

> is there anyway to save this USB or should I just give up and buy a new one?

Flash memory errors are not something you can recover. It's a pure hardware issue and I can guarantee that you are going to waste your time trying to salvage your drive. Just recycle it and get a new one.

I will now close this issue since there is no further help I can provide on issues that have to do with defective hardware.

closed 2020-08-17 by pbatard