pbatard/rufus #1606: Cross-compilation error on Debian buster:

nachtgeist, 2020-08-25:
I tried to cross-compile rufus on Debian Buster using:
```
git clone...
./autoreconf -fis
./configure --host=i686-w64-mingw32
make
```

which yielded:
```
[...]
  CC       rufus-stdfn.o
stdfn.c: In function ‘GetCpuArch’:
stdfn.c:243:7: error: ‘PROCESSOR_ARCHITECTURE_ARM64’ undeclared (first use in this function); did you mean ‘PROCESSOR_ARCHITECTURE_ARM’?
  case PROCESSOR_ARCHITECTURE_ARM64:
       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
       PROCESSOR_ARCHITECTURE_ARM
stdfn.c:243:7: note: each undeclared identifier is reported only once for each function it appears in
make[2]: *** [Makefile:459: rufus-stdfn.o] Error 1
```

I hack-fixed this by adding `#include "process.h"` to stdfn.c - not sure if this is the correct fix though.
However, re-issuing `make` produced a working executable.

Thanks
nachtgeist

pbatard, 2020-08-25:
Duplicate of #1478 and #1541.

Please ask the Debian maintainers to update the MinGW packages on Debian, because `PROCESSOR_ARCHITECTURE_ARM64` has been defined a long time ago in the MinGW headers and it's only because Debian uses an obsolete version of these that you (and other people) are still getting this error.

It will be **A LOT** more productive for you to get the Debian folks to update the MinGW package than ask **EVERY SINGLE MINGW PROJECT** that also happens to compile for ARM64 to redefine the macro, which is the reason why I have to insist that this is the way to go, because you will be helping a lot more people than folks that just happen to recompile Rufus on Linux.

closed 2020-08-25 by pbatard

nachtgeist, 2020-08-25:
Oh…thanks for the hint! Works fine on Bullseye.