pbatard/rufus #1611: NTFS instead of FAT32 on Rufus 3.12.1696.0

skipik, 2020-08-31:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [x] Running Rufus on a different computer.
- [x] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Rufus 3.12.1696.0 chooses NTFS instead of FAT32, but install.wim is 4 006 323 901 bytes. Why?

Log
---
```
Rufus x64 v3.12.1696 (Portable)
Windows version: Windows 10 64-bit (Build 19042.487)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0419 (ru-RU)
Will use default UI locale 0x0419
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'ru-RU'
Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
Found USB 2.0 device 'JetFlash Transcend 16GB USB Device' (8564:1000)
1 device found
Disk type: Removable, Disk size: 16 Гб, Sector size: 512 bytes
Cylinders: 1952, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {EB56317B-ABD5-47C4-8998-A16ABEE287CD}
Max parts: 128, Start Offset: 17408, Usable = 16055761408 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  ID: {9B12945E-CEB2-4C1E-91AC-66336730DAF7}
  Size: 15.0 GB (16053649920 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2 (UEFI:NTFS):
  Type: Microsoft Basic Data Partition
  Name: 'UEFI:NTFS'
  ID: {5C10ACFE-7A24-4E03-81C5-2E2EBD47F20D}
  Size: 512 KB (524288 bytes)
  Start Sector: 31356833, Attributes: 0x0000000000000000
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: '19042.487'
  Size: 4.3 Гб (Projected)
  Uses: EFI
  Uses: Bootmgr (BIOS and UEFI)
  Uses: Install.wim (version 0.13.1)
Using image: Win10_19042.487_x64_2020-08-31.iso (4.3 Гб)
```

skipik, 2020-08-31:
[image]

I can't make it FAT32 at all.

skipik, 2020-08-31:
Link to the iso https://www.dropbox.com/s/4vzxkb151s839p2/Win10_19042.487_x64_2020-08-31.iso?dl=1 if it can help to investigate the problem.

pbatard, 2020-08-31:
1. I don't support unofficial ISOs. Unless you obtained the ISO from Microsoft, and therefore its SHA-1 is a known official value, you are on your own.
2. All I'm seeing here is that you would like to be able to use FAT32, but I'm not seeing anything about the use of NTFS preventing you to boot. Especially, if you believe that you "need" FAT32 for UEFI boot, I would encourage you to read [this relevant FAQ entry](https://github.com/pbatard/rufus/wiki/FAQ#Blah_UEFI_Blah_FAT32_therefore_Rufus_should_Blah). UEFI systems are perfectly able to boot NTFS partitions.

pbatard, 2020-08-31:
Oh, and you seem to be using a 3.12 that you built yourself, which is **unreleased** (I'm not publishing any builds for that) and currently completely unstable. Unless you are seeing a similar issue in the 3.11 release or the 3.12 BETA when I start to publish it, I just can't accept reports against code that I know to be unstable.

closed 2020-08-31 by pbatard