pbatard/rufus #1614: UEFI:NTFS FAT32 option required

shahriar0247, 2020-09-03:
Summary: I need an option to create UEFI:NTFS partition as fat32 (it is always fat), 

Reason: NTFS and FAT support is not available in my relative's laptop. (FAT32 works)

Whole story: So i have a problem, a laptop which has uefi boot (cant change to legacy) can only detect fat32 stuff as bootable. So Uefi:ntfs mood in rufus is useless for me. So i manually created 2 partitions, one ntfs for windows installer, and another fat32 for UEFI:NTFS image. and tada it worked. took me 5 or 6 hours in 2 days to figure out. stupid uefi and stupid asus. thanks

pbatard, 2020-09-03:
I don't think I want to do this unless I start to get many reports of systems with a similar issue (care to give the full model number by the way, so that other people with the same model can find out about this issue?).

As per [design limits](https://en.wikipedia.org/wiki/Design_of_the_FAT_file_system#Size_limits) FAT32 requires a partition size of at least 32 MB, and because I am copying the partition as is (so that UEFI:NTFS is supported for  Windows 7, Windows 8 and Windows 10 early version users, who can **not** mount 2 partitions at once), it would mean I'd need to either include 32 MB of extra data in the application or, preferably, add code for decompression, which would require extra maintenance and testing (please remember that there's a lot more to adding new code &mdash; one must also ensure that it continues to work as the application gets modified).

And whereas you may bark that this "should" be easy to do, from past experiencea, I tend to be cautious about adding these seemingly "simple enough" features, as I've seen way too many of those resulting in  unexpected corollary issues, even when I could have sworn there's no way they would...

This means that, unless I start to see more reports of people using systems where FAT12 doesn't work for UEFI:NTFS, I'm going to consider that there's **always** going to be some poorly designed system that has one quirk or another that is hindering Rufus' operations, and consider the cost/benefit ratio of working around such quirk.

Right now, I'm afraid I'm not seeing the cost of implementing the feature you want as being low enough if it's only going to apply to a single system (for the record, this is the first report in many years of UEFI:NTFS usage that I'm seeing hinting that some UEFI systems appear not to be able to handle FAT12), so I'm going to put this on hold until such time as I have a better picture of the number of systems affected. And this is also why I would **really** like to have the exact model of your ASUS computer.

I'd also like to know if, by any chance, you may also have set the partition type to ESP, which could very well be the limitation instead of FAT21 vs FAT32, which is why I'm cautious on this, as I'd hate to spend days making sure that the UEFI:NTFS partition is created FAT32, only for you to report that it doesn't work and that the issue was the partition type rather than the file system. And there are plenty more non FAT12 vs FAT32 scenarios I can envision that could explain your issue, such as partition alignment, partition size limits and so on. There again, having more details on how exactly you created the partitions for the drive that work, what type of drive it was, how large, etc. would be helpful as, there again, I am very conscious of the cost/benefit ratio, and, for a misdiagnosed issue, this ratio is always infinite...

shahriar0247, 2020-09-05:
Okay cool so umm, this is the model of the laptop: vivobook 15_asus laptop x540ma x543ma

And yeah i agree with you, the ratio is pretty low and maybe its just this one system, or maybe the problem was with UEFI:NTFS image that rufus had and not the fat12 partition type. Becuase

1) I tried flashing just UEFI:NTFS from rufus, it didnt work (no idea the partition type, but it wasnt detected as bootable for sure)

2) I flashed uefi image (from github) without changing the partition type to fat32 or fat12 but it worked (maybe it was fat12) ( I assumed fat32 because i cant change partition type
[image]

actually there will be a dropdown with only one option if there is only one option, and there will be grayed out dropdown (like this) if you cant change partition type, which i just realised (for people reading it later))

3) I made two partitions in my pendrive with windows (no idea how windows detected both of them) and I copied and pasted files which worked so I assumed that UEFI:NTFS in rufus creates FAT12 and doesnt work, and uefi-ntfs image from github creates fat32 like me, and works.

But maybe whatever I am saying is totally meaningless and no idea what I am doing with my life lol. Have a nice day!

pbatard, 2020-09-06:
> I flashed uefi image (from github) without changing the partition type to fat32 or fat12 but it worked

This confirms that this is __not__ a FAT32 vs FAT12 issue, as the image will be written exactly as is, i.e. using FAT12 (the fact that FAT32 is __greyed out__ means that the FAT that will be used __is__ the one from the `.img`, i.e. FAT12), so this proves that your Asus x54#ma UEFI firmware is perfectly capable of handling bootloaders on a FAT12 partition. This isn't entirely surprising (and also the reason why I had some doubts about the use of FAT12 vs FAT32 being the issue at hand) as the FAT driver is standard with the UEFI source code (i.e. it's not something that computer manufacturers have to write themselves when designing a UEFI firmware), and includes support for FAT12 and FAT16 along FAT32. It has also been stable for many many years.

So, really, for a manufacturer to produce a computer that is able to handle FAT32 but not FAT12, they would have to edit the stable FAT driver code that comes natively from the UEFI public source and __remove__ features that make little sense to be removed (why would any manufacturer spend money doing that?).

This means that your issue is not with FAT12 vs FAT32 but something else. One thing could be where the partition starts or how it is aligned (By default, Rufus always places the UEFI:NTFS partition 512 KB before the end of the drive, after another main partitions, and I could see some systems taking objection with that).

All in all, now that we know that it's not a FAT12 vs FAT32 issue then without having access to a system where I can replicate the problem, there's very little I can do to try to fix it as I'm not willing to invest a lot of time (that I really don't have, and that I suspect you don't really have either) trying things at random and having you tell me if it works or not, especially as I already explained that, unless I get multiple reports of an issue affecting a significant number of systems, I am fine with having some features not work on systems that appear to have problematic quirks.

All in all then, I think I'm going to close this issue. Your tests and your report are very interesting (thanks for that) and I will definitely keep them in mind. But I'm afraid there's not much I want to do about this for the time being, sorry.

closed 2020-09-06 by pbatard