pbatard/rufus #1617: Why do you hate UWP? Seems promising tech

nainardev, 2020-09-12:
﻿

Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>

Log
---
```
<FULL LOG>
```

pbatard, 2020-09-12:
Why are you assuming I hate UWP?

The problem with UWP is that:
1. It didn't exist when I started developing Rufus in 2011
2. It's not compatible at all with Windows 7, which Rufus needed to support until early this year (and which I still need to "support", in a totally non-official manner, since Rufus is a utility that allows people to migrate **away** from Windows 7, therefore you do want Rufus to continue to work for a little while on that platform regardless of whether Microsoft officially ended support there or not). It also has limited compatibility with Windows 8, which Rufus officially support. NOT everybody is using Windows 10. If you are designing your app for Windows 10 only, you're doing a disservice to users.
3. It is **exceedingly limiting** when it comes to what an application can do because Microsoft's answer to security concerns for UWP apps has been to cripple the means in which UWP applications can perform low level operation, which is actually why the Windows Terminal team had to devote a lot of time an effort (they do have a large team and near unlimited resources + direct access to Microsoft internal developers to go though such endeavours, I don't) to eventually create a hybrid UWP Windows Terminal application. Especially, if you read https://devblogs.microsoft.com/commandline/building-windows-terminal-with-winui/ you will find that using UWP completely cripples the ability to perform system-level operation, such as accessing USB drives at the block level and many many other operations that Rufus needs to perform:
   > Therefore, if we wanted to build Terminal as a UWP app in 2018, then any of the shells that we would spawn (like cmd.exe, powershell.exe or bash) would be unable to do anything to the system. Could you imagine using your shell without being able to change directories, read file contents, or launch any other processes that you could interact with? Obviously, this was a non-starter for us.  

There are other reasons as well, that I'm not going to go into.

In other words, it's not because **you** were able to create a simple UWP application in less than 5 minutes, that does not need to perform any kind of system access, that UWP is suitable to something else. Just redesigning the UI of Rufus for version 3.0, to make it look more "fluentish" took me about 4 months full time work, and there was nothing that special about it: it's just how long it takes when you take something that looks **deceptively** simple like Rufus (_"How hard can it be to create a bootable drive, right? It's just creating a partition, formatting it and copying a bunch of files from an ISO, right?"_) and update the myriad of elements that need to be taken care of whenever you touch anything from a UI. And, yes, I did consider trying to go through a UWP layer at the time, just like the Windows Terminal folks did, but I can safely say that, had I done that, it would have taken me much longer than the 4 months I spent on the 2.x to 3.x redesign, since it would have required me to split the app into multiple layer (yeah, it sounds simple on paper, but wait until you actually try to do just that), and it would have meant that Windows 7 users would have been be stuck with Rufus 2.x, even though their platform was still officially supported then.

So please remember this:
1. It's not because someone doesn't do something that they are hating it.
2. Don't mistaken being **critical** of something (like Microsoft introducing YET ANOTHER UI LAYER that leaves users of older versions of Windows stranded, instead of maybe improving some of their existing APIs that are still widely used and long overdue for improvements, such as introducing a complete UTF-8 layer for instance) with being spiteful. One can enumerate the many issues that make UWP unsuitable for some tasks, and therefore dismiss it as a solution, and that doesn't mean that "They aren't using UWP because they hate UWP"
3. If you haven't done **proper extensive research** and looked at the many caveats a "promising" technology might have when applied practically to an existing project, switching from one technology to another might look painless and quick on paper. But I'm afraid I have acquired enough experience in software development project and have enough insight into what that effort would actually require for Rufus to know that it's anything but.

At this stage, it'd probably be 2023 (Windows 8.1 formal end of support) before I can realistically look at transitioning away from tried and trusted (and **universal**) Win32/GDI to something else, especially considering that Microsoft is still promising that they might try to get their shit together in terms of allowing Win32 apps to use modern UI APIs with the **NOT YET FINALIZED AND WAY TOO NEW TO TRANSITION TO** WinUI 3.0 (https://github.com/microsoft/microsoft-ui-xaml/blob/master/docs/roadmap.md#winui-3), which, realistically, should be a much better transition path than UWP.

As I believe I have answered your question, I will close this issue.

closed 2020-09-12 by pbatard

nainardev, 2020-09-12:
Woah! That's a very long reply! I was wrong. I thought UWP is a design system for windows apps plus some extra stuff. Like material design,etc.  And i wondered why u dont like a design system.  So, is it yet another desparate attempt by Microsoft to unify platforms by telling us rewrite our programs? 
Also is rufus  available for linux? if not Any similar tools for linux?

Add this to FAQ, so that you don't need to answer idiots like me again and again

pbatard, 2020-09-12:
> Also is rufus available for linux?

https://github.com/pbatard/rufus/wiki/FAQ#Do_you_plan_to_port_Rufus_to_LinuxMac_OSSome_other_OS

> Add this to FAQ,

Having an entry about UWP doesn't make sense to have in the FAQ (it is not a **frequently** asked question). And, unfortunately, as your follow up question shows, people don't seem to pay as much attention to the FAQ as they should... ;)