pbatard/rufus #1618: [Question] Release linux version?

VaGNaroK, 2020-09-12:
Do you have any intention of launching binaries for linux? an appimage for example to be compatible with several distros or flatpak? Why having to dual boot just to be able to use this tool is a headache.

pbatard, 2020-09-12:
The checklist is **NOT** optional. If you had followed it, you would have found the [answer to your question](https://github.com/pbatard/rufus/wiki/FAQ#Do_you_plan_to_port_Rufus_to_LinuxMac_OSSome_other_OS).

If someone asks you to do something before you ask a question, it's not to annoy you, but to make sure that you don't end up asking something that gets asked OVER and OVER and OVER again.

closed 2020-09-12 by pbatard

rzr, 2020-09-27:
What about supporting ReactOS too  or wine ?

pbatard, 2020-09-27:
It's up to ReactOS and/or Wine to provide the Windows APIs Rufus needs to work, not the other way around. I can't magically work around **missing** APIs (such as device enumeration or low level disk access or native Windows file systems) and pop out some kind of replacement of thin air &mdash; That's just not how it works.

If you want Rufus to work on ReactOS or Wine, then you have to ask their developers to provide the APIs it needs to make it so. As long as these API's don't exist or aren't properly implemented, you're pretty much asking me the same as creating a pure Linux version of Rufus, which isn't going to happen.