pbatard/rufus #1620: FEEDBACK: Option to ignore internal SD card readers (or just card readers in general)

agottschling, 2020-09-16:
﻿

Checklist
---------
- [x ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>
One more than one occasion, I've had to be extra careful when formatting drives using rufus on my Dell Latitude 7275 (Dell's version of the Surface Pro) as it has an internal microSD card reader that I use as a second disk. Is there a way to have Rufus ignore these cards or maybe add a feature that will do this?
Cheers!
Log
---
```
Rufus x86 v3.11.1678 (Portable)
Windows version: Windows 10 64-bit (Build 18363.1082)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found card reader device 'SDXC Card'
Using autorun.inf label for drive E: 'SD Card'
1 device found
Disk type: Removable, Disk size: 196 GB, Sector size: 512 bytes
Cylinders: 23934, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0EEE91E1
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: exFAT (0x07)
  Size: 183.3 GB (196864901120 bytes)
  Start Sector: 2048, Boot: No

```

pbatard, 2020-09-16:
This is a very specific problem that not many people are going to have, and I'm afraid that, as much as people would like it otherwise, I'm not going to add extra hand-holding to Rufus to prevent users from making mistakes that they should avoid by paying enough attention to the application.

The same applies to Virtual Hard Drives (which Rufus mounts by default). If you are unhappy about this, then you can alter the source and recompile your own version of Rufus that doesn't list SD drives.

&rarr; Rejected

closed 2020-09-16 by pbatard

agottschling, 2020-09-16:
Thanks for the response, and I understand where you're coming from. I just felt that more users would run into this eventually, but now that I think about it, if it was a big problem, you'd have added something already :D. 
Thanks for a fantastic tool. 
AG