pbatard/rufus #1621: Copying ISO files Kali Live stuck.

DobreinGitHub, 2020-09-23:
Issue description
-----------------
Hey, I've been trying to get Kali Live iso file on my usb and been stuck on 95.4% for quite a while now. This is my 2nd attempt now. First attempt has finished in "Extraction failure". This is not moving anywhere.

Log
---
```
Rufus x86 v3.11.1678 (Portable)
Windows version: Windows 10 64-bit (Build 18362.657)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 2.0 device 'Seagate Desktop USB Device' (0BC2:3300)
Device eliminated because it was detected as a Hard Drive (score 33 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
Found USB 2.0 device 'ADATA USB Flash Drive USB Device' (125F:C08A)
Using autorun.inf label for drive G: 'Run Debian GNU/Linux'
1 device found
Disk type: Removable, Disk size: 7.7GB, Sector size: 512 bytes
Cylinders: 932, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x01AFDC17
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 3.1 GB (3375880704 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2:
  Type: GNU/Linux (0x83)
  Size: 4.0 GB (4294950912 bytes)
  Start Sector: 6595565, Boot: No
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.04/20191223 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: '09_23_2020'
  Size: 3.0 GB (Projected)
  Uses: Syslinux/Isolinux v6.04
  Uses: EFI
Using image: kali-linux-2020.3-live-amd64.iso (2.9 GB)
Downloading https://rufus.ie/files/syslinux-6.04/20191223/ldlinux.sys
Unable to access file: 404
Extended version was not found, trying main version...
Downloading https://rufus.ie/files/syslinux-6.04/ldlinux.sys
File length: 66.5 KB
Successfully downloaded 'ldlinux.sys'
Download signature is valid ✓
Downloading https://rufus.ie/files/syslinux-6.04/20191223/ldlinux.bss
Unable to access file: 404
Extended version was not found, trying main version...
Downloading https://rufus.ie/files/syslinux-6.04/ldlinux.bss
File length: 512 bytes
Successfully downloaded 'ldlinux.bss'
Download signature is valid ✓

Format operation started
Requesting disk access...
Will use 'G:' as volume mountpoint
Deleting partitions (This may take a while)...
Deleting ALL partition(s) from disk '\\?\PhysicalDrive3':
● Partition 1 (offset: 1048576, size: 3.1 GB)
● Partition 2 (offset: 3376929280, size: 4.0 GB)
Opened \\.\PhysicalDrive3 for shared write access
[... 602 lines cut ...]
Extracting: G:\isolinux\libutil.c32 (23.4 KB)
Extracting: G:\isolinux\live.cfg (1 KB)
Extracting: G:\isolinux\menu.cfg (271 bytes)
Extracting: G:\isolinux\splash.png (61.6 KB)
Extracting: G:\isolinux\stdmenu.cfg (591 bytes)
Extracting: G:\isolinux\vesamenu.c32 (26.1 KB)
Extracting: G:\live\filesystem.packages (58.0 KB)
Extracting: G:\live\filesystem.packages-remove (1 bytes)
Extracting: G:\live\filesystem.size (11 bytes)
Extracting: G:\live\filesystem.squashfs (2.6 GB)
Extracting: G:\live\initrd.img (71.6 MB)
Extracting: G:\live\initrd.img-5.7.0-kali1-amd64 (71.6 MB)
Extracting: G:\live\memtest (96.6 KB)
Extracting: G:\live\vmlinuz (5.4 MB)
Extracting: G:\live\vmlinuz-5.7.0-kali1-amd64 (5.4 MB)
Extracting: G:\md5sum.txt (79.4 KB)
Extracting: G:\pool\main\a\acl\libacl1_2.2.53-8_amd64.deb (30.4 KB)
Extracting: G:\pool\main\a\adduser\adduser_3.118_all.deb (234.8 KB)
Extracting: G:\pool\main\a\apt\apt-utils_2.1.7_amd64.deb (423.9 KB)
Extracting: G:\pool\main\a\apt\apt_2.1.7_amd64.deb (1.4 MB)
Extracting: G:\pool\main\a\apt\libapt-pkg6.0_2.1.7_amd64.deb (970.1 KB)
Extracting: G:\pool\main\a\argon2\libargon2-1_0~20171227-0.2_amd64.deb (19.2 KB)
Extracting: G:\pool\main\a\attr\libattr1_1%3a2.4.48-5_amd64.deb (21.0 KB)

```

DobreinGitHub, 2020-09-23:
I cancelled it just now when it's been stuck for 30 minutes. Plugged USB directly to motherboard and checking for bad blocks now, before writing ISO.

pbatard, 2020-09-23:
Please provide the SHA-256 of `kali-linux-2020.3-live-amd64.iso`. At first glance it would look like a corrupted ISO. You can compute the SHA-256 in Rufus by pressing the `(✓)` button.

Makuru-DD, 2020-09-25:
i would advice using the dd mode instead of iso mode, i installed it once by me and kali didnt want to install without it

DobreinGitHub, 2020-09-26:
Solved it with buying new USB-Flash.

closed 2020-09-26 by DobreinGitHub