pbatard/rufus #1623: [WARN] AN NTFS SERVICE IS ALREADY LOADED    //   [FAIL] LOAD FAILURE [3] UNSUPORTED

Avotra27, 2020-09-23:
when trying to run efi\boot\bootx64.efi

﻿***  UEFI:NTFS (X64)  ***

[INFO] UEFI v2.31 (Amercan Megatrends, 0x00050008)
[INFO] Amercan Megatrends Inc. 3.11.2050
[INFO] Microsoft Corporation Surface Pro 3
[INFO] Disconnecting potentially blocking drivers
[INFO] Searching for target partion on boot disk
[INFO]   PciRoot (0)/Pci (0x14, 0x0)/usb(0x1 ,0x0)
[INFO] Fond NTFS target partition:
[INFO]   PciRoot (0)/Pci (0x14, 0x0)/usb(0x1, 0x0)/HD(1, GPT, 28C56060-0925-4157-8EF2-
C192D85891DE) 
[INFO] Checking if target partition needs the NTFS service
**[WARN] AN NTFS SERVICE IS ALREADY LOADED**
[INFO] Opening target partition:
[INFO]   Volume label is 'CCCOMA_X64FRE_EN_US_DV9'
[INFO] This system uses 64-bit x86 UEFI => searching for x64 EFI boot loader
[INFO] Launching 'efi\boot\bootx64.efi'...
**[FAIL] LOAD FAILURE [3] UNSUPORTED**

Press any key to exit.

pbatard, 2020-09-23:
I need the *full** log of how you created the media with Rufus.

If I were to take a guess, I'd say you altered the `efi\boot\bootx64.efi` so that it's not an x86 64-bit bootloader (e.g. renamed a `bootia32.efi` or something). Of course, since you chose not to follow the checklist and provide the log from Rufus an details of how you created the image, this is a complete guess.

As per the check list you chose to ignore, I will also need the SHA-1 of the ISO you used to create your media.

I simply cannot help you if you ignore elements that are being asked **not** to annoy you but because they are crucial to troubleshooting an issue.

Avotra27, 2020-09-23:
ok,
Rufus x86 v3.11.1678
Windows version: Windows 7 SP1 32 bit
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.04
System locale ID: 0x040C (fr-FR)
Will use default UI locale 0x040C
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'fr-FR'
Found USB 2.0 device 'Lexar USB Flash Drive USB Device' (21C4:0CC7)
1 device found
Disk type: Removable, Disk size: 16 Go, Sector size: 512 bytes
Cylinders: 1899, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {287165AF-C74D-4CA7-B88B-B43B391DB970}
Max parts: 128, Start Offset: 17408, Usable = 15623748096 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  ID: {28C56060-0925-4157-8EF2-C192D85891DE}
  Size: 14.5 GB (15621645312 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2 (UEFI:NTFS):
  Type: Microsoft Basic Data Partition
  Name: 'UEFI:NTFS'
  ID: {D059E685-59FA-4064-BDD9-1BE90F99EE81}
  Size: 512 KB (524288 bytes)
  Start Sector: 30513074, Attributes: 0x0000000000000000
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'CCCOMA_X64FRE_EN-US_DV9'
  Size: 4.8 Go (Projected)
  Has a >4GB file
  Uses: EFI
  Uses: Bootmgr (BIOS and UEFI)
  Uses: Install.wim (version 0.0.0)
Using image: Win10_2004_English_x64.iso (4.9 Go)
Search for conflicting processes was interrupted due to timeout
Search for conflicting processes was interrupted due to timeout

Format operation started
Requesting disk access...
Will use 'E:' as volume mountpoint
Deleting partitions (peut prendre du temps)...
Deleting ALL partition(s) from disk '\\?\PhysicalDrive1':
● Partition 1 (offset: 1048576, size: 14.5 Go)
● Partition 0 (offset: 15622693888, size: 512 Ko)
Opened \\.\PhysicalDrive1 for exclusive write access
No logical drive found (unpartitioned?)
Drive does not appear to be partitioned
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Partitioning (GPT)...
● Creating Main Data Partition (offset: 1048576, size: 14.5 GB)
● Creating UEFI:NTFS Partition (offset: 15622693888, size: 512 KB)
Writing UEFI:NTFS data...
Waiting for logical drive to reappear...
Formatting (NTFS)...
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Format completed.
Writing Master Boot Record...
Using Rufus protective MBR
[... 930 lines cut ...]
  10136796 Ko disponibles sur le disque.
      4096 octets dans chaque unit� d'allocation.
   3813878 unit�s d'allocation au total sur le disque.
   2534199 unit�s d'allocation disponibles sur le disque.
NTFS Fixup completed.

Found USB 2.0 device 'Lexar USB Flash Drive USB Device' (21C4:0CC7)
1 device found
Disk type: Removable, Disk size: 16 Go, Sector size: 512 bytes
Cylinders: 1899, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {3BCD8ECA-67CC-41F9-9581-C65887A50F03}
Max parts: 128, Start Offset: 17408, Usable = 15623748096 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  ID: {C8FEA2CE-CF69-439E-B1D6-AEA126CEB9F2}
  Size: 14.5 GB (15621645312 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2 (UEFI:NTFS):
  Type: Microsoft Basic Data Partition
  Name: 'UEFI:NTFS'
  ID: {4166CCCF-545A-4C0C-92C0-33B99A122F5D}
  Size: 512 KB (524288 bytes)
  Start Sector: 30513074, Attributes: 0x0000000000000000

Avotra27, 2020-09-23:
I have no idea for the iso SHA-1; I downloaded the iso from a friend's link, please help me,
where i can find it?

pbatard, 2020-09-23:
Okay, 3 possibilities:

1. Since it has a native NTFS service (i.e. UEFI driver) your Surface should be able to boot from the NTFS partition without using UEFI:NTFS (i.e. without having to go through the second partition). So you should have 2 choices of partitions to boot from from the media, and you should be able to pick the first one. Have you tried that? If so, what happened?
2. If you didn't get to choose between first and second partition it means that the NTFS driver that Microsoft provides on the Surface is crippled (which is strange considering that if someone should be able to produce a fully functional UEFI NTFS driver it should be Microsoft, but considering that they have never bothered helping UEFI/edk2 with providing one publicly, it's wouldn't be entirely surprising), in that case, sorry, but you won't be able to boot because the problem is that the Microsoft native NTFS driver cannot launch UEFI boot files from NTFS partitions. Rufus cannot replace that driver (which is why you get the message `AN NTFS SERVICE IS ALREADY LOADED`) so you are stuck with it, and if the bootloader cannot be launched when using it, then you're stuck.
3. Your ISO has been altered and the UEFI bootloader is either corrupted or has been replaced with one from a different architecture.

For the SHA-1, You can compute it in Rufus by pressing the `(✓)` button after having selected your ISO.

Avotra27, 2020-09-23:
Thank so much, I'm going to try it and i come back
 SHA1:	 6852eb4aea4fcf596b36462937fe63e1287510f5
before i installed windows 10 on my surface it says my ssd is 0MB unallocated,

pbatard, 2020-09-23:
Hmm, `6852eb4aea4fcf596b36462937fe63e1287510f5` [returns nothing here](https://msdn.rg-adguard.net/public.php?seach=+6852eb4aea4fcf596b36462937fe63e1287510f5&str=25) which means your ISO has been **modified from the official Microsoft retail one**.

I would encourage you to download an official retail ISO, which you can do in Rufus (see [here](https://github.com/pbatard/rufus/wiki/FAQ#Help_I_dont_see_the_ISO_download_button)) and use that.

I'm afraid I simply cannot support Windows ISOs that have been altered, and that is the case with yours.

Avotra27, 2020-09-23:
and have you any idea for this?
[image]

Avotra27, 2020-09-23:
thanks again, have you i iso windows for me?

pbatard, 2020-09-23:
> have you i iso windows for me?

I already explained how you can get the ISO. Either through the `DOWNLOAD` button in Rufus of by visiting [this page](https://www.microsoft.com/en-gb/software-download/windows10ISO).

closed 2020-10-02 by pbatard