pbatard/rufus #1625: USB Damaged after Rufus with Ubuntu 20.04

jok3rg, 2020-09-29:
The instalation process stopped at 85% more or less, after that i tried to format USB in Windows and Mac OS, without luck.
Tried with windows option, diskpart, disk utility in Mac and from mac terminal with no luck.
How can i fix the usb? If i plug the usb, its shows in Windows and Mac (wont load Finder or Utility Disk eternal loading)

pbatard, 2020-09-29:
> The instalation process stopped at 85% more or less,

This is indicative of a hardware failure. Your drive failed, therefore Rufus wasn't able to write to it past the 85% mark, as opposed to what you are assuming of Rufus having damaged your drive. If your drive becomes defective (and you should know that the flash memory used in USB drives is a lot less reliable than people think), then, of course, an application that is attempting to write a large chunck of data to it will fail. But this has nothing to do with the application.

> How can i fix the usb?

It's most likely dead. You can try a bad blocks check in Rufus (as suggested by the check list that you chose to disregards) or with any other application that allows for block level validation, but it's unlikely that you'll be able to recover it.

jok3rg, 2020-09-29:
Rufus log, my drive is not defective, its an Sandisk Ultra and i have been using it for a long time with 0 fails, i just used it yesterday with Etcher in Mac OS and worked fine, it just crashed this morning with Rufus and now is dead.

`Rufus x86 v3.11.1678
Windows version: Windows 10 64-bit (Build 19041.508)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0C0A (es-ES)
Will use default UI locale 0x0C0A
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'es-ES'
0 devices found
Checking for Rufus updates...
Checking release channel...
No new release version found.
Found USB 3.0 device 'SanDisk Ultra USB Device' (0781:5581)
1 device found
Disk type: Removable, Disk size: 64 GB, Sector size: 512 bytes
Cylinders: 7789, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x97DEC749
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 59.7 GB (64066945024 bytes)
  Start Sector: 2048, Boot: Yes
1 device found
Disk type: Removable, Disk size: 64 GB, Sector size: 512 bytes
Cylinders: 7789, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x97DEC749
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 59.7 GB (64066945024 bytes)
  Start Sector: 2048, Boot: Yes
1 device found
Disk type: Removable, Disk size: 64 GB, Sector size: 512 bytes
Cylinders: 7789, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x97DEC749
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 59.7 GB (64066945024 bytes)
  Start Sector: 2048, Boot: Yes
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.03/20171017 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'Ubuntu 18.04.5 LTS amd64'
  Size: 2 GB (Projected)
  Has a >64 chars filename
  Uses: Syslinux/Isolinux v6.03
  Uses: EFI
  Note: This ISO uses symbolic links, which will not be replicated due to file system limitations.
  Because of this, some features from this image may not work...
Using image: ubuntu-18.04.5-desktop-amd64.iso (2 GB)
Downloading https://rufus.ie/files/syslinux-6.03/20171017/ldlinux.sys
Unable to access file: 404
Extended version was not found, trying main version...
Downloading https://rufus.ie/files/syslinux-6.03/ldlinux.sys
File length: 67.0 KB
Successfully downloaded 'ldlinux.sys'
Download signature is valid ✓
Downloading https://rufus.ie/files/syslinux-6.03/20171017/ldlinux.bss
Unable to access file: 404
[... 327 lines cut ...]
write_sectors: Write error Windows error code 0x00000079 (FormatMessage error code 0x00003B01)
  StartSector: 0x00000065, nSectors: 0x1, SectorSize: 0x20000
Bad Blocks: 101
write_sectors: Write error Windows error code 0x00000079 (FormatMessage error code 0x00003B01)
  StartSector: 0x00000066, nSectors: 0x1, SectorSize: 0x20000
Bad Blocks: 102
write_sectors: Write error Windows error code 0x00000079 (FormatMessage error code 0x00003B01)
  StartSector: 0x00000067, nSectors: 0x1, SectorSize: 0x20000
Bad Blocks: 103
write_sectors: Write error Windows error code 0x00000079 (FormatMessage error code 0x00003B01)
  StartSector: 0x00000068, nSectors: 0x1, SectorSize: 0x20000
Bad Blocks: 104
write_sectors: Write error Windows error code 0x00000079 (FormatMessage error code 0x00003B01)
  StartSector: 0x00000069, nSectors: 0x1, SectorSize: 0x20000
Bad Blocks: 105
write_sectors: Write error Windows error code 0x00000079 (FormatMessage error code 0x00003B01)
  StartSector: 0x0000006a, nSectors: 0x1, SectorSize: 0x20000
Bad Blocks: 106
write_sectors: Write error Windows error code 0x00000079 (FormatMessage error code 0x00003B01)
  StartSector: 0x0000006b, nSectors: 0x1, SectorSize: 0x20000
Bad Blocks: 107
write_sectors: Write error Windows error code 0x00000079 (FormatMessage error code 0x00003B01)
  StartSector: 0x0000006c, nSectors: 0x1, SectorSize: 0x20000
Bad Blocks: 108
`

pbatard, 2020-09-29:
The bad block test shows that your hardware __is__ defective.

And yes, things tend to work until the day they break. But that doesn't mean that the last application you used was responsible for it. If you had happened to experience this failure while copying a file in Windows File Explorer, I doubt you'd be contacting Microsoft to tell them that you think File Explorer broke your drive. Well, the thing is, despite what you might think, Rufus is actually as safe as using File Explorer, so I can guarantee that, whatever you think might have happened, Rufus was not the cause of your hardware failure.

For more on this, please see [this relevant FAQ entry](https://github.com/pbatard/rufus/wiki/FAQ#Help_Rufus_damaged_my_flash_drive).

closed 2020-09-29 by pbatard