pbatard/rufus #1629: Terminate SearchIndexer.exe automatically

xmha97, 2020-10-06:
﻿

Checklist
---------
- [X] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [X] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [X] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [X] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [X] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_ and confirmed that my USB is not defective.
- [X] I also tried one or more of the following:
  - [X] Using a different USB drive.
  - [X] Plugging the USB into a different port.
  - [X] Running Rufus on a different computer.
- [X] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1, and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I get this error every time I try to create a bootable USB drive:
[image]
This error has been displayed since I enabled this option:
[image]
It's hard for me to terminate this process every time.
Please terminate SearchIndexer.exe automatically

Log
---
```
Rufus x64 v3.11.1678
Windows version: Windows 10 64-bit (Build 19041.508)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 2.0 device 'General USB Flash Disk USB Device' (8644:800B)
Could not get device number for device \\.\P:: [0x00000001] Incorrect function.
Could not get device number for device \\.\P:: [0x00000001] Incorrect function.
1 device found
Could not get Disk Extents: [0x00000001] Incorrect function.
Disk type: Removable, Disk size: 8 GB, Sector size: 512 bytes
Cylinders: 974, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0B124250
Drive has a Rufus Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 7.5 GB (8015314944 bytes)
  Start Sector: 2048, Boot: Yes
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'Win10Ex64IE'
  Size: 4.1 GB (Projected)
  Uses: EFI
  Uses: Bootmgr (BIOS and UEFI)
  Uses: Install.esd (version 0.14.0+)
Using image: Windows 10 20H1-2004 15in1 x64 - Integral Edition 2020.9.9.iso (4.1 GB)
Could not get device number for device \\.\P:: [0x00000001] Incorrect function.
Found potentially blocking process(es) against E:
● [011572] C:\Windows\system32\SearchIndexer.exe /Embedding (rwx)

Format operation started
Requesting disk access...
Could not get device number for device \\.\P:: [0x00000001] Incorrect function.
Will use 'E:' as volume mountpoint
Deleting partitions (This may take a while)...
Could not load VDS Service: [0x80070005] Access is denied.
Notice: Could not delete partitions: [0x80070005] Access is denied.
Could not get Disk Extents: [0x00000001] Incorrect function.
Opened \\.\PhysicalDrive2 for exclusive write access
Could not get Disk Extents: [0x00000001] Incorrect function.
Waiting for access on \\?\Volume{cd366f33-d08c-11ea-b5c6-90489ac6c452} [\Device\HarddiskVolume14]...
Warning: Could not obtain exclusive rights. Retrying with write sharing enabled...
WARNING: The following process(es) or service(s) are accessing \Device\HarddiskVolume14:
● [011572] C:\Windows\system32\SearchIndexer.exe /Embedding (rwx)
You should close these applications before attempting to reformat the drive.
I/O boundary checks disabled
Could not lock access to \\?\Volume{cd366f33-d08c-11ea-b5c6-90489ac6c452}: [0x00000005] Access is denied.
Could not lock volume
Could not get Disk Extents: [0x00000001] Incorrect function.
Re-mounted volume as E: after error

Found USB 2.0 device 'General USB Flash Disk USB Device' (8644:800B)
Could not get device number for device \\.\P:: [0x00000001] Incorrect function.
Could not get device number for device \\.\P:: [0x00000001] Incorrect function.
1 device found
Could not get Disk Extents: [0x00000001] Incorrect function.
Disk type: Removable, Disk size: 8 GB, Sector size: 512 bytes
Cylinders: 974, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0B124250
Drive has a Rufus Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 7.5 GB (8015314944 bytes)
  Start Sector: 2048, Boot: Yes
Found USB 2.0 device 'General USB Flash Disk USB Device' (8644:800B)
Could not get device number for device \\.\P:: [0x00000001] Incorrect function.
Could not get device number for device \\.\P:: [0x00000001] Incorrect function.
1 device found
Could not get Disk Extents: [0x00000001] Incorrect function.
Disk type: Removable, Disk size: 8 GB, Sector size: 512 bytes
Cylinders: 974, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0B124250
Drive has a Rufus Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 7.5 GB (8015314944 bytes)
  Start Sector: 2048, Boot: Yes
```

pbatard, 2020-10-06:
Thanks for the report.

I'm afraid I don't want to add the feature of automatically closing conflicting applications, even if I were to somehow try to whitelist "innocuous" ones or make that an optional setting, or as the user before closing the process, as Rufus is not a task manager, it is bad practice, and the report about potentially blocking processes is just there for information for the user to spend time acting on the process that is causing trouble.

If you find a recurring process that is blocking access to Rufus, then clearly the issue __is__ with that process, and asking other applications to take care of the problem is not the way to go. I mean, what if you decide to use something else than Rufus to format your drive? Are you going to also ask the developers of the application to add a check for conflicting processes and automatic closing of applications? It just doesn't make sense to place the burden of "fixing" an issue that has nothing to do with a specific application on that application, simply because, as opposed to the offending one, it happens to be Open Source and you can get in touch with its developer directly.

As such, considering that I am not seeing any other reports of `SearchIndexer.exe` being an issue for other people, despite the fact that Rufus is being used millions of time every single month, the only thing I can advocate is to investigate the cause and take that issue with Microsoft, because they are the ones who wrote the process that is causing the issue and therefore the ones who need to fix that process' intempestive access to a drive (especially in __write__ mode. Why the heck does `SearchIndexer.exe` need to access a drive in write mode? Something is very very wrong here, because a data indexer should never ever need write access to a drive). Anything else is tiptoeing around the actual issue, and is not going to achieve the result you want since, again, as soon as you start using a different application that needs exclusive access to the drive, you will end up in exactly the same position.

&rarr; Request rejected.

closed 2020-10-06 by pbatard

Hrxn, 2020-10-06:
What about adding the drive path to the *Excluded Folders* option?