pbatard/rufus #1631: Offer exFAT for bootable Windows media

marcosfrm, 2020-10-08:
ms-sys 2.7.0 exFAT boot record should make it possible.

https://sourceforge.net/p/ms-sys/code2/150/

pbatard, 2020-10-08:
exFAT doesn't work for Windows 7 UEFI boot (only Windows 8 installation media and later introduced full support for exFAT &mdash; I __tested__ it), and, since there no demonstrable advantage that exFAT brings over NTFS, and since proposing both would immediately result in users asking _"Which one should I use?"_, the better option is actually not to propose exFAT for now. If exFAT did have some advantage over NTFS, I'd be the first to rush to support it, but, currently, apart from a few very loud but very unsubstantiated claims that exFAT doesn't wear a drive as fast as NTFS (which, IMO, is complete bullshit), there really is little point of adding support for it if you already have a perfectly working NTFS support.

Still, it's nice to know that we have exFAT boot record for ms-sys, so thanks for pointing that out. I'll probably pick these in Rufus eventually, but it's far from a priority right now, sorry.

closed 2020-10-08 by pbatard

marcosfrm, 2020-10-08:
Thanks Pete!