pbatard/rufus #1632: Rufus 3.11 does not allow changing almost any option when trying to burn Pop! Os 20.04

PhantomR, 2020-10-10:
﻿

Checklist
---------
- [x ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [x] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
When using the Pop!_OS 20.04 LTS (NVIDIA) ISO, Rufus greys out all boot options.. I can't choose GPT, I can't choose MBR/UEFI, I can't change the partition name etc.

I see the log mentions something about not being able to find GRUB. Maybe Pop!_OS does not use Grub and this confuses Rufus? 

Log
---
```
Rufus x86 v3.11.1678
Windows version: Windows 10 64-bit (Build 20231.1000)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 3.0 device 'Patriot Memory USB Device' (13FE:6300)
Could not get device number for device \\.\D:: [0x00000001] Incorrect function.
Could not get device number for device \\.\F:: [0x00000001] Incorrect function.
Could not get device number for device \\.\G:: [0x00000001] Incorrect function.
Could not get device number for device \\.\H:: [0x00000001] Incorrect function.
Could not get device number for device \\.\D:: [0x00000001] Incorrect function.
Could not get device number for device \\.\F:: [0x00000001] Incorrect function.
Could not get device number for device \\.\G:: [0x00000001] Incorrect function.
Could not get device number for device \\.\H:: [0x00000001] Incorrect function.
1 device found
No volume information for drive 0x81
Disk type: Removable, Disk size: 64 GB, Sector size: 512 bytes
Cylinders: 7532, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x02079E18
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: GNU/Linux (0x83)
  Size: 57.7 GB (61956161536 bytes)
  Start Sector: 4096, Boot: No
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.03/20171017 (from '/isolinux/isolinux.bin')
Could not get ISO-9660 file information for file boot/grub/i386-pc/normal.mod
  Could not read Grub version from 'boot/grub/i386-pc/normal.mod'
  Could not detect Grub version
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'Pop_OS 20.04 amd64'
  Size: 2.4 GB (Projected)
  Has a >64 chars filename
  Uses: Syslinux/Isolinux v6.03
  Uses: EFI
  Note: This ISO uses symbolic links, which will not be replicated due to file system limitations.
  Because of this, some features from this image may not work...
Note: ISO mode will be disabled because this ISOHybrid is not compatible with ISO boot.
Using image: pop-os_20.04_amd64_nvidia_13.iso (2.4 GB)

Computing checksum for 'C:\Users\Alexandru Dan Duna\Downloads\pop-os_20.04_amd64_nvidia_13.iso'...
  MD5:	 3b0466c40f3fb897fed058f7525494ce
  SHA1:	 18600c6b15cb2f6c99d43c6e710d2be911f67bd8
  SHA256: 18acffc5b4956be331deb5f4f22e0ead6c1716809550665de8e29eeb041d97f2
```

pbatard, 2020-10-11:
As with any other Windows applications, options that are greyed out are options that are not applicable.

In this case, PopOS can only be written in DD mode, since its maintainers have made it incompatible with ISO mode, therefore options like partition scheme or file system, that only apply to ISO mode will be greyed out.

This is not a result of a bug. But if you're worried about the GRUB detection issue, you can try with Rufus 3.12 BETA (available [here](https://rufus.ie/downloads/)) and see for yourself.

If you want to be able to change the partition scheme, then please ask the PopOS maintainers to ensure that their ISOHybrid image follows what other Linux distros do, and can be booted when extracting the ISO content to a FAT32 drive.

closed 2020-10-11 by pbatard

PhantomR, 2020-10-11:
Thank you, I may try to contact them and I appreciate your help.

P.S I did try on that Beta (I edited my post and mentioned this) and I think it still gave thay grub error. Should it not? I can retry tomorrow if it helps.

pbatard, 2020-10-11:
Not seeing any GRUB issue with the beta:
```
Rufus x86 v3.12.1707 (Beta)
Windows version: Windows 10 64-bit (Build 19041.508)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0809 (en-GB)
Will use default UI locale 0x0809
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
0 devices found
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.03/20171017 (from '/isolinux/isolinux.bin')
  Detected Grub version: 2.02
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'Pop_OS 20.04 amd64'
  Size: 2.4 GB (Projected)
  Has a >64 chars filename
  Uses: Syslinux/Isolinux v6.03
  Uses: GRUB2
  Uses: EFI
  Note: This ISO uses symbolic links, which will not be replicated due to file system limitations.
  Because of this, some features from this image may not work...
Note: ISO mode will be disabled because this ISOHybrid is not compatible with ISO boot.
Using image: pop-os_20.04_amd64_nvidia_5.iso (2.4 GB)
Checking for Rufus updates...
Checking release channel...
No new release version found.
Checking beta channel...
No new beta version found.
```
Please retry.

PhantomR, 2020-10-12:
You are right, it does not happen, I apologize.

```
Rufus x86 v3.12.1707 (Beta)
Windows version: Windows 10 64-bit (Build 19041.508)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 2.0 device 'General USB Flash Disk USB Device' (21C4:8005)
1 device found
No volume information for drive 0x83
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1901, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x7DA206BC
Drive has an unknown Master Boot Record
Partition 2:
  Type: EFI FAT (0xef)
  Size: 4 MB (4194304 bytes)
  Start Sector: 324, Boot: No
Partition 3:
  Type: GNU/Linux (0x83)
  Size: 12.2 GB (13073711104 bytes)
  Start Sector: 5013504, Boot: No
Found USB 2.0 device 'General USB Flash Disk USB Device' (21C4:8005)
1 device found
No volume information for drive 0x83
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1901, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x7DA206BC
Drive has an unknown Master Boot Record
Partition 2:
  Type: EFI FAT (0xef)
  Size: 4 MB (4194304 bytes)
  Start Sector: 324, Boot: No
Partition 3:
  Type: GNU/Linux (0x83)
  Size: 12.2 GB (13073711104 bytes)
  Start Sector: 5013504, Boot: No
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.03/20171017 (from '/isolinux/isolinux.bin')
  Detected Grub version: 2.02
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'Pop_OS 20.04 amd64'
  Size: 2.4 GB (Projected)
  Has a >64 chars filename
  Uses: Syslinux/Isolinux v6.03
  Uses: GRUB2
  Uses: EFI
  Note: This ISO uses symbolic links, which will not be replicated due to file system limitations.
  Because of this, some features from this image may not work...
Note: ISO mode will be disabled because this ISOHybrid is not compatible with ISO boot.
Using image: pop-os_20.04_amd64_nvidia_13.iso (2.4 GB)

```