pbatard/rufus #1636: ISO → ESP mode timeout when reformating on v3.11.1678

lichforever1995, 2020-10-15:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I was trying out the Write in ISO → ESP mode feature of Rufus version 3.11 and this message popped up a while after I clicked start:
```
Error:[0xC00305B4] This operation returned because the timeout period expired.
```
And the installation failed.

After checking the log and read through the source code, I was confused how this mode is supposed to work. I first assumed this mode allows writting iso file contents into a ESP partition while left the rest of the drive unpartitioned. However, it fails after repartitioning the drive because the newly created ESP partition is not recognized by WaitForLogical() and eventually times out. 
The drive is left with a empty ESP partition and does not shows up on Explorer.
So how does this mode actually works?

PS: The image i used is a custom iso using grub2 efi mode, works just fine with ISO image mode

PS: I did a sketchy patch to create a normal data partition instead of ESP partition (by not assign the ESP GUID to that partition), and force convert it to ESP partition ( via the ToggleEsp()) after installation. And it works as how I assumed it would.

Log
---
```
Rufus x86 v3.11.1678
Windows version: Windows 10 64-bit (Build 18363.1082)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0804 (zh-CN)
NOTE: No Windows Language Pack is installed for English (English) on this system.
This means that some controls may still be displayed using the system locale.
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
Found USB 3.0 device 'SanDisk SDDR-B531 USB Device' (0781:CFCB)
1 device found
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3880, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {DFD7F50C-1671-43B4-B33A-DF4801317AFF}
Max parts: 128, Start Offset: 17408, Usable = 31914949120 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Basic data partition'
  ID: {264E0A12-745D-4C29-80D0-4564885EFEE9}
  Size: 29.7 GB (31913410560 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Scanning image...
ISO analysis:
  Image is an ISO9660 image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'RESCUE_DISK'
  Size: 67.8 MB (Projected)
  Uses: EFI
Using image: rescue-uefi_dev_1.1.0_20201012_140139.iso (68.2 MB)
Search for conflicting processes was interrupted due to timeout

Format operation started
Requesting disk access...
Will use 'E:' as volume mountpoint
Deleting partitions (This may take a while)...
Deleting ALL partition(s) from disk '\\?\PhysicalDrive2':
● Partition 1 (offset: 1048576, size: 29.7 GB)
Opened \\.\PhysicalDrive2 for exclusive write access
No logical drive found (unpartitioned?)
Drive does not appear to be partitioned
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Partitioning (GPT)...
● Creating EFI System Partition (offset: 1048576, size: 127 MB)
Waiting for logical drive to reappear...
Timeout while waiting for logical drive
Logical drive was not found - aborting

Found USB 3.0 device 'SanDisk SDDR-B531 USB Device' (0781:CFCB)
No logical drive found (unpartitioned?)
1 device found
No volume information for drive 0x82
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3880, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {D4459614-2910-45CA-AFB5-2257A552B768}
Max parts: 128, Start Offset: 17408, Usable = 31914949120 bytes
Partition 1:
  Type: EFI System Partition
  Name: 'EFI System Partition'
  ID: {6FA8FEBF-C02F-49F0-A5AE-227DEFAABF43}
  Size: 127 MB (133169152 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
```

pbatard, 2020-10-15:
Not seeing this issue, be it with Rufus 3.11 or 3.12.

Your assumption about what this mode does is correct, but where you are assuming wrong is that Windows won't let you write to an ESP. If you have the drive GUID (which is what `WaitForLogical()` waits on) and you are running elevated (which we are), you can write to an ESP even if it doesn't have a drive letter assigned. In other words, it's not because Windows was designed by Microsoft not to assign a drive letter to an ESP that it makes the ESP unavailable or that you can't write to it.

I know however that, for reasons that only Microsoft knows, Windows can sometime decide to be a pain in the ass when trying to access a logical drive that was just created, so I would advise you to try the following:
1. Try multiple times
2. Unplug and plug the USB drive into a different port
3. Try <kbd>Alt</kbd>-<bkd>Z</kbd> to zero the drive (you can cancel after a while) or try to create a FreeDOS bootable one, and then retry.

If all this fails, then I will need to check your ISO, because I am not seeing an issue with the ones I use.

For the record, here's an example of ISO &rarr; ESP mode with the latest `debian-testing-arm64-netinst.iso`  from https://cdimage.debian.org/cdimage/weekly-builds/arm64/iso-cd/ (which is something I and others test regularly as this the method that's advocated to [install vanilla Debian 11 on a Raspberry Pi 4](https://www.raspberrypi.org/forums/viewtopic.php?f=50&t=282839)):

<details>
    <summary>Log</summary>

```
Rufus x86 v3.11.1678
Windows version: Windows 10 64-bit (Build 19041.572)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0809 (en-GB)
Will use default UI locale 0x0809
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 3.0 device 'SanDisk Extreme USB Device' (0781:5580)
Using autorun.inf label for drive F: 'ISO Label'
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1946, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {E5A71871-F672-4CFE-9759-3B92DE0F82E2}
Max parts: 128, Start Offset: 17408, Usable = 16013908480 bytes
Partition 1:
  Type: EFI System Partition
  Name: 'EFI System Partition'
  ID: {EF0D326F-46AD-4488-99F9-9CE361867BBD}
  Size: 63 MB (66060288 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Scanning image...
ISO analysis:
  Image is an ISO9660 image
Disk image analysis:
  Image has a Zeroed Master Boot Record
  Image is a bootable disk image
ISO label: 'Debian testing arm64 n'
  Size: 325.9 MB (Projected)
  Note: File on disk is larger than reported ISO size by 3.3 MB...
  Uses: EFI
  Note: This ISO uses symbolic links, which will not be replicated due to file system limitations.
  Because of this, some features from this image may not work...
Using image: debian-testing-arm64-netinst-20201012.iso (310.3 MB)

Format operation started
Requesting disk access...
Will use 'F:' as volume mountpoint
Deleting partitions (This may take a while)...
Deleting ALL partition(s) from disk '\\?\PhysicalDrive5':
● Partition 3 (offset: 1048576, size: 63 MB)
Opened \\.\PhysicalDrive5 for exclusive write access
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Partitioning (GPT)...
● Creating EFI System Partition (offset: 1048576, size: 383 MB)
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (FAT32)...
[... 1007 lines cut ...]
Extracting: F:\.disk\base_components (5 bytes)
Extracting: F:\.disk\base_installable (0 bytes)
Extracting: F:\.disk\cd_type (13 bytes)
Extracting: F:\.disk\info (84 bytes)
Extracting: F:\.disk\mkisofs (1000 bytes)
Finalizing, please wait...
Created: F:autorun.inf
Created: F:autorun.ico

Found USB 3.0 device 'SanDisk Extreme USB Device' (0781:5580)
Using autorun.inf label for drive F: 'Debian testing arm64 n'
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1946, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {228D0EB6-FF5A-4DF3-AA99-DD862426A188}
Max parts: 128, Start Offset: 17408, Usable = 16013908480 bytes
Partition 1:
  Type: EFI System Partition
  Name: 'EFI System Partition'
  ID: {53D4E98B-873E-4F87-85C8-2C1BA8CCBCE8}
  Size: 383 MB (401604608 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
```
</details>

lichforever1995, 2020-10-16:
Hello, thanks for the reply! 

I did some more testing and it appears that this mode does not work with the micro sd card reader I was using. Partitions don't get refreshed until unplug and plug in again (only in this mode though, never had any issue when using iso image mode).

Swapped it for a generic usb thumb drive and it works great!

The model is SanDisk MobileMate USB 3.0 microSD Card Reader.

closed 2020-10-16 by lichforever1995

lichforever1995, 2020-10-16:
Oh welp...Tried on another computer and same thing happens again even on the generic usb drive. Seems like the card reader is not the only one to blame.

How I managed to make it works:
- Format the whole usb drive with Fat32
- Go to another computer running Windows 10, plug the drive in, run `rufus 3.11`
- Used `the debian-testing-arm64-netinst.iso`
- Run ISO -> ESP mode
- Works great

However, once I unplug and plug in again, it won't work anymore, not even if I use `Alt +Z` and reformat it to original state. Anyway, it seem to be a Windows problem instead of rufus's issue.

pbatard, 2020-10-16:
Yeah, the problem is Microsoft appears making it really really hard to work on removable drive with system partitions such as ESPs (see #1637 for an example), but unlike the issue in #1637, I have not been able to reproduce your issue so far.

I'll see if I can play with some more devices, but yeah, I'm afraid that this is fundamentally a Windows issue, as Microsoft are doing their darnedest to forcibly prevent access to system partitions, which can dramatically interfere with Rufus' ability to create a drive...

lichforever1995, 2021-01-25:
Just a quick update.
After I updated rufus to 3.12.1710 release, and windows to  20H2(19042.746), this issue has been resolved.
I can now reliably create boot drive with ESP mode with no problem! Great job!