pbatard/rufus #1637: Windows To Go - ISO Image Extraction Failure

buddyrtc, 2020-10-15:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [x] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
First, thanks for providing and maintaining this tool. I am attempting to install Windows To Go on a 64 GB Sandisk Cruzer USB. The ISO I am using is downloaded from the Windows 10 site without using the media creation tool; clicking the checkbox in Rufus and pasting the SHA1 hash into the website provided in the Rufus FAQ confirms the validity of the ISO. Bad block check has come back negative. 

After a successful but very long bad block check, progress halted around the 25% mark due to ISO Image Extraction Failure. This same error has occurred on two other USBs, plugged into two different drives on my laptop at around the same level of progress. Any help here (even if it's just to say "give up, not going to work") would be greatly appreciated!!!

Log
---
```
Apologies for not copy/pasting the log as instructued - I hit the maximum character limit for the post when I attempted to do so. Log from same attempt attached in .log and .txt format.

[rufus.log](https://github.com/pbatard/rufus/files/5388393/rufus.log)
[buddyrtc-rufuslog.txt](https://github.com/pbatard/rufus/files/5388394/buddyrtc-rufuslog.txt)

```

pbatard, 2020-10-16:
Yes, I'm afraid I've seen that issue to some extent with some drives with the removable attribute (drives with the fixed attribute seem to fare better) ever since I tried to switch to creating the Windows partition (ESP, MSR, Data) in the Microsoft recommended fashion.

The problem as far as I can see is that, for reasons that Microsoft will __really__ have to explain the logic of__, when it sees a drive that looks to it like a Windows system disk on a removable media, it may decide to suddenly remove the drive letter of the data partition as applications are accessing it. So this is what you are seeing here: The image apply errors are due to Rufus no longer finding the target drive, which is further confirmed by:
```
Device eliminated because it appears to contain no media
0 devices found
```

This is not something that I'm seeing consistently however, and, as I said, it doesn't seem to occur with FIXED media (which are better suited for Windows To Go anyway), but so far I haven't found a good workaround to tell Windows not to suddenly disconnect the drive. For instance, in case this was just another instance of Windows trying to hide an ESP from the user (since Windows forcibly prevents people from being able to access an ESP), I tried to create the ESP as a regular data partition, but this didn't seem to help. From what I can tell it seems to be a combination of Windows seeing a "small FAT partition/ESP + MSR + main data partition" in that order, on a drive with the removable attribute, that appears to trigger a potential drive removal from Windows.

So, if you are experiencing this issue, you may want to try to revert to Rufus 3.11, as we are using a "MSR + main data partition + ESP" layout that doesn't seem to produce that behaviour. Or you can try to use a drive with the fixed attribute, which are less common, but, again, more suitable to be used as a Windows To Go drive.

Or you can try to contact Microsoft support and ask them what the heck is their problem with forcibly removing a removable drive in the middle of a file copy operation, when it contains an ESP and an MSR, because it's certainly not Rufus that is triggering that removal...

pbatard, 2020-10-16:
I have tested some more. There's an unrelated issue due to ba406843f40d1a2e6af4b19d36fcfd37ea4a8fd0 (I may need to find another way to work around the `GetVolumePathNamesForVolumeName()` Windows bug, because this is sometimes needed to pick the right data partition for Windows To Go) that made my earlier testing inconclusive, but it looks like working with an ESP that was set as a regular partition may actually prevent Windows from disconnecting the main data partition while we are applying the image. So I think I'll amend Rufus so that we work with a regular partition and only switch to ESP type once the process is complete.

There are a few other issues I'm picking up though, so it may not be that simple...

buddyrtc, 2020-10-16:
Thank you for the response, Pete. Please let me know if I can provide any additional details that can be of use.

pbatard, 2020-10-20:
Okay, so by the looks of it, Windows has a **BUG** whereas, **for removable drives**, it will sometimes drop the last logical volume for not particular good reason.

This means that if you have something like:
* `\\?\Volume{ca7c9f6d-12b4-11eb-9c48-f0038c9a2e6f}\` @1048576 (ESP)
* `\\?\Volume{ca7c9f6f-12b4-11eb-9c48-f0038c9a2e6f}\` @273678336 (MSR)
* `\\?\Volume{ca7c9f70-12b4-11eb-9c48-f0038c9a2e6f}\` @407896064 (Main Data)

Then, depending on some Microsoft's internal lottery, Windows may only report:
* `\\?\Volume{ca7c9f6d-12b4-11eb-9c48-f0038c9a2e6f}\` @1048576 (ESP)
* `\\?\Volume{ca7c9f6f-12b4-11eb-9c48-f0038c9a2e6f}\` @273678336 (MSR)

And the thing is that, as we have seen, Windows might actually remove that last volume while you are already in the process of using it. You can actually see that with the `mountvol` command. If you create 3 partitions on a removable drive, `mountvol` will sometimes list 3 volume GUIDs for that drive, and sometimes only 2...

Now, when we used to have MSR, Main Data and ESP in that order, this wasn't much of an issue, because we have other means than the volume GUID to access the ESP, and Windows tends to leave it alone anyway, since it is designed to hide it regardless. But of course, once you move the Main Data partition to the last slot, as we did for Rufus 3.12, having Windows remove it right under your feet it a lot more problematic...

I guess the "solution" is to create another extra small partition after the Main Data, that Windows can then remove at its leisure, so that it leaves the Main Data partition alone, in a technique that I call _"dangling keys in front of a toddler to distract it"_, but I sure wish Windows had achieve object permanence, instead of forcing programmers to babysit it...

pbatard, 2020-10-26:
Okay, after much trial and error, I think I may have found a way to work around this issue, which will be part of Rufus 3.13.

If you are affected by this, can I please ask you to download [Rufus 3.13 ALPHA](https://rufus-web.akeo.ie/testing/rufus-3.13_ALPHA.exe) and report?

During my testing with this version, I did not see the previous errors when creating a Windows To Go drive on a removable media, but of course, I wouldn't mind some more testing to validate that the workaround works.

khan570, 2020-10-27:
That same problem happens to me but with linux (distro based on debian) as I would in that case

Will reuse 'ldlinux.sys' and 'ldlinux.bss' from './rufus_files/rufus_files/syslinux-6.04/20191223/' for Syslinux installation

Format operation started
Requesting disk access...
Will use 'D:' as volume mountpoint
Deleting ALL partition(s) from disk '\\?\PhysicalDrive1':
● Partition 1 (offset: 1048576, size: 3.0 GB)
● Partition 2 (offset: 3221229568, size: 12.0 GB)
Opened \\.\PhysicalDrive1 for shared write access
Analyzing existing boot records...
Drive has a Syslinux Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Partitioning (MBR)...
● Creating Main Data Partition (offset: 1048576, size: 3.0 GB)
● Creating Linux Persistence Partition (offset: 3221229568, size: 12.0 GB)
Closing existing volume...
Waiting for logical drive to reappear...
Using Debian-like method to enable persistence
Notice: Using physical device to access partition data
Formatting (ext3)...
393215 possible inodes out of 3145723 blocks (block size = 4096)
157286 blocks (5.0%) reserved for the super user
Creating 96 inode sets: [1 marker = 1.2 set(s)]
++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
Creating 8192 journal blocks: [1 marker = 102.4 block(s)]
++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
Created 'persistence.conf' file
Done
Formatting (FAT32)...
Using cluster size: 8192 bytes
Quick format was selected
Creating file system...
Format completed.
Writing Master Boot Record...
Partition is already FAT32 LBA...
Set bootable USB partition as 0x80
Using Syslinux MBR
Found volume \\?\Volume{3423a23b-1868-11eb-b883-1867b0779ec2}\
Installing Syslinux 6.04...
Opened \\?\Volume{3423a23b-1868-11eb-b883-1867b0779ec2} for shared write access
Using existing './rufus_files/syslinux-6.04/20191223/ldlinux.sys' ✓
Using existing './rufus_files/syslinux-6.04/20191223/ldlinux.bss' ✓
Successfully wrote 'ldlinux.sys'
Successfully wrote Syslinux boot record
Successfully remounted \\?\Volume{3423a23b-1868-11eb-b883-1867b0779ec2}\ as D:
Extracting files...
Image is an ISO9660 image
This image will be extracted using Rock Ridge extensions (if present)
Extracting: D:\.disk\archive_trace (1 bytes)
Extracting: D:\.disk\base_components (5 bytes)
Extracting: D:\.disk\base_installable (0 bytes)
Extracting: D:\.disk\cd_type (5 bytes)
Extracting: D:\.disk\info (100 bytes)
Extracting: D:\.disk\mkisofs (536 bytes)
Extracting: D:\.disk\udeb_include (49 bytes)
Extracting: D:\autorun.inf (133 bytes)
Extracting: D:\boot\grub\config.cfg (268 bytes)
Extracting: D:\boot\grub\dejavu-bold-14.pf2 (186.7 KB)
Extracting: D:\boot\grub\dejavu-bold-16.pf2 (207.2 KB)
Extracting: D:\boot\grub\efi.img (736 KB)
Extracting: D:\boot\grub\grub.cfg (2.9 KB)
  Added 'persistence' kernel option
Extracting: D:\boot\grub\i386-efi\acpi.mod (9.6 KB)
Extracting: D:\boot\grub\i386-efi\adler32.mod (1.3 KB)
Extracting: D:\boot\grub\i386-efi\ahci.mod (15.4 KB)
Extracting: D:\boot\grub\i386-efi\all_video.mod (500 bytes)
[... 1036 lines cut ...]
Retrying in 5 seconds...
Write error Windows error code 0x00000070 (FormatMessage error code 0x00003B01)
  Error writing file: Windows error code 0x00000070 (FormatMessage error code 0x00003B01)
Extracting: D:\pool-udeb\main\s\sysfsutils\libsysfs2-udeb_2.1.0+repack-6_amd64.udeb (13.9 KB)
  Unable to create file: Windows error code 0x00000003 (FormatMessage error code 0x00003B01)
Created: D:\syslinux.cfg → /isolinux/isolinux.cfg
Updating D:\md5sum.txt:
● /boot/grub/grub.cfg

Found USB 2.0 device 'Generic Flash Disk USB Device' (0000:7777)
Using autorun.inf label for drive D: 'Run Debian GNU/Linux'
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1958, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x03C5656B
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 3.0 GB (3220180992 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2:
  Type: GNU/Linux (0x83)
  Size: 12.0 GB (12884884992 bytes)
  Start Sector: 6291464, Boot: No

pbatard, 2020-10-27:
Please don't truncate the log, as I am missing important information.

For one thing, I have no idea if this was with 3.12 or 3.13 ALPHA. And you chose not to disclose the actual ISO you were using, which is annoying. I'm afraid I will need this data from you, as I can't do much without it.

pbatard, 2020-10-27:
Oh, and for the record, error code `0x00000070` is `ERROR_DISK_FULL`. Are you sure your persistent partition isn't too big?
Rufus tries to limit the maximum size you can choose for persistence according to how much it thinks the ISO will use, but this is an **approximation**, and if you are really trying to go for the largest size, the ISO partition might be too small.

What happens if you set the size of the persistent partition to 10 GB?

pbatard, 2020-11-14:
For anybody still facing this issue, please try with [Rufus 3.13 BETA](https://rufus.ie/downloads/) and let me know if you are still seeing issues.

pbatard, 2020-11-20:
Closing on account that this should be fixed in the 3.13 release

closed 2020-11-20 by pbatard

aucampia, 2021-02-28:
I had this problem when trying to use ISO from a network drive, worked fine after I copied it to a local drive. Just noting it in case someone else has this issue.

pbatard, 2021-02-28:
The network access issue is [documented in the FAQ](https://github.com/pbatard/rufus/wiki/FAQ#Why_can_Rufus_not_see_my_network_shares).

Please bear in mind that an application that runs elevated does not have the same credentials to access or map network shares as your regular user, and if you have network drives, you will see the exact same access problems if you to run `notepad.exe` elevated and try to access a document residing on a network drive.

So, as advised by the checklist, please don't forget to read the FAQ, as you may find that your issue is already known and explained there.