pbatard/rufus #1639: USB unrecognized by Windows need help

CyberFox2795, 2020-10-16:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Okay I literally don't have a log because of what happened. So I downloaded rufus to make a quick linux bootable USB. I selected my device, then the ISO and it said it needed 2 more files. Something about needing a .bss file or two that rufus didn't have. I clicked yes for it to download them and it hung on a 512 byte download for 5 minutes. That is INSANE. So naturally I just decided to cancel the operation and use a different software. This left my USB unrecognized by Rufus and Windows Disk Management. There is no disk to reformat and when I find it in device manager it says "This device cannot start. (Code 10) {Operation Failed}
The requested operation was unsuccessful." This is NOT my device simply running out of writes. This device is barely used and didn't show any sign of failure until now. Can you help me recover it so I can use it again? I closed Rufus thinking I could reformat it in Disk Management before I realized it wasn't even recognized. I'm just stuck now.

Log
---
```
<FULL LOG>
```

pbatard, 2020-10-16:
Most likely what happened is that your drive failed and that is the reason why Rufus was stuck after the download (and not the only way around, with Rufus "somehow" causing a hardware failure, which is [not something that Rufus can do](https://github.com/pbatard/rufus/wiki/FAQ#Help_Rufus_damaged_my_flash_drive))

And yes, even new devices can fail since the flash memory used in USB drives is a lot more unreliable than people realize. This happens regardless of the application being used and just for the record, there really is nothing fancy when it comes to repartitioning or formatting a drive which means that, no, there is no "critical data blocks" on a flash device that an application like Rufus could access and that would render the drive inoperable if a write was cancelled or garbage was written onto it.

A lot of people seem to reason that _"If application X didn't work as expected and the the drive failed, then it must mean that application X caused the failure"_ whereas it's really the other way around, i.e. _"If a drive fails while application X is trying to use it, then of course application X will not work as expected"_.

This means that, if your drive is unrecognised everywhere, there's not much that can be done because it basically amounts to the flash controller of your device declaring that the flash memory is dead. You can try to see if a Linux machine is able to see it (there are some weird Windows bugs that can prevent a drive from being accessed, but what you describe doesn't look like one of these), but if not, you will just have to recycle it.

CyberFox2795, 2020-10-17:
Damn. Still find it odd. I've been using it with 0 issue until this. Is it possible the drivers got corrupted? Its in device manager it just says there was an error. I'll try a linux machine but I don't believe its dead. I'm not trying to blame Rufus. If anything it's probably some windows shenanigans. I'm just looking for help from some technically minded people.

CyberFox2795, 2020-10-17:
After some more research and testing of the device I found that whatever the issue is windows will temporarily see the device as a generic USB then it will vanish from device manager. I can only conclude that you were right in that the device has simply given out for one reason or another. You ARE correct that software like Rufus cannot damage hardware or critical firmware and I am not blaming Rufus for this. I think the device has simply given up or windows is throwing a fit. Thank you for putting up with the community and continuing to make this software.

closed 2020-10-17 by CyberFox2795