pbatard/rufus #1640: Error: GUID drive could not be created.

EntenKoeniq, 2020-10-16:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I received an error message that "explorer.exe" was running and should be closed first, but after a few tries and restarting my computer, it did not help. At some point I got the following error message "Error: GUID drive could not be created."

I am trying to install the Windows 10 ISO on my WORKING USB stick using RUFUS.
It works with version 3.11.

Log
---
```
Rufus x86 v3.12.1710
Windows version: Windows 10 64-bit (Build 19042.508)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0407 (de-DE)
Will use default UI locale 0x0407
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'de-DE'
Found USB 2.0 device 'SanDisk Cruzer Fit USB Device' (0781:5571)
Found card reader device 'SDHC Card'
Removing D: from the list: This is the disk from which Rufus is running!
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1922, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {36DC16BE-6255-40E4-B0E3-CF62DA6B85D9}
Max parts: 128, Start Offset: 17408, Usable = 15816686080 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Basic data partition'
  ID: {1935C3A7-7CCB-4339-B962-05FA455DFB48}
  Size: 14.7 GB (15814623232 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'CCCOMA_X64FRE_DE-DE_DV9'
  Size: 5.8 GB (Projected)
  Has a >4GB file
  Uses: EFI
  Uses: Bootmgr (BIOS and UEFI)
  Uses: Install.wim (version 0.13.1)
Using image: Windows10_InsiderPreview_Client_x64_de-de_19042.iso (5.9 GB)

Format operation started
Requesting disk access...
Will use 'F:' as volume mountpoint
Deleting ALL partition(s) from disk '\\?\PhysicalDrive2':
● Partition 1 (offset: 1048576, size: 14.7 GB)
Opened \\.\PhysicalDrive2 for exclusive write access
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Partitioning (GPT)...
● Creating Main Data Partition (offset: 1048576, size: 14.7 GB)
● Creating UEFI:NTFS Partition (offset: 15815617024, size: 512 KB)
Writing UEFI:NTFS data...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (NTFS)...
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Format completed.
Writing Master Boot Record...
Using Rufus protective MBR
Writing protective message SBR
Found volume \\?\Volume{2c80c576-0fb7-11eb-81bd-1c697a6806e4}\
F:\ is mounted, but volume GUID doesn't match:
  expected \\?\Volume{2c80c576-0fb7-11eb-81bd-1c697a6806e4}\, got \\?\Volume{2c80c577-0fb7-11eb-81bd-1c697a6806e4}\
Retrying after dismount...
Could not remount \\?\Volume{2c80c576-0fb7-11eb-81bd-1c697a6806e4}\ as F: Windows error code 0x00000057 (FormatMessage error code 0x00003AFC)

Found USB 2.0 device 'SanDisk Cruzer Fit USB Device' (0781:5571)
Found card reader device 'SDHC Card'
Removing D: from the list: This is the disk from which Rufus is running!
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1922, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {8F143EB7-2FB2-4D12-A21C-824EC113B061}
Max parts: 128, Start Offset: 17408, Usable = 15816686080 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  ID: {76B024C5-32ED-4498-9D5F-2B213F70A96A}
  Size: 14.7 GB (15814568448 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2 (UEFI:NTFS):
  Type: Microsoft Basic Data Partition
  Name: 'UEFI:NTFS'
  ID: {3649300A-EA5C-4E87-BD04-CFD5D80AE3AE}
  Size: 512 KB (524288 bytes)
  Start Sector: 30889877, Attributes: 0x0000000000000000
```

EntenKoeniq, 2020-10-16:
I deleted all partitions with "Diskpart" and without creating new partitions, Rufus works!

pbatard, 2020-10-16:
Yes, this is a consequence of trying to work around a Windows bug in ba406843f40d1a2e6af4b19d36fcfd37ea4a8fd0 where Windows does not return the correct drive letter, and where I had to remove the code that handles GUID switching (which is what happens here).

Depending on how many people are faced with this issue I may produce a bugfix release, but the thing that worries me is that if I revert ba406843f40d1a2e6af4b19d36fcfd37ea4a8fd0 then some people may end up with data being copied to the wrong drive (though, in the lifetime of Rufus, one person reported that specific issue).

wdaviscbrokerz, 2020-10-18:
Hola amigo buenas noches al aplicar RUFUS  para recuperar un USB DE 64 G, inicialmente corre bien pero casi al fina me arroja un ERROR
 EN ESTADO 
DICE ... FALLO AL REALIZAR LA OPERACION
LUEGO SE ABRE UNA VENTANA ... ERROR NO SE PUDO MONTAR EL VOLUMEN GUID
[image]

COMO PUEDO RESOLVER ESTE ERROR ... NECESITO FORMATEAR MI USB

pbatard, 2020-10-18:
If you are affected by this, you should try using Rufus 3.11 (which can be downloaded [here](https://rufus.ie/downloads/).

ptlambert, 2020-10-30:
I was seeing the same problem as the OP here, but nothing in DiskPart / Disk Management / GhostBuster / USB Oblivion was helping for a workaround.  However [Rufus 3.13 ALPHA](https://rufus-web.akeo.ie/testing/rufus-3.13_ALPHA.exe) definitely solves this problem for me.  Thanks!

pbatard, 2020-11-14:
For anybody still facing this problem, please try with [Rufus 3.13 BETA](https://rufus.ie/downloads/) and let me know if you are still seeing issues.

pbatard, 2020-11-20:
Closing on account that this should be fixed in the 3.13 release

closed 2020-11-20 by pbatard