pbatard/rufus #1641: Can't write, gets stuck on the same file everytime.

TheMossyMinerYT, 2020-10-24:
﻿

Checklist
---------
- [✓] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [✓] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [✓ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [✓ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Hi! I'm trying to write ubuntu to my usb, and it's been stuck on one single file for the last hour, just wondering why! 
`Rufus x86 v3.12.1710 (Portable)
Windows version: Windows 10 64-bit (Build 19041.572)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0809 (en-GB)
Will use default UI locale 0x0809
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 2.0 device 'Verbatim STORE N GO USB Device' (18A5:0302)
1 device found
Disk type: Removable, Disk size: 3.9GB, Sector size: 512 bytes
Cylinders: 471, Tracks per cylinder: 255, Sectors per track: 63
Partition type: SFD (Super Floppy Disk) or unpartitioned
1 device found
Disk type: Removable, Disk size: 3.9GB, Sector size: 512 bytes
Cylinders: 471, Tracks per cylinder: 255, Sectors per track: 63
Partition type: SFD (Super Floppy Disk) or unpartitioned
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.04/20191223 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'Ubuntu 20.04.1 LTS amd64'
  Size: 2.6 GB (Projected)
  Uses: Syslinux/Isolinux v6.04
  Uses: EFI
  Note: This ISO uses symbolic links, which will not be replicated due to file system
  limitations. Because of this, some features from this image may not work...
Using image: ubuntu-20.04.1-desktop-amd64.iso (2.6 GB)
Downloading https://rufus.ie/files/syslinux-6.04/20191223/ldlinux.sys
Unable to access file: 404
Extended version was not found, trying main version...
Downloading https://rufus.ie/files/syslinux-6.04/ldlinux.sys
File length: 66.5 KB
Successfully downloaded 'ldlinux.sys'
Download signature is valid ✓
Downloading https://rufus.ie/files/syslinux-6.04/20191223/ldlinux.bss
Unable to access file: 404
Extended version was not found, trying main version...
Downloading https://rufus.ie/files/syslinux-6.04/ldlinux.bss
File length: 512 bytes
Successfully downloaded 'ldlinux.bss'
Download signature is valid ✓

Format operation started
Requesting disk access...
Will use 'E:' as volume mountpoint
[... 331 lines cut ...]
Extracting: E:\boot\grub\x86_64-efi\usb_keyboard.mod (5.8 KB)
Extracting: E:\boot\grub\x86_64-efi\verifiers.mod (3.6 KB)
Extracting: E:\boot\grub\x86_64-efi\video.lst (41 bytes)
Extracting: E:\boot\grub\x86_64-efi\video.mod (8.9 KB)
Extracting: E:\boot\grub\x86_64-efi\videoinfo.mod (5.3 KB)
Extracting: E:\boot\grub\x86_64-efi\videotest.mod (5.5 KB)
Extracting: E:\boot\grub\x86_64-efi\videotest_checksum.mod (3.8 KB)
Extracting: E:\boot\grub\x86_64-efi\video_bochs.mod (8.4 KB)
Extracting: E:\boot\grub\x86_64-efi\video_cirrus.mod (9.0 KB)
Extracting: E:\boot\grub\x86_64-efi\video_colors.mod (9.9 KB)
Extracting: E:\boot\grub\x86_64-efi\video_fb.mod (27.8 KB)
Extracting: E:\boot\grub\x86_64-efi\wrmsr.mod (2.4 KB)
Extracting: E:\boot\grub\x86_64-efi\xfs.mod (10.3 KB)
Extracting: E:\boot\grub\x86_64-efi\xnu.mod (41.1 KB)
Extracting: E:\boot\grub\x86_64-efi\xnu_uuid.mod (3.3 KB)
Extracting: E:\boot\grub\x86_64-efi\xnu_uuid_test.mod (3.4 KB)
Extracting: E:\boot\grub\x86_64-efi\xzio.mod (19.4 KB)
Extracting: E:\boot\grub\x86_64-efi\zfscrypt.mod (8.4 KB)
Extracting: E:\boot\grub\x86_64-efi\zstd.mod (78.1 KB)
Extracting: E:\casper\filesystem.manifest (55.4 KB)
Extracting: E:\casper\filesystem.manifest-minimal-remove (2.8 KB)
Extracting: E:\casper\filesystem.manifest-remove (3.7 KB)
Extracting: E:\casper\filesystem.size (11 bytes)
Extracting: E:\casper\filesystem.squashfs (2.0 GB)
`

pbatard, 2020-10-24:
Looks like your USB drive might be defective, especially as you seem to be getting errors. Either that or it is very slow for some reason.

At any rate, this is a problem that's external to Rufus, so there's not much I can do to help.

I would cancel the operation and run a bad blocks check. But more appropriately, I would try with the different USB drive.

pbatard, 2020-10-26:
No update &rarr; Closing.

closed 2020-10-26 by pbatard