pbatard/rufus #1642: How do a UEFI OS Loader be loaded into memory?

cpktnwt056, 2020-10-24:
As topic, the question is how do a UEFI OS Loader be loaded into memory? After I have read many articles and papers,I still don't understand how a UEFI OS Loader is loaded into memory. Can you give me a full description about how a UEFI OS Loader is loaded into memory?

pbatard, 2020-10-24:
Sorry, I don't do tutorial requests. I wish I had the time, but I really don't, especially for "full description".

Basically, the UEFI firmware looks for all devices, find the ones that match a relatively complex set of rules in order to decide whether they should be bootable or not (E.g. Does it use a file system UEFI can handle? If yes, does it contain a file that matches the set of bootloader naming conventions, like `/efi/boot/bootx64.efi`? If yes, and Secure Boot is enabled, does its signature fullfill the Secure Boot requirements? etc.) and then loads that bootloader to memory and executes it.

This is described in the [UEFI specs](https://uefi.org/sites/default/files/resources/UEFI%20Spec%202.8B%20May%202020.pdf) if you look for the many chapters that pertain to the boot process.

closed 2020-10-24 by pbatard

cpktnwt056, 2020-10-25:
Do you know anyone skilled in the knowledge about UEFI? Can you give me his/her contact method?