pbatard/rufus #1643: [0x80070032] The request is not supported - Manjaro

Den2908, 2020-10-24:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>

Log
---
```
Rufus x86 v3.12.1710 (Portable)
Windows version: Windows 10 64-bit (Build 19042.572)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0419 (ru-RU)
Will use default UI locale 0x0419
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'ru-RU'
Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
Found USB 2.0 device 'USB DISK 2.0 USB Device' (13FE:4200)
1 device found
Disk type: Removable, Disk size: 8 Гб, Sector size: 512 bytes
Cylinders: 974, Tracks per cylinder: 255, Sectors per track: 63
Partition type: SFD (Super Floppy Disk) or unpartitioned
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Detected Grub version: 2.04~manjaro
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'MANJARO_KDE_2012'
  Size: 2.9 Гб (Projected)
  Uses: GRUB2
  Uses: EFI
Note: ISO mode will be disabled because this ISOHybrid is not compatible with ISO boot.
Using image: manjaro-kde-20.1.2-201019-linux58.iso (2.9 Гб)

Format operation started
Requesting disk access...
Will use 'F:' as volume mountpoint
Deleting ALL partition(s) from disk '\\?\PhysicalDrive3':
● Partition 1 (offset: 0, size: 7.5 Гб)
Could not delete partitions: [0x80070032] The request is not supported.
Opened \\.\PhysicalDrive3 for exclusive write access
I/O boundary checks disabled
Writing Image...
Remounted \\?\Volume{f25c9f14-163a-11eb-85f7-309c23058bf6}\ as F:
\\?\Volume{f25c9f14-163a-11eb-85f7-309c23058bf6}\ is already mounted as F:
Re-mounted volume as F: after error

Found USB 2.0 device 'USB DISK 2.0 USB Device' (13FE:4200)
1 device found
Disk type: Removable, Disk size: 8 Гб, Sector size: 512 bytes
Cylinders: 974, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000000
Drive has an unknown Master Boot Record
Partition 2:
  Type: EFI FAT (0xef)
  Size: 4 MB (4194304 bytes)
  Start Sector: 6013888, Boot: No
Found USB 2.0 device 'USB DISK 2.0 USB Device' (13FE:4200)
1 device found
Disk type: Removable, Disk size: 8 Гб, Sector size: 512 bytes
Cylinders: 974, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000000
Drive has an unknown Master Boot Record
Partition 2:
  Type: EFI FAT (0xef)
  Size: 4 MB (4194304 bytes)
  Start Sector: 6013888, Boot: No

pbatard, 2020-10-24:
Try to unplug/replug the drive, or zero it first with <kbd>Alt</kbd>-<kbd>Z</kbd> (you can cancel the zeroing if it takes too long).

Do you see that issue consistently, or was it a one-of error?

Den2908, 2020-10-24:
All the time, but the problem was solved with the help of disk cleanup diskpart. I tried before to switch the flash drive, format via properties. I spent 2 hours of time on all of this. I eventually decided to create a theme on github. Then I remembered diskpart. Lol.

pbatard, 2020-10-25:
Well, I tested `manjaro-kde-20.1.2-201019-linux58.iso` with Rufus 3.11 and saw no errors.

Since you managed to sort your problem, I will close this issue.

closed 2020-10-25 by pbatard