pbatard/rufus #1644: Question on the numbers shown during bad blocks checking

RinMinase, 2020-10-25:
﻿

Checklist
---------
- [x ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.

Not Applicable to my question:
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
A question, since I am unable to find these stated anywhere I looked.
I just wanted to know for pure information only.

I ran bad blocks checking and it stated `#/#/# errors`.

My questions are:
1. Why are there 3 numbers indicated?
2. What does each of the number mean / correspond?

RinMinase, 2020-10-25:
I'll be closing this issue as I found it at the end of bad blocks checking.

If this is not stated in the documentation, `A/B/C errors` during checking mean:
`A` - Read error(s)
`B` - Write error(s)
`C` - Corruption error(s)

I get what `read` and `write` errors mean when checking the drive.
As to what `corruption` errors mean and on how they are checked, that I do not know.

If anyone can comment below on this, it would be very much appreciated. 😁

closed 2020-10-25 by RinMinase

pbatard, 2020-10-25:
* Read error: Can't read the specific block on the media
* Write error: Can't write the specific block on the media
* Corruption error: **Can** read and write the specific block on the media, but the data read doesn't match the data written (i.e. the media is corrupting the data).

Those are the 3 fairly standard errors from any data validation utility.