pbatard/rufus #1647: Create Windows installations Without UEFI:NTFS partition & wimsplit

AkitoTheGambler, 2020-11-04:
Issue description
-----------------

Hi, pls make it possible to have a checkbox to disable the creation if the UEFI:NTFS partition (for systems that already support ntfs boot) and also one to split the wim file to allow the installation of Windows on older devices with Secure boot on but without UEFI:NTFS boot support

pbatard, 2020-11-04:
This gets asked on a semi regular basis, but since the solution you propose (as well as the one that requires one FAT32 partition for everything but `install.wim` and one NTFS partition for `install.wim`) has a lot of drawbacks compared to UEFI:NTFS (splitting files is very costly, compared to simply copying them wholesale into a single partition) **and** there does not exist a single x86 based system where you cannot disable Secure Boot **and** the fact that [temporarily disabling Secure Boot is not a real security liability](https://github.com/pbatard/rufus/wiki/FAQ#Why_do_I_need_to_disable_Secure_Boot_to_use_UEFINTFS) **and** that Secure Boot proponents (i.e. those who don't want to disable it ever) curiously don't seem to actually be bothered enough to complain to Microsoft that they are abusing their power by forbidding anything GPLv3 from being signed for Secure Boot (which is the **ONE AND ONLY** reason you have to disable Secure Boot at all when using UEFI:NTFS. If Microsoft allowed GPLv3 software to be signed for Secure Boot, I'd have a Secure Boot compliant UEFI:NTFS in a jiffy), I have absolutely no plan to add what is effectively a lesser solution to Rufus, when UEFI:NTFS does remain the much better approach.

Oh, and since you mention _"systems that already support ntfs boot"_ then you'll be pleased to know that, on such systems, you can use an UEFI:NTFS drive as created by Rufus **and not have to disable Secure Boot at all**, since your system will happily boot from the first partition (another reason why UEFI:NTFS is the better solution, because it does cover systems that have an NTFS driver, and doesn't require to go through unwarranted splitting).

So, for all the _"But why should I have to disable Secure Boot? It's annoying!"_ fanatics out there, you're going to have to come up with better excuses as to why UEFI:NTFS can't fit your needs, because I've heard them all, and not one of them makes an even remotely compelling argument.

Still, if you want to be annoyed at somebody for not giving you what you want, then be annoyed at Microsoft and let them know, because UEFI:NTFS could easily be made compatible with Secure Boot **IF** Microsoft weren't abusing their power by refusing to sign GPLv3 software.

&rarr; Rejected.

closed 2020-11-04 by pbatard

AkitoTheGambler, 2020-11-06:
Im aware its not a security problem but users often call because of the
error message secure boot violation and so im not saying it should be set
as default but it would be very useful to have as a advanced option and as
an advanced option to disable the creation of the rufus uefi:ntfs if we
already know the oem has a uefi ntfs driver just to avoid the secureboot
violation message and there are a lot of users who dont know how to go into
buos etc and because of covid we cannot go to the customers.and doing bios
settings via phone is very difficult if the user doesn't know how
Im very sure I'm not the only one with this issue and there are a lot of
others with this problem just like there are a lot of people who are not IT
professionals but want to install windows

AkitoTheGambler, 2020-11-06:
And as "excuse" its more for having a possibility to avoid it for pcs that
dont support it and for users who are not pros  with pcs and changing bios
its extremely time consuming to do over phone because of covid especially
when you have to do 20 different users and more in the future, and ik im
not the only one with this issue

So yeah its more just to have the possibility as an advanced setting to
split wim or put wim on ntfs and everything else in fat32

Yes maybe there are performance issue but its still less time consuming
than explaining users how to disable and re-enable secure boot

pbatard, 2020-11-06:
> but users often call because of the error message secure boot violation

Then you just have to explain that they need to temporarily disable Secure Boot, which, actually, Rufus has a VERY EXPLICIT message about, that you can't miss, when you create a UEFI:NTFS drive, that points to the FAQ and to links that will help users figure out how to disable Secure Boot. So, if users call, it's because they clearly haven't paid attention to what Rufus was telling them.

> there are a lot of users who dont know how to go into buos etc

I'm going to be harsh here, but these users shouldn't be using Rufus to install an OS, because, and this always seems to surprise a lot of people who think that the application should be designed in such a manner that a toddler should be able to use it, it is actually designed for people who have a minimum a technical proficiency. The reason for that is that installing an OS is __not__ a trivial operation, and treating it as something that everybody and their grandmother should be able to do is not the way to go.

If you disagree with that, then I will kindly ask that you use a different application, because Rufus was not and will never be designed so that people can avoid having to __learn__ even the bare minimum about booting a computer. If I had designed Rufus to include such users, it would look very differently.

> because of covid we cannot go to the customers.and doing bios settings via phone is very difficult if the user doesn't know how

I think you're always going to have other problems if you're trying to support OS installation by phone, so the BIOS issue (which is usually not as complicated as you indicate, if the user can actually tell you the computer model they have, especially in this day an age where most user manuals are available on the internet), so I find it difficult to buy that I should add what you suggest because of COVID.

> Im very sure I'm not the only one with this issue and there are a lot of others with this problem

As the author of the application, who sees the amount of requests I actually get for this, I'm very sure that there aren't that many people with this problem, and also, I am 100% certain that most of these requests are bullshit excuse for people who irrationally scream at the idea of ever disabling Secure Boot (because it says "Secure") and who, and this was my earlier point, who mysteriously seem to stop caring about it altogether, whereas they are adamant that Rufus should allow them not to disable Secure Boot, when I tell them that they can very much have that, **if** they complain to Microsoft about their own irrational policy about not signing GPLv3 binaries. Because, the minute Microsoft allows GPLv3 binaries to be signed, the minute you get a Secure Boot friendly of UEFI:NTFS in Rufus. So, if people are that inconvenienced by it, why on earth aren't they using that energy to tell Microsoft that their policy of not signing GPLV3 binaries is hurting them. I mean, if you think that the extra difficulties in disabling Secure Boot due to COVID is a good justification as to why I should add the feature you request to Rufus, then surely, the same justification should work with Microsoft. Or are you not contacting Microsoft because you don't believe in what you are putting forward when you state that it's actually preventing a lot of people from installing Windows?

Again, the issue is not that UEFI:NTFS can't be made compatible with Secure Boot. It easily can, and you could have a version of Rufus where you don't have to disable Secure Boot ever, **if** you were to do the RIGHT thing and let Microsoft know that their policy of not signing GPLv3 code is bullshit and explain to them how it is penalizing you.

>  its more for having a possibility to avoid it for pcs that dont support it 

Please create an issue report for PC that don't appear to support UEFI:NTFS if you see them. As this and the UEFI:NTFS issue tracker demonstrates, I am often willing to go out of my way to troubleshoot UEFI:NTFS reports to ensure that it works on as many PCs as possible. I actually purchased a couple of PCs, that were reported as problematic, just to do just that, and make sure UEFI:NTFS would work on those.

Or, if your "support it" is about disabling Secure Boot, then, as I mentioned earlier, there is no a single x86 based PC that I know of right now, that doesn't support disabling Secure Boot.

> Yes maybe there are performance issue

There are. MASSIVE ones, coz, if you've ever tried that, you'll see that splitting a WIM is a very slow operation.

And I haven't even started on mentioning that this pretty much excludes Windows 7 users (which I still actively support, since Rufus is an application that allows then to move **away** from their obsolete OS), since Windows 7 can't mount an ISO, so Rufus would first have to extract a 4 GB `install.wim` onto the host computer (provided that the user has enough space) and, after it added an already time consuming file copy operation, split that file. But you;'re going to tell me that I should only make that option available for Windows 8 or later users, just like Windows To Go, but if you do that, then I'm going to start to wonder how committed you are to the idea that you need to support a wide variety of people remotely, that aren't tech savvy enough and need to media creation utility to do everything for them. Or are you mysteriously only supporting phone users that don't use Windows 7 at all (which o [... cut ...]
If you are inconvenienced by having to temporarily disable Secure Boot on your machine, I'm afraid you're still going to have to do better than that, because I'm far from being convinced by any of the "arguments" you (and others) have desperately tried to put forward.

And there again, I have to stress out once more that you could easily get the "problem" solved if, instead of complaining to me, you complained to Microsoft about their Secure Boot signing policy. Get Microsoft to change their irrational stance, and I'll happily sign UEFI:NTFS for Secure Boot, so that you won't have to bother spending 30 seconds disabling and re-enabling it when installing Windows.

AkitoTheGambler, 2020-11-07:
we have to use customized versions of the iso for the users of our customer
companies, if i could use the media creation tool i would but we cant use
the default iso from MS media creation tool
and i was just saying having the option like chill i got it you dont wanna
do it alright
already told MS but they wont budge on that GPLv3 thing for now

pbatard, 2020-11-07:
> it doesnt have to be necesserily wim splitting it can also be 1 fat partition with everything and only the wim file on ntfs, then you dont have the compatiblity problem with windows 7

Oh but you do. Because until Windows 10 1703, you cannot mount 2 partitions at the same time for removable media (99% of USB Flash Drives out there). So that makes the 1 FAT32 partition + 1 NTFS partition solution a major pain in the ass to add to Rufus for any non Windows 10 users.

That's actually the reason why UEFI:NTFS is the __much better__ solution, because I'm just flat copying the 512 KB FAT partition wholesale (i.e. I don't have to rely on a system to be able to mount that partition), which works for all versions of Windows.

Please bear in mind that it's a developer's job to analyse the various options and, in that process, it's exceedingly common to find that something that looks good on paper has a major flaw that makes it very unattractive as a solution. Considering that UEFI:NTFS required a lot of effort (because I basically had to produce a UEFI NTFS driver, and therefore had to create a whole new [EfiFs](https://github.com/pbatard/efifs) project in the process), you can bet that I did carefully analyse the various solutions, and settled on the one that looked objectively better (which, despite the need to __temporarily__ disable Secure Boot, I maintain UEFI:NTFS still is to this day).

Unfortunately, a lot of folks come to a developer with only the _"Well, this looks good on paper"_ part and do not spend much time, if any at all, on the _"Alright, now where would this seemingly great solution unravel"_ part.

> i got it you dont wanna do it alright

It's not that I don't want to do it. It's that the solutions you advocate are less efficient than UEFI:NTFS, no matter how big you think temporarily disabling Secure Boot is (which, IMO, is actually a good skill for users of a computer to acquire, unless they fell that relinquishing control of the machine they own to a remote corporation, that the becomes the judge, executor and jury of what they can and cannot run is a good idea).

> already told MS but they wont budge on that GPLv3 thing for now

Which is precisely why we need more and more people to complain to Microsoft that it's hindering their workflow when that is the case.

AkitoTheGambler, 2020-11-07:
Oh but you do. Because until Windows 10 1703, you cannot mount 2 partitions at the same time for removable media (99% of USB Flash Drives out there). So that makes the 1 FAT32 partition + 1 NTFS partition solution a major pain in the ass to add to Rufus for any non Windows 10 users.

i see, well im not sure if it would work but lets says it creates first an ntfs partition with the correct size at the end of the usbdrive, it moves the wim file on it, then it then it creates the fat32 partition as first partition, then yeah windows would not be able to mount both so it will hopefully mount the first partition in the usb drive, then it copies all the other files. after that, you have your usb drive, you cant access the second drive anymore in windows 7 but at this point you wouldnt need to access it anymore anyway, do you think that would work? and as i said its not to set it as default option its an advanced alternative setting

idk how windows behaves when adding a second partition to usb drive but if it will always only mounts the first partition like i think it could work even if less performant, might be able to test it with windows 7 and a 3rd party partitioning software

pbatard, 2020-11-07:
> but lets says it creates first an ntfs partition with the correct size (...)

Yes, __of course__ I thought about doing that, and that is why I say it's a pain in the ass. It'd doable, but it adds a lot of extra complexity (that needs to be __TESTED__ and __MAINTAINED__ separately) just to add older platforms whereas UEFI:NTFS adds literally none, as it's just creating an extra partition at the end, which we don't care about mounting at all, and flat copying 512 KB, which is pretty much instant. Oh, and considering the trouble I have with ensuring that partitions get properly mounted where and when expected with Windows, which I'm only starting to resolve in the next version, even if adding the code and maintaining it was peanuts (which it isn't), I see just way too possibilities of Windows to screw this up on Pre 1703 systems if you're going to be shuffling mountpoints around.

Again, solutions that look good on paper and easy to add only tend to do so until you actually have to implement them... And it takes developer experience, and a special kind of having had to fight against the unintuitive behaviour of Windows and it's API, to come to the conclusion that it's usually the solution that has the less requirements on Windows to be cooperative that will work best in practice.

Also, at this stage, i really think you should not be using Rufus but deploying your own solution, as, if you are running a post 1703 machine, a good script that calls on diskpart to create the partitions, and 7-zip or something to extract the ISO, will be a much better fit to your workflow than a GUI application that was never designed for the purpose you would like it to accomplish.

ThiloL, 2020-11-27:
It is annoying, having to disable secure boot before each installation and enable afterwards. 
Often BIOS/UEFI is password locked, so switching off and on secure boot is not possible.
A few minutes more creating the bootable stick or five minutes longer installation is not worth that pain.
@pbatard Is there really no argument to add wmi split feature?

pbatard, 2020-11-27:
>  Is there really no argument to add wmi split feature?

None. If your BIOS/UEFI is password locked, and you don't know the password, then it probably means you shouldn't be trying to install Windows on that machine, so, once again, you'll have to find a better argument than that.

Again, if you ACTUALLY DO CARE about this, you WILL complain to Microsoft for their abusive "no GPLv3 software can be signed for Secure Boot" policy because I can only consider that people complaining to me instead of complaining to Microsoft don't care as much about Secure Boot as they'd like to pretend. If Microsoft allowed GPLv3 binaries to be signed, like they allow GPLv2 ones, you wouldn't have this annoyance as I'd have made UEFI:NTFS Secure Boot compatible a long time ago. But Microsoft ARE ABSOLUTELY ABUSING THEIR POSITION in rejecting GPLv3 software for Secure Boot signing, and, instead of complaining to Microsoft, people who allegedly care about Secure Boot somehow chose to complain to the one person who is trying to tell them that some of their software freedoms are being taken away by a large corporation and that, if they don't want more of this to happen, they should make a stand for it.

So that's the matter you should be taking on, instead of the **minor** annoyance of having to go to your UEFI settings when installing Windows (knowing, once again, that you only need to temporarily disable Secure Boot, and that this will not prevent you at all from running Windows with Secure Boot enabled once it's installed).

GalaxyNetworks, 2021-02-01:
I've used Rufus a lot but even if I often disable secure boot for other reasons, I'm not going to create a bootable drive that requires secure boot to be disabled... That's just not something that should be required and there are other ways to do it.

For a long time I've deployed Windows using DISM but now I needed a new bootable USB with Windows Server 2019...
It's really not that hard to run "dism /Split-Image" so there's absolutely no reason not to let people choose FAT32 for what ever reason they may have...

Disabling "Secure Boot" may be a minor annoyance for some, but not for everyone. Maybe we should complain to MS, but you're the one who's limiting us when there's a perfectly viable way to do it. A way you've simply chosen not to support, but that's okay, it's your software, but understand that people have a right to be disappointed with your limited support!

I guess it's all my fault for downloading Rufus instead of just making a PowerShell script. One thing you need to know though! Not everyone who may use Rufus are comfortable with adjusting UEFI/BIOS settings, it's a simple tool and it's easy to follow guides, but I'd never ask someone to disable secure boot, even if it doesn't put them at a security risk.

pbatard, 2021-02-01:
> It's really not that hard to run "dism /Split-Image"

You know, I'm growing tired of having to re-explain the same thing again for people who conveniently appear not to consider the full picture or be open to the idea that their solution **actually has major flaws**.

I've already explained that over and over, to every proponent of the "just split the image": To be able to achieve that, you must be able to mount the ISO (because `dism` will not split a file that resides into an ISO file), **which makes this approach completely useless for Windows 7 users** (because native mounting of ISOs was only introduced as a feature in Windows 8).

But hey, feel free to tell me how I should just say _"screw you"_ to all the Windows 7 users out there who might precisely be trying to use Rufus to install Windows 10 and stop using a version of Windows that is no longer supported.

And there's also the matter of speed between just flat copying a file or relying on `dism` usually atrocious throughput when processing anything (which, curiously, the proponents of the "just split the image" always seem to gloss over).

> but you're the one who's limiting us

Wow, that really takes the cake.

Somehow, I have become responsible for a downright user-unfriendly licence limitation, that Microsoft introduced, and that I had absolutely no say on.

Again, if you are really that annoyed about disabling Secure Boot, make your voice heard to Microsoft, because I am not the one who's limiting you here.

I did devise the **much** better technical solution to all the alternatives (no time wasted in splitting, no need to mount an ISO on system that might not have mount capabilities, no need to mount 2 partitions on systems that might not be able to mount 2 partitions on a removable drive, a.k.a. pre Windows 10 1703), and the **only** reason you are not able to use it in the way you would like (i.e. without having to disable Secure Boot) is **ENTIRELY** on account of Microsoft abusing their position because they forced a choice that goes against everybody's freedoms. But, somehow, **I** am the person limiting you? In what universe does that make any shred of logical sense?

Maybe you can live with seeing your freedoms abused by people who have the power to do so. But please don't try to put the blame on the very people who aren't happy with that, and are doing what they can so that this kind of abuse doesn't become the norm, **for your and everybody else's sake**.

_"Those who would give up essential freedom, for the sake of convenience, deserve neither freedom nor convenience."_

I am not the one limiting you here. Microsoft are. As I've explained many many times, there is no actual legal or technical justification to Microsoft preventing GPLv3 being used for Secure Boot binaries. So please pick the right person(s) to complain to if you want things to change.

GalaxyNetworks, 2021-02-01:
I'm not saying you should say screw you to any, but giving people options is a good thing, right?
You are limiting us by not making it an option, and that's okay, but as people clearly wish for this feature, I really dislike your view of simply blaming MS - IN ALL CAPS. Are they to blame? Sure! But do you really think they care?

It's perfectly okay that you don't want to support FAT32 and I'm not denying that your UEFI:NTFS boot option is better than being limited by FAT32. But in my case I'd much rather spend a few minutes longer on splitting the install.wim, even though that progress only takes seconds if done in memory or on SSD/NVMe storage before copying to the USB partition.

If I tell a client to disable Secure Boot, then they'd most likely start questioning my intentions. Trust me, just because a company have hired an IT-guy to manage their systems, that doesn't guarantee he/she knows anything about it.
I don't mind advising them how to boot from the USB, but to disable Secure Boot, even temporarily, then I'd in some way be liable if they fail to enable it again. They may not legally have anything against me, but even if I tell them a million times to enable Secure Boot again, it would still be me who initially told them to disable it. That's just an unnecessary risk I don't want to take.

And trust me! I do understand your hate against MS! I just wish for an option that doesn't require the user to disable Secure Boot, that's all.

pbatard, 2021-02-01:
> but giving people options is a good thing, right?

Not when one has to develop the same feature, **twice**, for no good reason.
My time is not unlimited and the product is free, so you do the math about my willingness to invent the wheel two times in a row. If I feel I can improve on the wheel, I sometimes don't have a problem reinventing it. But I'm the one who invited the wheel in the first place here, and, regardless of cavemen that come with a policy that states that I am not allowed to bring my chariot to their section of the cave unless it uses the same square wheels as theirs, my wheel is still the obvious superior product. 

> Are they to blame? Sure! But do you really think they care?

Well, they are certainly not going to care much if, instead of complaining directly to them about what should be labelled as a blatant abuse policy, people decide to complain to people who are opposed to that policy, because the law of the strongest should prevail.

> But in my case I'd much rather spend a few minutes longer on splitting the install.wim

Then do that. You have access to `dism` and you are clearly indicating that you don't care that much about the extra time it should take you. So there's nothing preventing you from formatting a drive to FAT32 and then performing the splitting steps manually if you are that adamant that Secure Boot shouldn't be disabled even for one second. Plus, if you are dealing with IT pros, they're going to have no problem creating bootable drives themselves using command line tools instead of third party utilities, especially if they are that paranoid about security.

So please take a moment to consider that you are asking me to invest time, that I don't have, re-implementing a feature that I already implemented, at a cost that is going to go way beyond the "5 minutes job" that you seem to believe it is, on acocunt that **you** don't seem to want to have to invest your own time performing something that isn't that difficult to accomplish manually (and that, in this instance, is pretty much a 5 minutes job in terms of mounting a drive, formatting another drive, figuring out the dism command and performing the rest of the file copy).

You see the problem here? Why should I have to invest **weeks** of my time (because that's what I will take to do things properly, especially as I'm going to have to ask users what method they should use, and all the other things that people who propose solutions that they don't actually have to implement, disregard as no-time, no-effort tasks) just because you don't want to invest 5 minutes of your own time or give in to the paranoia that Secure Boot should not ever be disabled, even temporarily? Where's the fair exchange there?

I've got more than enough on my plate not to have to duplicate a solution for a problem that I have already solved, properly, and, as I also mentioned previously, and since the amount of time needed to be invested would be roughly the same either way, **if I were to take action on this** then I'd rather do that in a way that will be a lot more beneficial for the world as a whole (including non Rufus users) by:
1. Crafting my own non-GPLv3 exFAT driver as part of the EDK2 (or even my own NTFS driver if I feel like a challenge, since, if I am forced into this by some abusive third party policy, I might as well derive some enjoyment from the process).
2. Switch to using that for UEFI:NTFS, and thus get UEFI:NTFS Secure Boot signed, as it's only the license of the driver (which comes from GRUB 2.0) that is a problem.

I'll take an aside on this to point out that, even though Microsoft are part of the UEFI committee and that, from being a very regular contributor to the EDK2 (which is part of the reason why my time is limited and I'd rather not have to solve the same problem twice) they have active developers contributing to the EDK2 (which is the Open Source core of UEFI), they sure don't seem even remotely intent on contributing an exFAT or NTFS there, which one can't help but find strange considering that every person that ever dabbles in UEFI knows the massive pain points of only having a FAT driver there. Had they done any of that over the many years we've had UEFI:NTFS, we certainly wouldn't be in the situation we are now.

Thus, if you don't feel like taking on Microsoft on their Secure Boot signing policy (which you should), you may also want to ask them why they still haven't contributed something that UEFI users have been requesting for years, namely: exFAT or NTFS UEFI driver.

I have tested exFAT (in fact the UEFI:NTFS that Rufus uses does include exFAT UEFI drivers alongside the NTFS ones) and the only limitation I've seen with switching to exFAT is that exFAT support for Windows installs only started with Windows 8 (i.e. Windows 7 ISOs only support NTFS for the installation media). But in this instance, since we are certainly not expecting people to install Windows 7 any longer, this is a non issue.

So, to all the people who are suggesting I do it this way or I do it that way, please bear in mind that, the one and only solution I am going to go for, if I decide that there's no hope on the Microsoft policy front (because you guys, who should be the ones letting Microsoft know that their policy is creating issues, have decided to **give up** altogether), is the development of a new non GPLv3 exFAT driver, that I can "donate" to the EDK2, so that I can continue to use the better solution, which is UEFI:NT [... cut ...] tively coerced Microsoft to run whatever you want in the first place). Or ask them when the heck they are going to contribute some resources to put an exFAT or NTFS driver in the EDK2, instead of leaving that matter to willing (or unwilling) non MS developers, who tend to have much better things to do and a lot less resources to do them with.

Right now, and as as been the case for all the years UEFI:NTFS has existed, your best option to solve the Secure Boot situation is with Microsoft. Not me.

ThiloL, 2021-02-02:
OMG

pbatard, 2021-02-02:
> OMG

Yes, that was exactly my reaction the first time I saw the no GPLv3 rule that Microsoft added to their Secure Policy signing policy...

And as long as people appear to be more offended by the views of an independent Open Source software developer than by the unfair behaviour of a large corporation, and believe that it should be reasonable for the former to just bend to the will of the latter, we're going to continue to have a problem.

Hrxn, 2021-02-02:
Just for the record, and not that my opinion really matters, but I completely agree with @pbatard here.

pbatard, 2021-02-02:
Thanks.

For the record, I'm going to point out to another example of Microsoft's anticompetitive/abusive behaviour when it comes to Secure Boot, as illustrated by the terms and conditions they enforced at https://uefi.org/revocationlistfile, and more specifically this:

> You may copy and use the UEFI Revocation List for your internal, reference purposes and to design, develop, and test your software, firmware or hardware, as applicable; and you may distribute the UEFI Revocation List to end users solely as part of the distribution of an operating system software product, or as part of the distribution of updates to an operating system product; and you may distribute the UEFI Revocation List to end users or through your distribution channels solely as embodied in a firmware product or hardware product that embodies nontrivial additional functionality.

This legalese, which was put in by the entity that the UEFI forum entrusted with the revocation list update process, i.e. Microsoft, means that, if I wanted to, say, develop easy-to-use software that grants you the freedom to install your own keys to gain full control of your computer's Secure Boot process (rather than rely on the ones installed by your hardware manufacturer, which is actually a major security risk) and allow you to sign your own UEFI binaries for Secure Boot, while obviously still preserving the ability to boot Microsoft signed bootloaders and prevent known bad ones, that software would **not** be able to download the UEFI revocation list, a critical part of the whole process, because, per Microsoft's **restriction** only manufacturers of OSes or hardware are allowed to do that.

So in effect, by adding these rather limiting terms and conditions, Microsoft is ensuring that it is not possible for users of a Secure Boot environment to be in a position where they are given the ability to decide, for themselves, what UEFI binaries they want to sign for Secure Boot, through the convenience of a software that would simplify all the required operations for them. And that is because, in order to do that, that software would need to re-install an up to date revocation list as part as its operations, which Microsoft made sure it cannot legally obtain it through a direct download from the UEFI forum...

Considering that, as any security researchers will tell you, revocation entries that are part of a public key infrastructure should be made available publicly with not strings attached (See OCSP or CRLs, which **anybody** can always access through an internet connection), it certainly does appear that, through yet another example of very unfair terms and conditions, Microsoft is following some kind of agenda to ensure that they keep as much control of the Secure Boot process as they can, by stifling the ability of third parties to provide user-friendly software that might restore some element of control of this whole process to the end user...

All this to say, it's not just the no GPLv3 condition that has me riled up against Microsoft when it comes to their Secure Boot policies. There are other factors as well. And that is also why I feel entitled to insist that we have a fundamental anticompetitive problem here, that no single person can tackle on their own, and that requires more involvement than a lone developer can address...