pbatard/rufus #1649: Suggestion: EXT3:NTFS

Yuraconst, 2020-11-07:
Rufus-EXT3 profile (if EXT3 driver installed on WIndows)
Same as UEFI:NTFS but with the bulk partition being EXT3 instead of NTFS.
The user then simply extracts the ISO into the bulk partition and extracts the ISO-bundled efi.img into the EFI-FAT partition to add the necessary files, including EXT3 driver.

pbatard, 2020-11-07:
> if EXT3 driver installed on WIndows

What ext3 driver? Is there a native freely available version of an ext3 driver for Windows that I am not aware of?
I know that Microsoft is planning to improve on ext accessibility for WSL, but I am not aware of a any native Windows ext3 driver that allows you to mount ext file systems read write outside of very expensive and non open source custom drivers.

Short of a freely available r/w ext3 driver being readily available on Windows, I am not going to add a feature to Rufus that relies on people having to pay for some proprietary non Open Source driver, as a matter of policy (and also because it would drastically reduce the population of users that could benefit from this).

> and extracts the ISO-bundled efi.img into the EFI-FAT

The EFI-FAT partition exists only for the purpose of providing the UEFI driver. It's not designed to accommodate any extra files.

Also UEFI:NTFS was designed purely with the purpose of accommodating the Windows installation process (and I don't believe you are talking about UEFI:EXT3, which should be the name you want to designate that feature as opposed to EXT3:NTFS which makes no sense, in the context of Windows installation), so I fail to see what problem you are actually trying to resolve.

For one thing, most of the distros I know, that may face the 4 GB limitation of FAT32 have added support for NTFS (e.g. Debian, Ubuntu, etc, which will happily boot from NTFS media created in Rufus), so it's not like there appears to be a real issue that only UEFI:EXT3 could solve here.

It would certainly help for you to describe exactly what problem you are trying to solve, for which specific image, because, so far, I am not seeing anything that even remotely justifies the adding of the feature you request.

pbatard, 2020-11-09:
No update &rarr; Closed.

closed 2020-11-09 by pbatard