pbatard/rufus #1651: Making it not change the USB permission for formatting?

spirulin, 2020-11-15:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Sometimes when I create a bootable USB drive for some reason it messes up the permission of the USB and later it's a pain to format the USB. For example, I made a PopOS 20.10 live USB to install it, and later on Windows 10 it was a pain to format it due to some messed up permissions. Had to use Rufus to create a bootable drive with another distro and it finally allowed me to format it.

I am using the latest version: 3.12 but it happened before on older versions too.

pbatard, 2020-11-15:
> it messes up the permission of the USB

You're going to have to be a bit more explicit. Windows does not apply specific permissions to USB Flash Drives, so I'm tempted to say that your assumption that Rufus messes "permissions" is erroneous.

Most likely you are interpreting an `Access denied` message as a permissions problem, whereas [the FAQ clearly indicates that this is an issue with **other** applications accessing the drive that Rufus is attempting to format](https://github.com/pbatard/rufus/wiki/FAQ#Error_0x00000005_Access_is_denied).

So, please don't assume that `Access denied` means that Rufus is setting up some weird permissions on the drive, as Rufus simply does not restrict any of the file permissions when creating a drive or copying its content (which you can validate by checking our source code if needed).

I am therefore afraid that your problem lies elsewhere. Or, if you believe that it's really a permission issue, you will have to demonstrate exactly what you mean, and what permission you see that are being altered.

pbatard, 2020-11-19:
No update &rarr; Closing

closed 2020-11-19 by pbatard