pbatard/rufus #1653: Likvidace disků

Bobromil700, 2020-11-21:
﻿

Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>

Log
---
```
<FULL LOG>
```
[image]
Rufus likviduje flash disky jak je vidět na přiloženém snímku. Proto žádám o náhradu disku typu: ADATA UV150/32GB USB 3.1 
[image]

pbatard, 2020-11-21:
I'm sorry but you're going to have to explain your issue. Adding screenshots and expecting people to understand what you want from these is not the way to go.

Bobromil700, 2020-11-21:
Snažil jsem se přes program RUFUS vytvořit boot usb linuxu ale místo vyzvpření bootobatelné fleshky jí zničil jak je vydět na snímku.

pbatard, 2020-11-21:
Well, it looks to me like your hardware failed (and it would have failed regardless of the application used), and therefore Rufus reported an error.

Flash memory, especially the one used in flash drives, has a way shorter life than people imagine, and it's not uncommon at all to have brand new flash drives fail, especially cheap ones. And that's not even talking of "fake" drives, that are designed by unscrupulous people to report more capacity than they actually have (e.g. they might only have 8 GB of flash memory but report 32 GB), and that will automatically fail as soon as you copy more data than the actual capacity.

In short there are lots of ways for hardware to fail, and I can assure you that, [no, Rufus does not have the ability to destroy USB drives](https://github.com/pbatard/rufus/wiki/FAQ#Help_Rufus_damaged_my_flash_drive).

Unless you can recover that drive in another machine (because sometimes, even Windows has bugs that make it look like a drive is dead), you will just have to replace it.

Since there is nothing that can be done with Rufus, I will close this issue.

closed 2020-11-21 by pbatard

Bobromil700, 2020-11-21:
Adata není zrovna levná značka a není to poprvé co mi tento software zničil flash disk a velikost disku je  skutečně 32gb