pbatard/rufus #1654: Rufus log claims the Hebrew translation requires an update, but the Hebrew language is already up-to-date

ItielMaN, 2020-11-21:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Running Rufus 3.13 (same with 3.12, haven't tested previous versions), the log (seen below) claims the Hebrew translation requires an update, but in fact the translation is up-to-date, according to Pollock 1.3:
[image]

Note that some Fido strings aren't localized, not sure if this is related to this or not.
(Off-topic, I'd like to know how to translate those. Couldn't find anything in pbatard/Fido)

(I was going to email you about this but decided to post this here, seeing that this could potentially be a Rufus issue and not localization issue)

Log
---
```
Rufus x86 v3.13.1730 (Portable)
Windows version: Windows 10 64-bit (Build 18363.1198)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.04
System locale ID: 0x040D (he-IL)
Will use default UI locale 0x040D
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'he-IL'
NOTE: The Hebrew (עברית) translation requires an update, but the current translator hasn't submitted one. Because of this, some messages will only be displayed in English.
If you think you can help update this translation, please e-mail the author of this application
Found USB 2.0 device 'SanDisk Dual Drive USB Device' (0781:55A9)
NOTE: This device is a USB 3.0 device operating at lower speed...
Found USB 3.0 device 'JetFlash Transcend 32GB USB Device' (8564:1000)
2 devices found
Disk type: Removable, Disk size: ‎256‎ GB, Sector size: 512 bytes
Cylinders: 30399, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0xB1EF5793
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 232.9 GB (250047610880 bytes)
  Start Sector: 32, Boot: No
```

pbatard, 2020-11-21:
Hmmm, I can't seem to replicate the issue.

Are you sure you don't have an old `rufus.loc` in the same directory as the one where you have the .exe. Coz that would certainly explain your issue...

ItielMaN, 2020-11-21:
Darn it, it's the second time this happens to me!
Thanks, that was it =\

In any case, I'd still be interested in this:
> (Off-topic, I'd like to know how to translate those. Couldn't find anything in pbatard/Fido)

closed 2020-11-21 by ItielMaN

pbatard, 2020-11-21:
Yeah, just so you know I am planning a translation round for Rufus 3.14, so we'll get a chance to discuss whatever messages need to be clarified. So expect a message from me on that in the coming months...