pbatard/rufus #1656: Windows XP - latest supporting Rufus build

PaulNevada, 2020-11-29:
What is it? Please.

pbatard, 2020-11-29:
2.18 available [here](https://rufus.ie/downloads/). Note that if you choose to use this build, you are 100% unsupported.

closed 2020-11-29 by pbatard

PaulNevada, 2020-11-29:
Thank you. But can I have it from github as well?

JonnyTech, 2020-11-29:
All releases are on GitHub, if you looked you would have found it: [rufus-2.18.exe](https://github.com/pbatard/rufus/releases/download/v2.18/rufus-2.18.exe)

PaulNevada, 2020-11-29:
I have and I just wanted to know - are they totally identical to the ones came to rufus.ie... Checked hash sums already - they are...

PaulNevada, 2020-11-30:
I tried it as well as 1.3.4 - Windows XP just doesn’t load at all. As far as I know, the tool should make the usb drive as a dvd-rom type device... What’s wrong here and where can I read about that?